Barracuda09/SATPI #65: Add: Save button in web interface

Barracuda09, 2019-03-15:
At time the is no Save button in the web interface.

It will save with every change or 'Enter' key press.

Saentist, 2019-09-03:
If I can add some enhancement to request.
Is SAVE button can save config file with will be automatically loaded on next startup.
ex. 
satpi.conf located in /etc or location of satpi binary

Barracuda09, 2019-09-03:
Hi @Saentist 

Do you mean you could and an 'Save' option in the web interface?
If so I am happy to see your work.

Saentist, 2019-09-09:
> Hi @Saentist
> 
> Do you mean you could and an 'Save' option in the web interface?
> If so I am happy to see your work.

No this is more feature request. 
Just want binary automatically to save settings 
and next time to run just binary without any variables.

If someone want to be more secure:
Let's binary will be started with "**-save-config**" all modifications made will be saved, 
and next run will load them automatically.
 
This also will be more easy to deploy and error debug.