Barracuda09/SATPI #66: SYSTEMD file to start as a service?

brimur, 2019-03-16:
I have tried to start SatPI as a service but I am failing. There is something hard coded so that it will only run with relative paths. i.e if I run it as /opt/satPI/satPI then it will not run properly. I have to... cd /opt/satPI and then run it as ./satPI. 

Starting the daemon using /opt/saPI/satPI I have tried adding the app and http as parameters but the tuners still do not show up in my TVHeadend server and the web page on 8875 does not load.

Have you created a service file yourself that works? Can you share it?

Thanks

Barracuda09, 2019-03-16:
Hi,

No I have not made an service file myself, I will look into this as the web path seems indeed broken

Barracuda09, 2019-03-17:
Hi,

The problem is that the XML settings file is overriding the command options.

So the quick fix would be to delete the SatPI.xml file and try again. I will try to fix this as soon as possible. And I use this to test start SatPI:

``
/usr/satpi/satpi --no-daemon --http-path /usr/satpi/web
``

Thanks for giving this a trying

brimur, 2019-09-03:
A little late, sorry, but using the above it is running as a service fine now. Thanks

closed 2019-09-03 by brimur