Barracuda09/SATPI #67: how to use oscam

taojishou, 2019-04-18:
我配置好了，也链接到了oscam。 但是并没有发送任何数据到oscam 。。
2019/04/18 19:56:16 00000000 s (emmcache) loaded 2 emmstat records from /tmp/.oscam/oscam.emmstat in 1 ms
2019/04/18 19:56:16 00000000 s (anticasc) anti cascading disabled
2019/04/18 19:56:16 562F621A c   (client) plain dvbapi-client granted (au, au=off)
2019/04/18 19:56:16 562F621A c   (dvbapi) dvbapi channelcache loaded from /var/tuxbox/config/oscam.ccache
2019/04/18 19:56:16 562F621A c   (dvbapi) Using TCP listen socket, API forced to DVBAPIv3 (0), userconfig boxtype: 11
2019/04/18 19:56:20 562F621A c   (dvbapi) Client connected: 'SatPI 1.5.260~gb06a87' (protocol version = 2)

src/mpegts/TableData.cpp:179] Stream: 0, SDT - PID 17: sectionLength: 137  tableDataSize: 188  secNr: 0  lastSecNr: 0  currSecNr: 0
339  Thu Apr 18 08:13:39.0997 2019  6  [                           src/mpegts/SDT.cpp:061] Stream: 0, SDT - Section Length: 137  Transport Stream ID: 6200  Version: 203  secNr: 0  lastSecNr: 0  NetworkID: 6666  CRC: 0x8A3974E0
340  Thu Apr 18 08:13:39.0997 2019  6  [                           src/mpegts/SDT.cpp:086] Stream: 0,  serviceID: 0x183D - 06205  EIT: 0xFF  NetworkName: WHDTV  ChannelName: CCTV-3¸ßÇå
341  Thu Apr 18 08:13:39.0998 2019  6  [                           src/mpegts/SDT.cpp:086] Stream: 0,  serviceID: 0x1842 - 06210  EIT: 0xFF  NetworkName: WHDTV  ChannelName: CCTV-5¸ßÇå
342  Thu Apr 18 08:13:39.0998 2019  6  [                           src/mpegts/SDT.cpp:086] Stream: 0,  serviceID: 0x184C - 06220  EIT: 0xFF  NetworkName: WHDTV  ChannelName: CCTV-5+¸ßÇå
343  Thu Apr 18 08:13:39.0998 2019  6  [                           src/mpegts/SDT.cpp:086] Stream: 0,  serviceID: 0x1856 - 06230  EIT: 0xFF  NetworkName: WHDTV  ChannelName: CCTV-6¸ßÇå
344  Thu Apr 18 08:13:39.1300 2019  6  [                     src/mpegts/TableData.cpp:179] Stream: 0, PAT - PID 0: sectionLength: 25  tableDataSize: 188  secNr: 0  lastSecNr: 0  currSecNr: 0
345  Thu Apr 18 08:13:39.1301 2019  6  [                           src/mpegts/PAT.cpp:059] Stream: 0, PAT: Section Length: 25  TID: 6200  Version: 209  secNr: 0 lastSecNr: 0  CRC: 0xA023256A
346  Thu Apr 18 08:13:39.1301 2019  6  [                           src/mpegts/PAT.cpp:073] Stream: 0, PAT: Prog NR: 0x183D - 06205  PMT PID: 6212
347  Thu Apr 18 08:13:39.1301 2019  6  [                           src/mpegts/PAT.cpp:073] Stream: 0, PAT: Prog NR: 0x1842 - 06210  PMT PID: 6164
348  Thu Apr 18 08:13:39.1301 2019  6  [                           src/mpegts/PAT.cpp:073] Stream: 0, PAT: Prog NR: 0x184C - 06220  PMT PID: 6180
349  Thu Apr 18 08:13:39.1302 2019  6  [                           src/mpegts/PAT.cpp:073] Stream: 0, PAT: Prog NR: 0x1856 - 06230  PMT PID: 6196

taojishou, 2019-04-18:
oscam  received and disarranged successfully, but still could not play.

taojishou, 2019-04-18:
oscam live log
04/18 21:09:29 00C80C55 c      (ecm) au (4A02@000000/8D00/01F5/8E:40E6BC557FBCBF81605018D14AA5AAA0): found (189 ms) by card1 (L/1/3/3)
2019/04/18 21:09:44 00C80C55 c      (ecm) au (4A02@000000/8D00/01F5/8E:419D6A0F7D9047F76941FCB86F16C433): found (188 ms) by card1 (L/1/3/3)
2019/04/18 21:09:59 00C80C55 c      (ecm) au (4A02@000000/8D00/01F5/8E:FC8FDED8FA773613E28FAC1A979B7962): found (188 ms) by card1 (L/1/3/3)
2019/04/18 21:10:14 00C80C55 c      (ecm) au (4A02@000000/8D00/01F5/8E:FDEB9C9D581FD1C816D7BB6CF4B2DE86): found (189 ms) by card1 (L/1/3/3)
2019/04/18 21:10:29 00C80C55 c      (ecm) au (4A02@000000/8D00/01F5/8E:83D198306E4E7D11D88087C3959642DA): found (189 ms) by card1 (L/1/3/3)
2019/04/18 21:10:44 00C80C55 c      (ecm) au (4A02@000000/8D00/01F5/8E:1C66A394226C317A9015305B4AEB4215): found (188 ms) by card1 (L/1/3/3)

Barracuda09, 2019-04-19:
Hi,

Did you add all the necessary PIDs to the request?

taojishou, 2019-04-19:
ho to get all the PIDs？

Barracuda09, 2019-04-19:
What are the pids, you have in your request

Barracuda09, 2019-10-02:
Hi @taojishou 

Could you give this another try? I fixed some problems with PMT (or table) filtering.

Thanks for giving this a try!

Barracuda09, 2022-09-23:
This should already be fixed.

closed 2022-09-23 by Barracuda09