Barracuda09/SATPI #70: Too high cpu usage

Ntemis, 2019-10-21:
%CPU  %MEM
24698 me  20   0  248812   4176   3856 S             98,3      0,1       0:37.06 satpi       
Without doing anything(on idle)

Barracuda09, 2019-10-21:
Which thread is consuming the cpu time?

What and how did you build SatPI?

Ntemis, 2019-10-21:
run make on ubuntu bionic

Barracuda09, 2019-10-21:
Which thread? could you check with top

Ntemis, 2019-10-21:
https://ibb.co/jfSFqSL

Barracuda09, 2019-10-21:
weird.. I don't see this here on my Vu+ and Raspberry.. or on my PC with Xenial.

Barracuda09, 2019-10-21:
If you show the threads in top

Barracuda09, 2019-10-21:
<shift> H

Ntemis, 2019-10-21:
25440 demetris  20   0  248812   4120   3800 R 97,4  0,1  10:07.98 HttpServer

Barracuda09, 2019-10-21:
Strange.. have to try this on bionic probably.  Which gcc version do you use in bionic?

Ntemis, 2019-10-21:
gcc (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0

Barracuda09, 2019-10-22:
Is that 64-bit or 32-bit

Ntemis, 2019-10-22:
64bit

Barracuda09, 2019-10-22:
Strange, I can not seem to reproduce this in 64 or 32 bit Ubuntu bionic.

You have done make clean and try make again. Or do you have some more info I can reproduce this?

Ntemis, 2019-10-22:
No because this was a fresh make. I can try and let you know