Barracuda09/SATPI #71: How to advertize as DVB-C only

lars18th, 2019-10-22:
Hi @Barracuda09 ,

I want to configure one SATPI instance with only `file` and `childpipe` inputs and advertized as DVB-C. 
The idea is to use only the `mapping.m3u` without any real tuner.

Any idea about how to achieve this?
Thank you.

Barracuda09, 2019-10-22:
Hi @lars18th 

No at the moment you can only advertise as DVB-S2

lars18th, 2019-10-23:
> No at the moment you can only advertise as DVB-S2

Ok. Thank you to comment it.
Regards.

Barracuda09, 2019-10-23:
you could change it in the mean time in:
void TSReader::addDeliverySystemCount(..) -> _transform.advertiseAsDVBS2()

lars18th, 2019-10-23:
> you could change it in the mean time in:
> void TSReader::addDeliverySystemCount(..) -> _transform.advertiseAsDVBS2()

:+1: 
https://github.com/Barracuda09/SATPI/blob/b7f09c665e1b682e9cb1ab30071d63a26a86f4a4/src/input/childpipe/TSReader.cpp#L89

Thank you!

lars18th, 2019-11-26:
Hi @Barracuda09 ,

Previous to close this issue (it's resolved) I ask if you want to add to the option to advertize as DVB-T (and/or DVB-T2). I feel it will be also useful.

Regards.

lars18th, 2020-10-03:
All is done. So I close the issue.
Thank you!

closed 2020-10-03 by lars18th