Barracuda09/SATPI #72: Problem: Mapping with multicast doesn't work

lars18th, 2019-10-22:
Hi @Barracuda09 ,

I have configured one MPTS in the multicat address @224.0.0.1:12345 (plain UDP). So if I configure the `mapping.m3u` with 
```
#EXTM3U

#EXTINF:-1 satip-freq="10744", Translation to: Stream 1 - Multicast UDP input
udp://@224.0.0.1:12345
```
Then when doing some test with VLC as a client (HTTP protocol):
- `http://satpi:8875/?msys=streamer&uri=udp://224.0.0.1:12345` it works
- `http://satpi:8875/?freq=10744&msys=dvbs2` it doesn't work

Any idea?

Barracuda09, 2019-10-22:
Hi @lars18th 

Sorry, I am not sure what you mean. Could you explain the test in more detail?

lars18th, 2019-10-23:
> Sorry, I am not sure what you mean. Could you explain the test in more detail?

OK. I try it:

- In my network I start a process that multicasts an entire MPTS from a transporter to the multicast address @224.0.0.1:12345.
- Then I start in another machine the SATPI process with the file `mapping.m3u` described above.
- Then using the VideoLAN I open the network address: `http://satpi:8875/?msys=streamer&uri=udp://224.0.0.1:12345`. And then in the VLC window I can see the video. So all is OK.
- However, if I open in the VideoLAN the address `http://satpi:8875/?freq=10744&msys=dvbs2` then the LOG of the SAPTI indicates that the translation is selected, but I don't receive anything. In fact, in the machine when the SATPI process runs I can't see any traffic from the multicast (using the tool `iftop`).

Perhaps the remapping only works when translating to one SAT>IP command to another SAT>IP command, and not when targetting an unicast/multicast input?

I hope it's more clear now.
Regards.

Note: `satpi` is the network name of the machine where the SATPI process runs. So the URL `http://satpi:8875` is resolved to the correct IP address.

Barracuda09, 2019-10-23:
Ok thanks @lars18th I will try this out, but I am not sure indeed if streamer has transform capability. I Have to check this.

This is in TSReader:
https://github.com/Barracuda09/SATPI/blob/b7f09c665e1b682e9cb1ab30071d63a26a86f4a4/src/input/file/TSReader.h#L126

lars18th, 2019-10-23:
> Ok thanks @lars18th I will try this out, but I am not sure indeed if streamer has transform capability. I Have to check this.
> 
> This is in TSReader:
> https://github.com/Barracuda09/SATPI/blob/b7f09c665e1b682e9cb1ab30071d63a26a86f4a4/src/input/file/TSReader.h#L126

Thank you to check it. In any case, I do the test with a DVB-S2 tuner at frontend 0. But, I prefer that it works with both: when using a real tuner and when using virtual inputs (file and pipe).

Regards.

lars18th, 2019-10-25:
Hi @Barracuda09 ,

The problem is not related with the multicast input, but with the _translation_ support.

If I try to translate to some "childpipe" the result is the same... no output.
The LOG prints this:
```
150  Fri Oct 25 13:25:06.4798 2019  6  Stream: 0, Parsing transport parameters...
151  Fri Oct 25 13:25:06.4799 2019  6  Stream: 0, Request Transformed
152  Fri Oct 25 13:25:06.4799 2019  7  Stream: 0, Request Transformed to SETUP rtsp://127.0.0.1:8554/?msys=childpipe&exec=cat%20test.ts RTSP/1.0
153  Fri Oct 25 13:25:06.4800 2019  3  Stream: 0, Not supported delivery system
154  Fri Oct 25 13:25:06.4800 2019  7  Stream: 0, Parsing transport parameters (Finished)
155  Fri Oct 25 13:25:06.4801 2019  7  Stream: 0, Found Streaming type: RTSP Unicast
```
This is with one DVB-S tuner configured (tuner 0), and with this `mapping.m3u`:
```
#EXTINF:-1 satip-freq="11111", Translation to: Stream 1
rtsp://127.0.0.1:8554/?msys=childpipe&exec=cat%20test.ts
```

So I feel something is bad in the code when translating from DVB-* to `file`,`childpipe` and `udp/rtp`.

Please, can you check it?
Thank you!

Barracuda09, 2019-10-25:
Hi @lars18th 

Yes, I will check this out. I believe it is not implemented for file or childpipe. But it has been a will that I did some work on this so I need to dive into it.

lars18th, 2019-10-25:
> Hi @lars18th
> 
> Yes, I will check this out. I believe it is not implemented for file or childpipe. But it has been a will that I did some work on this so I need to dive into it.

OK. Thank you for targeting this issue.
Regards.

lars18th, 2019-11-06:
Hi @Barracuda09 ,

I'm testing the _last_ version (after the childpipe changes), and this problem continues. The MULTICAST input doesn't work at all!

- Here my `mapping.m3u`:
```
#EXTINF:-1 satip-freq="11494", Translation to: Stream 1 - Multicast UDP input
udp://@239.1.1.1:12345
```

And when compiling with `make debug` and run it the LOG is...
```
[                        src/StreamManager.cpp:170] Found StreamID x - SessionID: 0288515163
[                               src/Stream.cpp:215] Stream: 0, StreamClient[0] with SessionID 0288515163 for dvbs2
[                   src/input/dvb/Frontend.cpp:422] Stream: 0, Parsing transport parameters...
[                 src/input/Transformation.cpp:150] Stream: 0, Request Transformed
[                 src/input/Transformation.cpp:183] Stream: 0, Request Transformed to SETUP udp://@239.1.1.1:12345 RTSP/1.0
[                   src/input/dvb/Frontend.cpp:429] Stream: 0, Parsing transport parameters (Finished)
[                               src/Stream.cpp:270] Stream: 0, Found Streaming type: RTSP Unicast
[                   src/input/dvb/Frontend.cpp:433] Stream: 0, Updating frontend...
[                   src/input/dvb/Frontend.cpp:720] Stream: 0, Opened /dev/dvb/adapter0/frontend0 fd: 10
[                          src/HttpcServer.cpp:204] RTSP/1.0 200 OK
```

But the server doesn't receive the stream @239.1.1.1:12345 at all (tested with `iftop`), and the SATPI doesn't read anything from the network.

Futhermore, I noticed that when using the address @224.0.0.1:12345 the server receives the stream, but the SATPI not. The difference here is that this address is registered for switch multicast notification.

So, please, check the code.
Thank you!

Barracuda09, 2019-11-06:
Hi @lars18th 

Yes the previouse fix was only for #69 and #71.

With this fix I am busy testing myself. By the way it will be something:

```
#EXTINF:-1 satip-freq="202", Translation to: Streamer - Multicast UDP input
rtsp://%1/?msys=streamer&uri=udp@224.0.1.3:12345
```

lars18th, 2019-11-06:
> Hi @lars18th
> 
> Yes the previouse fix was only for #69 and #71.
> 
> With this fix I am busy testing myself. By the way it will be something:
> 
> ```
> #EXTINF:-1 satip-freq="202", Translation to: Streamer - Multicast UDP input
> rtsp://%1/?msys=streamer&uri=udp@224.0.1.3:12345
> ```

Hi @Barracuda09 ,

Thank you for your fast response.
Then I'll wait until you will push a new fix for testing it.
Regards.

lars18th, 2019-11-06:
Hi @Barracuda09 ,

And another comment: With the last version the `TS Reader` and `Child PIPE` frontends have the option `Advertise as` but not the frontend `Streamer`. Why not? I feel it will have it, right?

Regards.

Barracuda09, 2019-11-06:
Yes it will have it after my fix. At the moment is does not.

lars18th, 2019-11-11:
Hi @Barracuda09 ,

After the last commits the problem contiues, even with the new format in the translation `mapping.m3u` file... I think two things require a fix:

1. The `Streamer` frontend doesn't have any option for remapping. That's annoying. The other two (TSReader and ChildPIPE) have this functionality. So in order to remap to some multicast address we need to use (and enable) almost one real tuner. I think this requirement doesn't have sense. Please, can you enable the functionality of remapping to all virtual tuners?
2. When using the remap with a real tuner, it doesn't reads anything from the multicast address when using `#EXTINF:-1 satip-freq="10744", Translation to: Stream 1 - Multicast UDP input
rtsp://127.0.0.1:8554/?msys=streamer&uri=udp@239.0.0.1:12345`. And I feel this is a bug, as the socket doesn't reads anything from the multicast address. Perhaps you don't join the multicast group prior to open the socket. Please, can you target this problem?

Thank you for this great SAT>IP server!

Barracuda09, 2019-11-11:
Hi @lars18th

Latest commit, adds transformation to 'Virtual' Tuner Streamer

lars18th, 2019-11-11:
> Hi @lars18th
> 
> Latest commit, adds transformation to 'Virtual' Tuner Streamer

Yes! Now the options appear, and I feel the multicast input works too! :+1:

lars18th, 2019-11-26:
Hi @Barracuda09 ,

I'm closing it because it's already fixed.
Thank you!

closed 2019-11-26 by lars18th