Barracuda09/SATPI #73: DD maxsx8 dont get attached

mfu-mcosys, 2019-10-22:
It seems like the device cant attached because satpi cant read the signal stuff? Everything related to "monitor" is 0.

But minsatip and tvh works.

dvb-fe-tool -a 0 -m
Lock   (0x1f) Signal= 29,88dBm C/N= 15,08dB
Lock   (0x1f) Signal= 30,32dBm C/N= 15,10dB
Lock   (0x1f) Signal= 29,88dBm C/N= 15,03dB

`./satpi --http-path /usr/src/satpi/web/ --no-daemon --app-data-path /usr/src/satpi/ --iface-name enp0s20f0 

 src/Satpi.cpp:456] --- Starting SatPI version: 1.5.280~gb7f09c6 ---
[                                src/Satpi.cpp:457] Number of processors online: 8
[                                src/Satpi.cpp:458] Default network buffer size: 104 KBytes
[                        src/InterfaceAttr.cpp:130] enp0s20f0: 192.168.81.4 [0c:c4:7a:31:7e:30]
[                        src/StreamManager.cpp:070] Current DVB_API_VERSION: 5.11
[                        src/StreamManager.cpp:071] Enumerating all devices...
[                   src/input/dvb/Frontend.cpp:167] Detecting frontends in: /dev/dvb
[                src/decrypt/dvbapi/Client.cpp:370] Setting up DVBAPI client
[                   src/input/dvb/Frontend.cpp:436] Frontend Name: Digital Devices MaxSX8 MCI DVB-S/S2/S2X
[                   src/input/dvb/Frontend.cpp:509] Frontend Type: Satellite (DVB-S)
[                   src/input/dvb/Frontend.cpp:513] Frontend Type: Satellite (DVB-S2)
[                   src/input/dvb/Frontend.cpp:555] Frontend Freq: 950000 Hz to 2150000 Hz
[                   src/input/dvb/Frontend.cpp:556] Frontend srat: 100000 symbols/s to 100000000 symbols/s
[                   src/input/dvb/Frontend.cpp:436] Frontend Name: Digital Devices MaxSX8 MCI DVB-S/S2/S2X
[                   src/input/dvb/Frontend.cpp:509] Frontend Type: Satellite (DVB-S)
[                   src/input/dvb/Frontend.cpp:513] Frontend Type: Satellite (DVB-S2)
[                   src/input/dvb/Frontend.cpp:555] Frontend Freq: 950000 Hz to 2150000 Hz
[                   src/input/dvb/Frontend.cpp:556] Frontend srat: 100000 symbols/s to 100000000 symbols/s
[                   src/input/dvb/Frontend.cpp:436] Frontend Name: Digital Devices MaxSX8 MCI DVB-S/S2/S2X
[                   src/input/dvb/Frontend.cpp:509] Frontend Type: Satellite (DVB-S)
[                   src/input/dvb/Frontend.cpp:513] Frontend Type: Satellite (DVB-S2)
[                   src/input/dvb/Frontend.cpp:555] Frontend Freq: 950000 Hz to 2150000 Hz
[                   src/input/dvb/Frontend.cpp:556] Frontend srat: 100000 symbols/s to 100000000 symbols/s
[                   src/input/dvb/Frontend.cpp:436] Frontend Name: Digital Devices MaxSX8 MCI DVB-S/S2/S2X
[                   src/input/dvb/Frontend.cpp:509] Frontend Type: Satellite (DVB-S)
[                   src/input/dvb/Frontend.cpp:513] Frontend Type: Satellite (DVB-S2)
[                   src/input/dvb/Frontend.cpp:555] Frontend Freq: 950000 Hz to 2150000 Hz
[                   src/input/dvb/Frontend.cpp:556] Frontend srat: 100000 symbols/s to 100000000 symbols/s
[                   src/input/dvb/Frontend.cpp:436] Frontend Name: Digital Devices MaxSX8 MCI DVB-S/S2/S2X
[                   src/input/dvb/Frontend.cpp:509] Frontend Type: Satellite (DVB-S)
[                   src/input/dvb/Frontend.cpp:513] Frontend Type: Satellite (DVB-S2)
[                   src/input/dvb/Frontend.cpp:555] Frontend Freq: 950000 Hz to 2150000 Hz
[                   src/input/dvb/Frontend.cpp:556] Frontend srat: 100000 symbols/s to 100000000 symbols/s
[                   src/input/dvb/Frontend.cpp:436] Frontend Name: Digital Devices MaxSX8 MCI DVB-S/S2/S2X
[                   src/input/dvb/Frontend.cpp:509] Frontend Type: Satellite (DVB-S)
[                   src/input/dvb/Frontend.cpp:513] Frontend Type: Satellite (DVB-S2)
[                   src/input/dvb/Frontend.cpp:555] Frontend Freq: 950000 Hz to 2150000 Hz
[                   src/input/dvb/Frontend.cpp:556] Frontend srat: 100000 symbols/s to 100000000 symbols/s
[                   src/input/dvb/Frontend.cpp:436] Frontend Name: Digital Devices MaxSX8 MCI DVB-S/S2/S2X
[                   src/input/dvb/Frontend.cpp:509] Frontend Type: Satellite (DVB-S)
[                   src/input/dvb/Frontend.cpp:513] Frontend Type: Satellite (DVB-S2)
[                   src/input/dvb/Frontend.cpp:555] Frontend Freq: 950000 Hz to 2150000 Hz
[                   src/input/dvb/Frontend.cpp:556] Frontend srat: 100000 symbols/s to 100000000 symbols/s
[                   src/input/dvb/Frontend.cpp:436] Frontend Name: Digital Devices MaxSX8 MCI DVB-S/S2/S2X
[                   src/input/dvb/Frontend.cpp:509] Frontend Type: Satellite (DVB-S)
[                   src/input/dvb/Frontend.cpp:513] Frontend Type: Satellite (DVB-S2)
[                   src/input/dvb/Frontend.cpp:555] Frontend Freq: 950000 Hz to 2150000 Hz
[                   src/input/dvb/Frontend.cpp:556] Frontend srat: 100000 symbols/s to 100000000 symbols/s
[                   src/input/dvb/Frontend.cpp:170] Frontends found: 8

[image]
[image]

Barracuda09, 2019-10-23:
Hi @mfu-mcosys 

Thanks for reminding me, because I still read these stats the 'old' way.

See in function void Frontend::monitorSignal(const bool showStatus) in src/input/dvb/Frontend.cpp

I have to read this via FE_GET_PROPERTY and if that fails the 'old' way should be used.

Barracuda09, 2019-10-24:
Hi @mfu-mcosys 

Could you try with latest commit

mfu-mcosys, 2019-10-24:
src/Satpi.cpp:456] --- Starting SatPI version: 1.5.283~g5f9ea79 ---

no changes, still not attached and no signal :/

Barracuda09, 2019-10-24:
Could you please provide an log of an tuning (syslog)

mfu-mcosys, 2019-10-24:
you i should log satpi while i request some channels? Or a debug-log from the dvb-device?

Barracuda09, 2019-10-24:
Just the syslog from start of satpi and one request

mfu-mcosys, 2019-10-24:
/usr/src/satpi # ./satpi --http-path /usr/src/satpi/web/ --no-daemon --app-data-path /usr/src/satpi/ --iface-name enp0s20f0 
[                                src/Satpi.cpp:456] --- Starting SatPI version: 1.5.283~g5f9ea79 ---
[                                src/Satpi.cpp:457] Number of processors online: 8
[                                src/Satpi.cpp:458] Default network buffer size: 104 KBytes
[                        src/InterfaceAttr.cpp:130] enp0s20f0: 192.168.81.4 [0c:c4:7a:31:7e:30]
[                src/decrypt/dvbapi/Client.cpp:370] Setting up DVBAPI client
[                        src/StreamManager.cpp:070] Current DVB_API_VERSION: 5.11
[                        src/StreamManager.cpp:071] Enumerating all devices...
[                   src/input/dvb/Frontend.cpp:173] Detecting frontends in: /dev/dvb
[                   src/input/dvb/Frontend.cpp:502] Frontend Name: Digital Devices MaxSX8 MCI DVB-S/S2/S2X
[                   src/input/dvb/Frontend.cpp:575] Frontend Type: Satellite (DVB-S)
[                   src/input/dvb/Frontend.cpp:579] Frontend Type: Satellite (DVB-S2)
[                   src/input/dvb/Frontend.cpp:621] Frontend Freq: 950000 Hz to 2150000 Hz
[                   src/input/dvb/Frontend.cpp:622] Frontend srat: 100000 symbols/s to 100000000 symbols/s
[                   src/input/dvb/Frontend.cpp:502] Frontend Name: Digital Devices MaxSX8 MCI DVB-S/S2/S2X
[                   src/input/dvb/Frontend.cpp:575] Frontend Type: Satellite (DVB-S)
[                   src/input/dvb/Frontend.cpp:579] Frontend Type: Satellite (DVB-S2)
[                   src/input/dvb/Frontend.cpp:621] Frontend Freq: 950000 Hz to 2150000 Hz
[                   src/input/dvb/Frontend.cpp:622] Frontend srat: 100000 symbols/s to 100000000 symbols/s
[                   src/input/dvb/Frontend.cpp:502] Frontend Name: Digital Devices MaxSX8 MCI DVB-S/S2/S2X
[                   src/input/dvb/Frontend.cpp:575] Frontend Type: Satellite (DVB-S)
[                   src/input/dvb/Frontend.cpp:579] Frontend Type: Satellite (DVB-S2)
[                   src/input/dvb/Frontend.cpp:621] Frontend Freq: 950000 Hz to 2150000 Hz
[                   src/input/dvb/Frontend.cpp:622] Frontend srat: 100000 symbols/s to 100000000 symbols/s
[                   src/input/dvb/Frontend.cpp:502] Frontend Name: Digital Devices MaxSX8 MCI DVB-S/S2/S2X
[                   src/input/dvb/Frontend.cpp:575] Frontend Type: Satellite (DVB-S)
[                   src/input/dvb/Frontend.cpp:579] Frontend Type: Satellite (DVB-S2)
[                   src/input/dvb/Frontend.cpp:621] Frontend Freq: 950000 Hz to 2150000 Hz
[                   src/input/dvb/Frontend.cpp:622] Frontend srat: 100000 symbols/s to 100000000 symbols/s
[                   src/input/dvb/Frontend.cpp:502] Frontend Name: Digital Devices MaxSX8 MCI DVB-S/S2/S2X
[                   src/input/dvb/Frontend.cpp:575] Frontend Type: Satellite (DVB-S)
[                   src/input/dvb/Frontend.cpp:579] Frontend Type: Satellite (DVB-S2)
[                   src/input/dvb/Frontend.cpp:621] Frontend Freq: 950000 Hz to 2150000 Hz
[                   src/input/dvb/Frontend.cpp:622] Frontend srat: 100000 symbols/s to 100000000 symbols/s
[                   src/input/dvb/Frontend.cpp:502] Frontend Name: Digital Devices MaxSX8 MCI DVB-S/S2/S2X
[                   src/input/dvb/Frontend.cpp:575] Frontend Type: Satellite (DVB-S)
[                   src/input/dvb/Frontend.cpp:579] Frontend Type: Satellite (DVB-S2)
[                   src/input/dvb/Frontend.cpp:621] Frontend Freq: 950000 Hz to 2150000 Hz
[                   src/input/dvb/Frontend.cpp:622] Frontend srat: 100000 symbols/s to 100000000 symbols/s
[                   src/input/dvb/Frontend.cpp:502] Frontend Name: Digital Devices MaxSX8 MCI DVB-S/S2/S2X
[                   src/input/dvb/Frontend.cpp:575] Frontend Type: Satellite (DVB-S)
[                   src/input/dvb/Frontend.cpp:579] Frontend Type: Satellite (DVB-S2)
[                   src/input/dvb/Frontend.cpp:621] Frontend Freq: 950000 Hz to 2150000 Hz
[                   src/input/dvb/Frontend.cpp:622] Frontend srat: 100000 symbols/s to 100000000 symbols/s
[                   src/input/dvb/Frontend.cpp:502] Frontend Name: Digital Devices MaxSX8 MCI DVB-S/S2/S2X
[                   src/input/dvb/Frontend.cpp:575] Frontend Type: Satellite (DVB-S)
[                   src/input/dvb/Frontend.cpp:579] Frontend Type: Satellite (DVB-S2)
[                   src/input/dvb/Frontend.cpp:621] Frontend Freq: 950000 Hz to 2150000 Hz
[                   src/input/dvb/Frontend.cpp:622] Frontend srat: 100000 symbols/s to 100000000 symbols/s
[                   src/input/dvb/Frontend.cpp:176] Frontends found: 8
[                  src/input/file/TSReader.cpp:052] Setting up TS Reader using path: /usr/src/satpi/
[                src/input/stream/Streamer.cpp:054] Setting up TS Streamer
[             src/input/childpipe/TSReader.cpp:052] Setting up Child PIPE - TS Reader using path: /usr/src/satpi/
[                           src/Properties.cpp:074] Setting HTTP Port to: 8875
[                           src/Properties.cpp:079] Setting RTSP Port to: 554
[                           src/Properties.cpp:083] Setting WEB Path to: /usr/src/satpi/web/
[                           src/Properties.cpp:087] Setting App Data Path to: /usr/src/satpi/
[                           src/HttpServer.cpp:057] Setting up HTTP server
[                           src/RtspServer.cpp:046] Setting up RTSP server
[... cut ...]
ClientChallenge: 9e26d33f2984236010ef6253fb1887f7
PlayerStarttime: [28/03/2003:22:50:23 00:00]
CompanyID: KnKV4M4I/B2FjJ1TToLycw==
GUID: 00000000-0000-0000-0000-000000000000
RegionData: 0
ClientID: Linux_2.4_6.0.9.1235_play32_RN01_EN_586
--- LINE END ---
[                          src/HttpcServer.cpp:204] RTSP/1.0 200 OK
CSeq: 1
Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
--- LINE END ---
[                     src/socket/TcpSocket.cpp:109] RTSP Client 192.168.81.253:53686 Connection closed with fd: 11

mfu-mcosys, 2019-10-24:
oO was testing this stream with minisatip, sounds like it was a radio channel from a generated playlist.. but should not matter i guess :D

Barracuda09, 2019-10-24:
What version of VLC are you using

mfu-mcosys, 2019-10-24:
VLC media player 2.2.8 Weatherwax (revision 2.2.7-14-g3cc1d8cba9)

Barracuda09, 2019-10-24:
Try this
`http://192.168.81.4:8875/?src=1&freq=12265.50&pol=h&ro=0.35&msys=dvbs2&mtype=qpsk&plts=on&sr=27500&fec=34&pids=0,17,18,520,521`

mfu-mcosys, 2019-10-24:
uhhh, this works. great! but.. with vlc-3.0.8 (windows) and rdp it dont work too, so it seems like a rdp problem for now?

Barracuda09, 2019-10-24:
The signal is now Ok ?

mfu-mcosys, 2019-10-24:
just short info.. the radio stream ""http://192.168.81.4:8875/?src=1&freq=12265.50&pol=h&ro=0.35&msys=dvbs2&mtype=qpsk&plts=on&sr=27500&fec=34&pids=0,17,18,520,521"

But still no signal on the interface:

[image]

A freq with video dont work

Barracuda09, 2019-10-24:
Strange here vlc-3.0.6 works with rtsp...

Well fudge.. just show that the signal is not read in HTTP mode here as well..

mfu-mcosys, 2019-10-24:
Video works too (need to remove the mapping.m3u)... 

[                 src/input/Transformation.cpp:150] Stream: 0, Request Transformed to GET #udp://@224.0.0.1:12345&pids=0,17,18,5100,5101,5102,5103,5104,5105,5106 RTSP/1.0
[                   src/input/dvb/Frontend.cpp:426] Stream: 0, Parsing transport parameters (Finished)

:)

Something is confusing.. i see the stream (ARD) in vlc but the satpi-iface tells me "there is nothing" :)

OK, but the tunercard works now, so, thx! :)

Barracuda09, 2019-10-24:
Yes I will look in the other problems later this day

Thanks for giving this a try

Barracuda09, 2019-10-24:
You can disable the `Transformation enable` per frontend in the web interface

mfu-mcosys, 2019-10-24:
yeah, but this just work if no content is in the input for the mapping :)

Barracuda09, 2019-10-25:
> yeah, but this just work if no content is in the input for the mapping :)

Ok, that is not how it should work. I look into that as well.