Barracuda09/SATPI #74: Bug when SETUP with pids=none

lars18th, 2019-11-06:
Hi @Barracuda09 ,

This is tested with the _last_ version...

When using a SAT>IP client that initializes the session with a SETUP command _without_ `pids=` or _with_ `pids=0...` (aka, with any list) then the SATPI works. No problem at all.

However, if the client uses the SETUP command with `pids=none` then the SATPI crash:
```
--- LINE END ---
[                          src/HttpcServer.cpp:150] RTSP Stream data from client None with IP 192.168.1.27 on Port 49771: SETUP rtsp://192.168.1.27:8554/?src=1&freq=10773&msys=dvbs2&mtype=8psk&pol=h&sr=22000&fec=34&ro=0.35&plts=off&pids=none RTSP/1.0
CSeq: 2
Transport: RTP/AVP;unicast;client_port=6504-6505
--- LINE END ---
[                        src/StreamManager.cpp:170] Found StreamID x - SessionID: 0261489270
[                               src/Stream.cpp:215] Stream: 0, StreamClient[0] with SessionID 0261489270 for dvbs2
[                   src/input/dvb/Frontend.cpp:422] Stream: 0, Parsing transport parameters...
[               src/input/dvb/FrontendData.cpp:136] Stream: 0, New frequency requested, clearing old channel data...
terminate called after throwing an instance of 'std::invalid_argument'
  what():  stoi
Aborted
```

Please, can you fix this case?
Thank you!

Barracuda09, 2019-11-06:
Ofcourse I will fix this as well.. Thanks for you testing and persistency!

lars18th, 2019-11-06:
FYI @Barracuda09 ,

Identical problem if the SETUP commands uses `pids=all`.

lars18th, 2019-11-11:
I'm testing it. :wink:

lars18th, 2019-11-20:
The crash is fixed. And although the issues #81 and #75 aren't fixed, I close this to reduce the list of opened issues.

Thank you @Barracuda09 ! :wink:

closed 2019-11-20 by lars18th