Barracuda09/SATPI #76: DESCRIBE should follow Transformed response

Barracuda09, 2019-11-08:
Hi @Barracuda09 ,

Another issue detected with the ChildPIPE frontend: The RTCP packets are not confirmant. See the content (from the LOG) when using it with the real frontend disabled and the "transformation" and "advertise" enabled for the virtual frontend:
```
v=0
o=- 0253037672 0253037672 IN IP4 192.168.1.27
s=SatIPServer:1 0,0,0
t=0 0
m=video 0 RTP/AVP 33
c=IN IP4 0.0.0.0
a=control:stream=3
a=fmtp:33 ver=1.5;tuner=4,240,1,15;exec=None
a=sendonly
```

So, when using the _TRANSFORMATION_ functionality the `a=fmtp:33 ...` line requires to be conformant with the selected _ADVERSITE_ mode. If not then some clients refuse to play.

Regards.

_Originally posted by @lars18th in https://github.com/Barracuda09/SATPI/issues/69#issuecomment-550260961_

lars18th, 2019-11-11:
However, I think this issue requires low priority. Not all SAT>IP clients parse the full `a=fmtp:33` line.

lars18th, 2021-03-13:
> However, I think this issue requires low priority. Not all SAT>IP clients parse the full `a=fmtp:33` line.

I've discovered some clients that parse this line. They use it for checking if the requested frequency are tuned when changing the transponder. So, it's required to provide a correct output for these clients.

Barracuda09, 2022-11-12:
Should be fixed with 7107fa566541e6c443887188c8c4793dac38a4ef

lars18th, 2022-11-13:
> Should be fixed with [7107fa5](https://github.com/Barracuda09/SATPI/commit/7107fa566541e6c443887188c8c4793dac38a4ef)

Yes. So thank you. Only try to fix the current empty pid list bug:
`a=fmtp:33 ver=1.0;src=1;tuner=4,240,1,15,11953.5000,h,dvbs,qpsk,off,27500,34;pids=`
(described in https://github.com/Barracuda09/SATPI/issues/169#issuecomment-1312714160)

So if you want you can close this issue.

Barracuda09, 2022-11-13:
Close as requested

closed 2022-11-13 by Barracuda09