Barracuda09/SATPI #77: Bug with the childPIPE, subprocess not killed

Barracuda09, 2019-11-08:
Hi @Barracuda09 ,

I found a __SERIOUS__ bug with the childPIPE !!!

- The child process is not killed. When using something like `exec=wget -qO- ...` then multiple "wget" processes are created (check it with `ps -aex | grep wget`). In fact, for every _new_ call a new process is created.

Please, check to kill the subprocess every time you _close_ the stream.
Thank tou!

_Originally posted by @lars18th in https://github.com/Barracuda09/SATPI/issues/69#issuecomment-550274249_

lars18th, 2019-11-11:
Hi @Barracuda09 ,

This problem is not fixed and it continues. I can't do futher tests until this is resolved, as the server collapses with a lot of child process. So testing is difficult with last commit.

I hope you can target it soon.
Thank you!

Barracuda09, 2019-11-11:
Hi @lars18th 

Do you have an example using wget to read from something. I can not seem to get a use case.

lars18th, 2019-11-11:
> Do you have an example using wget to read from something. I can not seem to get a use case.

You can use the well-kown `udproxy` or similar UDP-2-HTTP relay : https://github.com/pcherenkov/udpxy
So you can convert any multicast to HTTP... then you can receive the stream with `exec=wget -qO- http://udpxy_ip:5056/udp/239.0.0.1:12345`.

Regards.

Barracuda09, 2019-11-12:
This should be fixed now.
```
exec=wget -qO- http://udpxy_ip:5056/udp/239.0.0.1:12345
```

should be something like this (I think):
```
exec=wget%20-qO-%20http:%2F%2Fudpxy_ip:5056%2Fudp%2F239.0.0.1:12345
```

lars18th, 2019-11-20:
Thank you @Barracuda09 ,

The issue is fixed with commit https://github.com/Barracuda09/SATPI/commit/6257b4b3e0b6f82f6b289129525a36f23fca992a

So, please close it. :wink:

Barracuda09, 2019-11-26:
Thanks, and as requested by @lars18th I will close this-one.

closed 2019-11-26 by Barracuda09