Barracuda09/SATPI #79: Suggestion: PID list in Frontend page

lars18th, 2019-11-11:
Hi,

To more easy debug it will be interesting to print in the Frontend page (similar to the `Path` entry) the PIDS List. And perhaps too the last (or current) RTSP command (PLAY, OPTIONS, DESCRIBE, TEARDOWN) with the used Stream.

I hope you agree @Barracuda09 .
Regards.

Barracuda09, 2019-11-11:
Hi @lars18th 

You mean it for virtual tuners? as for an real tuner the PID list is shown.

lars18th, 2019-11-12:
> You mean it for virtual tuners?

YES, please!

lars18th, 2021-05-10:
Hi @Barracuda09 ,

To be more precise in this case too: In the "Frontend Info" page, in the section "Stream Info", the idea is to add a row indicating: `Session pids:   0,16,17,18,20,21,100,101`.

You agree?