Barracuda09/SATPI #81: BUG: After PLAY &pids=none the Streaming fail

lars18th, 2019-11-19:
Hi @Barracuda09 ,

I'm not sure if this problem is related to _all_ frontends or only to the _Streamer_, but I detected it using this frontend _with_ a translation.

One SAT>IP client sends the SETUP command, then different PLAY commands with addpids, and after a PLAY with pids=none. I all cases it uses the session delivered in the first SETUP. However, the problem is that after the "pids=none" it's impossible to receive anything. When using this session with a PLAY addpids the client doesn't receive anything.

Please, note that I'm using _translation_ and the input is a multicast address using the _Streamer_ frontend.

Please, try to reproduce it!
Regards.

lars18th, 2019-11-19:
Hi @Barracuda09 ,

After more tests the problem isnt' related to "pids=none", but this a second PLAY-SETUP after initial SETUP. So, it's like this:

- SETUP: OK
- PLAY: OK (streaming starts)
- PLAY with change pids: OK.
- PLAY with new parameters (new frequency, but the freq it's the same): FAIL.

Any idea?

lars18th, 2019-11-19:
Hi @Barracuda09 ,

This ditry patch resolves the problem:
```
diff --git a/src/Stream.cpp b/src/Stream.cpp
index a9bf59b..26c95a1 100644
--- a/src/Stream.cpp
+++ b/src/Stream.cpp
@@ -360,7 +360,7 @@ bool Stream::processStreamingRequest(const std::string &msg, const int clientID,
        // Channel changed?.. stop/pause Stream
        if (_device->hasDeviceDataChanged()) {
                _device->clearMPEGFilters();
-               if (_streaming) {
+               if (_streaming && method == "SETUP") {
                        _streaming->pauseStreaming(clientID);
                }
        }
```

However, I feel it's not very orthodox.
Please, can you try it?
Regards.

lars18th, 2019-11-20:
Hi @Barracuda09 ,

With the previous patch the problem is only solved partially. So please, can you try to fix the issue? The problema is related to a PLAY command with _new_ tunning parameters over current stream. In this case the retune fails (almost with the Streamer frontend and with translation).

I hope you provide a solution soon.
Thank you for supporting this project!

Barracuda09, 2019-11-20:
Hi @lars18th 

Thanks for testing.

Yes I will try to solve this problem as soon as I can. I have to rethink this logic when 'changing' channels.

lars18th, 2019-11-26:
Hi @Barracuda09 ,

I'm not sure if this the best solution, however it works in my setup: PR #83 .

Please, review it.
Regards.

Barracuda09, 2019-11-26:
Hi @lars18th 

Thanks for you PR.

But I feel there is a deeper problem in the logic when changing channels. When in the PLAY requested is an 'freq='

I had this bug, reported by the programmer of NextPVR not so long ago. I did some testing back then and came up with some solution as well, which I only tested shortly. But the problem was a bug in TvHeadend #4809 which I tried to workaround. but in time I lost why and how exactly.

lars18th, 2019-11-27:
Hi @Barracuda09 ,

My PR #83 works but don't fix the problem: in case you change the `freq=` then the frontend isn't retuned (I don't see the LOG message about the remapping `Request transformed to SETUP ...`). So it's true: the bug continues.

Futhermore, some comments about the SAT>IP logic:
- After a SETUP command the pid list can or cannot be deleted. The reason is that in a multicast environment the "stream" can be shared. And that's true even in a unicast environment.
- The SAT>IP server needs to accept `addpids=` and `delpids=` commands with REPEATED values. So, for example, if the stream has pid 0 and you receive the command to add the pid 0, then this isn't an error. Another time think about the "shared" environment: the client requests about his requirements but the stream is controlled by the server and it can be shared between different clients. However, this doesn't apply to syntax errors like: `addpids=0,1,0,10,0&delpids=0,1`. However, it's _recommended_ to try to accept these commands.
- Any PLAY command with tuning parameters are equal to a SETUP command. With the exception that the stream is identified. And in this case the SAT>IP server can or cannot serve the command with the same stream (so it can change the id, however it's common to not change it if it's not required).

As a summary: Remember that the stream can be shared, so don't do assumptions!

I hope this helps.

Barracuda09, 2019-12-22:
Hi @lars18th,

I have fixed this issue. Give it a try.

lars18th, 2020-04-14:
> Hi @lars18th,
> 
> I have fixed this issue. Give it a try.

Yes. Fixed, so I close this issue.

closed 2020-04-14 by lars18th