Barracuda09/SATPI #86: Unable to build on DietPi Buster (undefined reference to `__atomic_load_8')

psxde, 2019-12-14:
Hi,

I tried to build SATPI on a recently installed Raspbian OS but the build fails with:

```
g++ -c -I ./src -std=c++11 -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG s                                                                                                             rc/Version.cpp -o obj/Version.o
g++ obj/Version.o obj/InterfaceAttr.o obj/HttpServer.o obj/HttpcServer.o obj/Log                                                                                                             .o obj/Properties.o obj/RtspServer.o obj/Satpi.o obj/Stream.o obj/StreamClient.o                                                                                                              obj/StreamManager.o obj/StringConverter.o obj/base/M3UParser.o obj/base/Thread.                                                                                                             o obj/base/ThreadBase.o obj/base/TimeCounter.o obj/base/XMLSaveSupport.o obj/bas                                                                                                             e/XMLSupport.o obj/input/DeviceData.o obj/input/Transformation.o obj/input/dvb/F                                                                                                             rontend.o obj/input/dvb/FrontendData.o obj/input/dvb/delivery/DiSEqc.o obj/input                                                                                                             /dvb/delivery/DiSEqcEN50494.o obj/input/dvb/delivery/DiSEqcEN50607.o obj/input/d                                                                                                             vb/delivery/DiSEqcSwitch.o obj/input/dvb/delivery/DVBC.o obj/input/dvb/delivery/                                                                                                             DVBS.o obj/input/dvb/delivery/DVBT.o obj/input/dvb/delivery/Lnb.o obj/input/file                                                                                                             /TSReader.o obj/input/file/TSReaderData.o obj/input/childpipe/TSReader.o obj/inp                                                                                                             ut/childpipe/TSReaderData.o obj/input/stream/Streamer.o obj/input/stream/Streame                                                                                                             rData.o obj/mpegts/Filter.o obj/mpegts/PacketBuffer.o obj/mpegts/PAT.o obj/mpegt                                                                                                             s/PCR.o obj/mpegts/PidTable.o obj/mpegts/PMT.o obj/mpegts/SDT.o obj/mpegts/Table                                                                                                             Data.o obj/output/StreamThreadBase.o obj/output/StreamThreadHttp.o obj/output/St                                                                                                             reamThreadRtcpBase.o obj/output/StreamThreadRtcp.o obj/output/StreamThreadRtcpTc                                                                                                             p.o obj/output/StreamThreadRtp.o obj/output/StreamThreadRtpTcp.o obj/output/Stre                                                                                                             amThreadTSWriter.o obj/socket/HttpcSocket.o obj/socket/TcpSocket.o obj/socket/So                                                                                                             cketAttr.o obj/socket/UdpSocket.o obj/upnp/ssdp/Server.o -o satpi -pthread -lrt
/usr/bin/ld: obj/Stream.o: in function `Stream::getRtpPayload() const':
Stream.cpp:(.text+0x94): undefined reference to `__atomic_load_8'
/usr/bin/ld: obj/Stream.o: in function `Stream::addRtpData(unsigned int, long)':
Stream.cpp:(.text+0x100): undefined reference to `__atomic_load_8'
/usr/bin/ld: Stream.cpp:(.text+0x124): undefined reference to `__atomic_store_8'
/usr/bin/ld: obj/Stream.o: in function `Stream::addToXML(std::__cxx11::basic_str                                                                                                             ing<char, std::char_traits<char>, std::allocator<char> >&) const':
Stream.cpp:(.text+0x3414): undefined reference to `__atomic_load_8'
collect2: error: ld returned 1 exit status
make: *** [Makefile:173: satpi] Error 1
```

Thanks,
Patrick

Barracuda09, 2019-12-14:
Just pushed a commit that should fix this issue

Please check it out

psxde, 2019-12-14:
Thanks for the fast patch! The build process now run without any issue.

closed 2019-12-14 by psxde