Barracuda09/SATPI #87: Suggestion: Reload mapping file from UI

lars18th, 2020-03-27:
Hi @Barracuda09 ,

The current implementation of the SATPI works very well with streaming inputs and the frontend remapping. However, I want to do some tests and I feel it will be useful to have one option to reload the mapping file from the UI.

You agree with that?
Regards.

lars18th, 2020-04-14:
Hi @Barracuda09 ,

The idea is simple: a button for triggering a reload of the mapping file.

I hope you want to implement it.

Barracuda09, 2020-04-14:
Hi @lars18th 

If you press enter on the 'Transformation Mapping M3U' input field it should reload as well

lars18th, 2020-10-03:
Hi @Barracuda09 

> If you press enter on the 'Transformation Mapping M3U' input field it should reload as well

Thank you! Yes it works.
However, I don't have any feedback. Almost I suggest to print in the LOG that the `mapping.m3u` is reloaded. And if you want, put a "reload" button close to the input field replicating the < ENTER > press. This will be more intuitive.

Regards.