Barracuda09/SATPI #88: Sugestion: Home UI page

lars18th, 2020-03-27:
Hi @Barracuda09 ,

I feel it will be desisable to print additional info the HOME PAGE. For example the listening RTP PORT and the ADDRESS.

You agree?

lars18th, 2020-10-03:
Hi @Barracuda09 ,

I update this old request: Inside the "Server Info Page" I suggest to print the listening address and ports (RTSP & HTTP), plus if the SSDP is enabled or not. That's the only missing info that I see inside the UI.

Thank you!

Barracuda09, 2020-11-26:
Implemented by @issue  Many thanks for your support.

Will close this one.

closed 2020-11-26 by Barracuda09

lars18th, 2020-11-26:
> Implemented by @issue Many thanks for your support.

I hope will be soon a PR & merge of the contribution from @issue :wink:

Barracuda09, 2020-11-26:
@lars18th 

It is merged already, It requires to update the web directory.