Barracuda09/SATPI #89: Add FreeBSD support

decke, 2020-07-14:
I have tried to compile satpi 1.6.0 on FreeBSD 12.1/amd64 and found a few Linuxisms and compile issues on the way. In the end I could successfully build the binary with clang 8.0.1. Now before I spend more time on creating clean patches and a proper pull request I thought I'd ask if there is any interest in what I am doing and if it has a chance to get merged.

To avoid bias on FreeBSD support:
* FreeBSD also runs on RPI
* various DVB tuners connected via USB are supported on FreeBSD (via v4l / webcamd)

Barracuda09, 2020-07-14:
Hi Bernhard,

First of all thanks for your interest in SatPI.

I am always interested in enhancing SatPI and it's use on  a different OS.
And definitely in help with doing this in any shape or form.

I started this project, out of interest and thought it would be useful for
myself, which it turned out to be and use it daily. I do have less time to
do any useful programming and progress, because of some personal issues and
lack of time.

So please take the time to make a patch or pull request and I'm happy to
look at it and merge it. If you have any other suggestions or changes
please let me know or create a pull request as well.

Appreciate you help and interest and with kind regards,

Marc Postema

decke, 2020-11-21:
@Barracuda09:
Sorry, that it took me so long to come back working on this. I try to come up with some sane patches to add FreeBSD support and noticed that there is some Scheduler Affinity code in Thread.cpp and ThreadBase.cpp which is not portable and seems to be unused. Do you prefer to keep this 4 functions (Thread::getScheduledAffinity(), Thread::setAffinity(int UNUSED(cpu)), ThreadBase::getScheduledAffinity(), ThreadBase::setAffinity(int cpu)) or remove them alltogether?

Barracuda09, 2020-11-21:
@decke 

Thanks for your reaction.
Yes my preference is to keep the functions for Linux (for now), just make it not compile for FreeBSD.

Thanks for your time.

Reagards,

Marc

Barracuda09, 2021-02-22:
Done 91d5bd93d25a7bb9bd6ed1688a9a21799d082fa0

closed 2021-02-22 by Barracuda09