Barracuda09/SATPI #90: Unable to bind to secondary IPs of network interfaces

kripton, 2020-08-30:
Hi, this is probably related to #64 and the changes done via https://github.com/Barracuda09/SATPI/commit/df1bc1831a228600e367054b560ed31f69a6b414#diff-211f330b7372feef8c1d5938beb42e2eR91:

My situation is that I have only one network interface `br0` but that has several IP addresses. Right now, SatPI binds to the first IP address of that interface. Since the address that I would want it to bind to is in another subnet, I cannot access the web interface from my laptop, for example.

Could that code be modified to bind to a given IP instead? Or maybe easier: just bind to `0.0.0.0` if the `--iface-name` parameter has not been specified?

Barracuda09, 2020-08-30:
Hello @kripton 

Yes that code could be changes to bind to a specified IP (maybe with --bind-to-ip argument)

SaulGoodman1337, 2022-12-28:
any news here?
i need this too :)

Barracuda09, 2024-04-08:
Fixed by f2de19b