Barracuda09/SATPI #91: SEGFAULT in mpegts::SDT::parse on Astra19.2E

kripton, 2020-08-30:
Hi, I got SEGFAULTS from time to time and I've attached `gdb` to look where it happens. Didn't yet dig deeper into it but wanted to share anyways. Maybe someone who is deep into the code can provide a fix faster than me.

Environment: 2x TechniSat SkyStar USB HD connected to the same Unicable-LNB pointing at Astra 19.2E. TvHeadend using SATPI as SatIP server and scanning MUXes. It's pretty rainy today and reception is not perfect, so I cannot rule out corrupted data. Are there any CRC checks to make sure the data processed is sane?

```[                          src/HttpcServer.cpp:153] RTSP Stream data from client None with IP 192.168.189.249 on Port 40924: SETUP rtsp://192.168.189.249/stream=0?src=1&fe=1&freq=11875.5&sr=27500&msys=dvbs2&mtype=8psk&pol=h&fec=34&ro=0.35&pids=0 RTSP/1.0
Transport: RTP/AVP;unicast;client_port=50698-50699
Session: 0275564858
CSeq: 91
--- LINE END ---
[                        src/StreamManager.cpp:187] Found StreamID 0 - SessionID 0275564858
[                               src/Stream.cpp:234] Stream: 0, StreamClient[0] with SessionID 0275564858 for dvbs2
[                   src/input/dvb/Frontend.cpp:395] Stream: 0, Parsing transport parameters...
[               src/input/dvb/FrontendData.cpp:120] Stream: 0, New frequency requested, clearing old channel data...
[                   src/input/dvb/Frontend.cpp:402] Stream: 0, Parsing transport parameters (Finished)
[          src/output/StreamThreadRtcpBase.cpp:075] Stream: 0, Pause RTCP/UDP stream to 192.168.189.249:49219
[              src/output/StreamThreadBase.cpp:151] Stream: 0, Pause RTP/UDP stream to 192.168.189.249:49218 (Streamed 53.571 MBytes)
[                   src/input/dvb/Frontend.cpp:406] Stream: 0, Updating frontend...
[                   src/input/dvb/Frontend.cpp:640] Stream: 0, Closing /dev/dvb/adapter0/frontend0 fd: 11
[                   src/input/dvb/Frontend.cpp:678] Stream: 0, Opened /dev/dvb/adapter0/frontend0 fd: 11
[              src/input/dvb/delivery/DVBS.cpp:154] Stream: 0, Start tuning process for DVB-S(2)...
[     src/input/dvb/delivery/DiSEqcEN50607.cpp:126] Stream: 0, Sending DiSEqC: [70] [44] [97] [03] - DiSEqC Src: 0 - UB: 8
[     src/input/dvb/delivery/DiSEqcEN50607.cpp:126] Stream: 0, Sending DiSEqC: [70] [44] [97] [03] - DiSEqC Src: 0 - UB: 8
[     src/input/dvb/delivery/DiSEqcEN50607.cpp:126] Stream: 0, Sending DiSEqC: [70] [44] [97] [03] - DiSEqC Src: 0 - UB: 8
[     src/input/dvb/delivery/DiSEqcEN50607.cpp:126] Stream: 0, Sending DiSEqC: [70] [44] [97] [03] - DiSEqC Src: 0 - UB: 8
[     src/input/dvb/delivery/DiSEqcEN50607.cpp:126] Stream: 0, Sending DiSEqC: [70] [44] [97] [03] - DiSEqC Src: 0 - UB: 8
[     src/input/dvb/delivery/DiSEqcEN50607.cpp:126] Stream: 0, Sending DiSEqC: [70] [44] [97] [03] - DiSEqC Src: 0 - UB: 8
[     src/input/dvb/delivery/DiSEqcEN50607.cpp:126] Stream: 0, Sending DiSEqC: [70] [44] [97] [03] - DiSEqC Src: 0 - UB: 8
[     src/input/dvb/delivery/DiSEqcEN50607.cpp:126] Stream: 0, Sending DiSEqC: [70] [44] [97] [03] - DiSEqC Src: 0 - UB: 8
[     src/input/dvb/delivery/DiSEqcEN50607.cpp:126] Stream: 0, Sending DiSEqC: [70] [44] [97] [03] - DiSEqC Src: 0 - UB: 8
[              src/input/dvb/delivery/DVBS.cpp:181] Stream: 0, Set Properties: Frequency 1260000
[                   src/input/dvb/Frontend.cpp:689] Stream: 0, Waiting on lock...
[                   src/input/dvb/Frontend.cpp:700] Stream: 0, Tuned and locked (FE status 0x1F)
[                   src/input/dvb/Frontend.cpp:784] Stream: 0, Updating PID filters...
[                   src/input/dvb/Frontend.cpp:735] Stream: 0, Set DMX buffer size to 18874368 Bytes
[                   src/input/dvb/Frontend.cpp:749] Stream: 0, Opened /dev/dvb/adapter0/demux0 fd: 15
[                   src/input/dvb/Frontend.cpp:755] Stream: 0, Set filter PID: 0000
[                   src/input/dvb/Frontend.cpp:755] Stream: 0, Set filter PID: 0001
[                   src/input/dvb/Frontend.cpp:755] Stream: 0, Set filter PID: 0016
[                   src/input/dvb/Frontend.cpp:755] Stream: 0, Set filter PID: 0017
[                   src/input/dvb/Frontend.cpp:755] Stream: 0, Set filter PID: 0018
[                   src/input/dvb/Frontend.cpp:434] Stream: 0, Updating frontend (Finished)
[          src/output/StreamThreadRtcpBase.cpp:086] Stream: 0, Restart RTCP/UDP stream to 192.168.189.249:50699
[              src/output/StreamThreadBase.cpp:173] Stream: 0, Restart RTP/UDP stream to 192.168.189.249:50698
[                          src/HttpcServer.cpp:210] RTSP/1.0 200 OK
CSeq: 91
Session: 0275564858;timeout=60
Transport: RTP/AVP;unicast;client_ip=192.168.189.249;client_port=50698-50699
com.ses.streamID: 0
--- LINE END ---
[                          src/HttpcServer.cpp:153] RTSP Stream data from client None with IP 192.168.189.249 on Port 40926: SETUP rtsp://192.168.189.249/stream=1?src=1&fe=2&freq=12051&sr=27500&msys=dvbs&mtype=qpsk&pol=v&fec=34&ro=0.35&pids=0 RTSP/1.0
Transport: RTP/AVP;unicast;client_port=41152-41153
Session: 0263777460
CSeq: 92
--- LINE END ---
[                        src/StreamManager.cpp:187] Found StreamID 1 - SessionID 0263777460
[                               src/Stream.cpp:234] Stream: 1, StreamClient[0] with SessionID 0263777460 for dvbs
[                   src/input/dvb/Frontend.cpp:395] Stream: 1, Parsing transport parameters...
[               src/input/dvb/FrontendData.cpp:120] Stream: 1, New frequency requested, clearing old channel data...
[                   src/input/dvb/Frontend.cpp:402] Stream: 1, Parsing transport parameters (Finished)
[          src/output/StreamThreadRtcpBase.cpp:075] Stream: 1, Pause RTCP/UDP stream to 192.168.189.249:56065
[              src/output/StreamThreadBase.cpp:151] Stream: 1, Pause RTP/UDP stream to 192.168.189.249:56064 (Streamed 225.079 MBytes)
[                   src/input/dvb/Frontend.cpp:406] Stream: 1, Updating frontend...
[                   src/input/dvb/Frontend.cpp:640] Stream: 1, Closing /dev/dvb/adapter1/frontend0 fd: 12
[                   src/input/dvb/Frontend.cpp:678] Stream: 1, Opened /dev/dvb/adapter1/frontend0 fd: 12
[              src/input/dvb/delivery/DVBS.cpp:154] Stream: 1, Start tuning process for DVB-S(2)...
[                     src/mpegts/TableData.cpp:180] Stream: 0, PAT - PID 0000: sectionLength: 13  tableDataSize: 188  secNr: 0  lastSecNr: 0  currSecNr: 0
[                           src/mpegts/PAT.cpp:056] Stream: 0, PAT: Section Length: 13  TID: 14  Version: 229  secNr: 0 lastSecNr: 0  CRC: 0xD4B97175
[                           src/mpegts/PAT.cpp:069] Stream: 0, PAT: Prog NR: 0x0000 - 00000  NIT PID: 0016
[     src/input/dvb/delivery/DiSEqcEN50607.cpp:126] Stream: 1, Sending DiSEqC: [70] [4d] [47] [01] - DiSEqC Src: 0 - UB: 9
[     src/input/dvb/delivery/DiSEqcEN50607.cpp:126] Stream: 1, Sending DiSEqC: [70] [4d] [47] [01] - DiSEqC Src: 0 - UB: 9
[     src/input/dvb/delivery/DiSEqcEN50607.cpp:126] Stream: 1, Sending DiSEqC: [70] [4d] [47] [01] - DiSEqC Src: 0 - UB: 9
[     src/input/dvb/delivery/DiSEqcEN50607.cpp:126] Stream: 1, Sending DiSEqC: [70] [4d] [47] [01] - DiSEqC Src: 0 - UB: 9
[     src/input/dvb/delivery/DiSEqcEN50607.cpp:126] Stream: 1, Sending DiSEqC: [70] [4d] [47] [01] - DiSEqC Src: 0 - UB: 9
[     src/input/dvb/delivery/DiSEqcEN50607.cpp:126] Stream: 1, Sending DiSEqC: [70] [4d] [47] [01] - DiSEqC Src: 0 - UB: 9
[                     src/mpegts/TableData.cpp:180] Stream: 0, SDT - PID 0017: sectionLength: 12  tableDataSize: 188  secNr: 0  lastSecNr: 0  currSecNr: 0
[                           src/mpegts/SDT.cpp:060] Stream: 0, SDT - Section Length: 12  Transport Stream ID: 14  Version: 193  secNr: 0  lastSecNr: 0  NetworkID: 0133  CRC: 0x58852CD4
--Type <RET> for more, q to quit, c to continue without paging--

Thread 578 "Streaming0" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff48af700 (LWP 9988)]
0x00005555555ce5ed in mpegts::SDT::parse (this=0x7fffd805eac0, streamID=0) at src/mpegts/SDT.cpp:70
70                                              const int eit       =  ptr[i + 2u];
(gdb) bt
#0  0x00005555555ce5ed in mpegts::SDT::parse (this=0x7fffd805eac0, streamID=0) at src/mpegts/SDT.cpp:70
#1  0x00005555555c36c1 in mpegts::Filter::addData (this=0x7ffff79c5268, streamID=0, buffer=...) at src/mpegts/Filter.cpp:108
#2  0x0000555555590c7a in input::DeviceData::addFilterData (this=0x7ffff79c51d8, streamID=0, buffer=...) at src/input/DeviceData.cpp:113
[... cut ...]
#5  0x00005555555d4ffe in output::StreamThreadBase::threadEntry (this=0x7fffe807d3a0) at src/output/StreamThreadBase.cpp:84
#6  0x000055555558ec11 in base::ThreadBase::threadEntryBase (this=0x7fffe807d3a0) at src/base/ThreadBase.cpp:148
#7  0x000055555558ed4e in base::ThreadBase::threadEntryFunc (arg=0x7fffe807d3a0) at ./src/base/ThreadBase.h:115
#8  0x00007ffff7bcdf40 in start_thread (arg=<optimized out>) at pthread_create.c:479
#9  0x00007ffff7b05c9f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
(gdb)```

Barracuda09, 2020-08-30:
Hi @kripton 

Yes there is an crc check.

If it is reproducible could you un-comment line 58 in mpegts::SDT::parse to trace the SDT data.

kripton, 2020-08-30:
Done @Barracuda09:
```
--- LINE END ---
[                          src/HttpcServer.cpp:153] RTSP Stream data from client None with IP 192.168.189.249 on Port 41908: PLAY rtsp://192.168.189.249/stream=0?delpids=222,223 RTSP/1.0
Session: 0276722210
CSeq: 10
--- LINE END ---
[                        src/StreamManager.cpp:187] Found StreamID 0 - SessionID 0276722210
[                               src/Stream.cpp:237] Stream: 0, StreamClient[0] with SessionID 0276722210
[                   src/input/dvb/Frontend.cpp:395] Stream: 0, Parsing transport parameters...
[                   src/input/dvb/Frontend.cpp:402] Stream: 0, Parsing transport parameters (Finished)
[                   src/input/dvb/Frontend.cpp:406] Stream: 0, Updating frontend...
[                   src/input/dvb/Frontend.cpp:784] Stream: 0, Updating PID filters...
[                   src/input/dvb/Frontend.cpp:768] Stream: 0, Remove filter PID: 0222 - Packet Count: 6 - PMT
[                   src/input/dvb/Frontend.cpp:768] Stream: 0, Remove filter PID: 0223 - Packet Count: 6 - PMT
[                   src/input/dvb/Frontend.cpp:434] Stream: 0, Updating frontend (Finished)
[                          src/HttpcServer.cpp:210] RTSP/1.0 200 OK
RTP-Info: url=rtsp://192.168.189.249/stream=0
CSeq: 10
Session: 0276722210
Range: npt=0.000-
--- LINE END ---
[                     src/mpegts/TableData.cpp:180] Stream: 1, PAT - PID 0000: sectionLength: 13  tableDataSize: 188  secNr: 0  lastSecNr: 0  currSecNr: 0
[                           src/mpegts/PAT.cpp:056] Stream: 1, PAT: Section Length: 13  TID: 14  Version: 229  secNr: 0 lastSecNr: 0  CRC: 0xD4B97175
[                           src/mpegts/PAT.cpp:069] Stream: 1, PAT: Prog NR: 0x0000 - 00000  NIT PID: 0016
[                     src/mpegts/TableData.cpp:180] Stream: 1, SDT - PID 0017: sectionLength: 12  tableDataSize: 188  secNr: 0  lastSecNr: 0  currSecNr: 0
[                           src/mpegts/SDT.cpp:058] Stream: 1, SDT data
47 40 11 1E 00 42 F0 0C 00 0E C1 00 00 00 85 FF  G@...B..........
58 85 2C D4 FF FF FF FF FF FF FF FF FF FF FF FF  X.,.............
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
FF FF FF FF FF FF FF FF FF FF FF FF              ............
--- LINE END ---
END
[                           src/mpegts/SDT.cpp:060] Stream: 1, SDT - Section Length: 12  Transport Stream ID: 14  Version: 193  secNr: 0  lastSecNr: 0  NetworkID: 0133  CRC: 0x58852CD4

Thread 8 "Streaming1" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff48af700 (LWP 24076)]
0x00005555555ce61a in mpegts::SDT::parse (this=0x7fffe8005060, streamID=1) at src/mpegts/SDT.cpp:69
69                                              const int serviceID = (ptr[i + 0u] << 8u) | ptr[i + 1u];
(gdb) bt
#0  0x00005555555ce61a in mpegts::SDT::parse (this=0x7fffe8005060, streamID=1) at src/mpegts/SDT.cpp:69
#1  0x00005555555c36c1 in mpegts::Filter::addData (this=0x7ffff7984268, streamID=1, buffer=...) at src/mpegts/Filter.cpp:108
#2  0x0000555555590c7a in input::DeviceData::addFilterData (this=0x7ffff79841d8, streamID=1, buffer=...) at src/input/DeviceData.cpp:113
#3  0x000055555559809a in input::dvb::Frontend::readFullTSPacket (this=0x7ffff7984020, buffer=...) at src/input/dvb/Frontend.cpp:260
#4  0x00005555555d59a1 in output::StreamThreadBase::readDataFromInputDevice (this=0x7ffff50b1010, client=...) at src/output/StreamThreadBase.cpp:188
#5  0x00005555555d504e in output::StreamThreadBase::threadEntry (this=0x7ffff50b1010) at src/output/StreamThreadBase.cpp:84
#6  0x000055555558ec11 in base::ThreadBase::threadEntryBase (this=0x7ffff50b1010) at src/base/ThreadBase.cpp:148
#7  0x000055555558ed4e in base::ThreadBase::threadEntryFunc (arg=0x7ffff50b1010) at ./src/base/ThreadBase.h:115
#8  0x00007ffff7bcdf40 in start_thread (arg=<optimized out>) at pthread_create.c:479
#9  0x00007ffff7b05c9f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
(gdb)
```

kripton, 2020-08-30:
Oh, and yes, this is already a debug-build. Did you expect more debugging output? And just for the reference: Built from git commit 62f7f7e45761b9f65ac97e567299f994abb6a6fe

Barracuda09, 2020-08-30:
@kripton   It seems to be an empty SDT. I seem not to check this at all..

Barracuda09, 2020-08-30:
@kripton  No you are correct it was debug... Sorry

Barracuda09, 2020-08-30:
@kripton  could you add after line 64 in in mpegts::SDT::parse this:

`		
				if (len < 16) {
					continue;
				}
`

Barracuda09, 2020-08-30:
@kripton 

Well that previous  fix will not work.. I need to rethink this, I did not take into account that the the SDT could be 'empty'.

Barracuda09, 2020-08-31:
Hi @kripton 

I did rethink and I think I have the solution this should change
`				// 4 = CRC   9 = SDT Header from section length`
`				const std::size_t len = tableData.sectionLength - 4u - 9u;`

To:

`				// 4 = CRC   8 = SDT Header from section length`
`				const std::size_t len = tableData.sectionLength - 4u - 8u;`

The header is not 9 but 8 bytes

Could you try this?

kripton, 2020-08-31:
That in combination with the condition to just skip the SDT solves it! Great, thank you!

Barracuda09, 2020-08-31:
Thanks @kripton  for your report, it should be fixed now with latest commit.

kripton, 2020-09-01:
I can confirm that the fix is working. I've rescanned all Muxes and SATPI is not crashing anymore. Thanks for fixing so fast :+1:

commit 52eaeb93 (2020-08-31): Fix #91 - Segmentation fault in parsing SDT
closed 2020-09-01 by kripton