Barracuda09/SATPI #92: Vu+ Uno 4k > only frontend0 to frontend3 are working

neo7530, 2020-09-07:
Tested satpi today and got it only working for frontend0 to frontend3. Would like to use frontend6 and 7 for sat>IP. The webgui shows, that it is in use, but there is no video/audio.

Barracuda09, 2020-09-07:
Hi @neo7530 

First of all thanks for testing, and time to make an issue here.

I expect you build the current code from master?
How did you build it?

What is your setup:
- DVB-S2 or DVB-C?
- Enigma 2 still running?

neo7530, 2020-09-08:
Oh sorry, forgot to post my settings. Yea. Dvb-c and enigma is running. Even without enigma running only the first 4 tuners are working.

Barracuda09, 2020-09-08:
Hi @neo7530 

Thanks for the additional info. I have to take a look what is going on.

Did you build with something like this?: `make debug ENIGMA=yes`
It should not be required anymore, but you could try it anyway?

neo7530, 2020-09-08:
Have rebuild with debug ENIGMA=yes. will test, when i'm @home.

root@raspberrypi:~# g++ --version
g++ (Raspbian 8.3.0-6+rpi1) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Barracuda09, 2020-09-08:
@neo7530 

I use for Vu+ Uno 4K (DVB-S2 and C) this compiler: https://github.com/Broadcom/stbgcc-6.3/releases

neo7530, 2020-09-08:
What do you Change for compiling?

Barracuda09, 2020-09-08:
@neo7530 

From the Wiki:

- If you like to run it on an Enigma2 box **_(With the correct toolchain)_**, use:

    `source env.sh` _(Only once, to enable build environment for Broadcom cross-compiler)_<br/>
    `make debug ENIGMA=yes`<br/>

neo7530, 2020-09-08:
works now… so i have to test, if tuner 4..7 are working now with this build…

Barracuda09, 2020-09-08:
@neo7530  Yes please test this.

By the way which image are you using?

neo7530, 2020-09-08:
Vti 14.x

Barracuda09, 2020-09-08:
@neo7530 

Could you try current code with fix: 6d294a28eb94c025f1f46216fb370b5f501b4ab0

neo7530, 2020-09-08:
Hi. Have tried. Sadly the sat>ip client on my 2nd box get any video. Vlc works okay. Will try dvbviewer tomorrow with this fix.

neo7530, 2020-09-09:
Tested any further... Dvbviewer works with frontend6 and 7 now. But sat>ip client on my second vu+ won't show any video nor audio. What can I do now to get this working?

Barracuda09, 2020-09-09:
@neo7530 

Could you provide a more complete log, with start of request and lock etc.

neo7530, 2020-09-09:
Yep, will provide tomorrow. Have tried openatv image with sat-ip client. Same as with vti image.

Barracuda09, 2020-09-09:
Did you use stock image. I use the original image

neo7530, 2020-09-09:
The sat ip client works if I use the satpi server on my Ubuntu computer. I think it has to do with buffering or so. No clue. Is it possible to stream via tcp instead via udp?

Barracuda09, 2020-09-09:
Yes it has to do with drivers. Slow PID(scanning channels). I now use the stock image it works best for me.

neo7530, 2020-09-09:
Okay. Will try Stock image tomorrow...

Barracuda09, 2020-09-09:
Yes you can use http like:
http://192.168.0.112:8875/?fe=4&freq=418&sr=6900&msys=dvbc&mtype=256qam&fec=35&specinv=0&pids=0,1,16,17,18,103,2601,2611,2700

neo7530, 2020-09-10:
Here is a longer LOG from non working Stream. Have tried Stock Firmware. With this i can get A/V after a reboot. But it dissapears after changing the channel.

5  Thu Sep 10 07:48:22.7792 2020  7  RTSP Stream data from client None with IP 10.24.10.11 on Port 44739: PLAY rtsp://10.24.10.10:554/stream=5?addpids=220 RTSP/1.0
6  Thu Sep 10 07:48:22.7792 2020  7  CSeq: 6
7  Thu Sep 10 07:48:22.7792 2020  7  Session: 0273321084
8  Thu Sep 10 07:48:22.7792 2020  7  --- LINE END ---
9  Thu Sep 10 07:48:22.7794 2020  6  Found StreamID 5 - SessionID 0273321084
10  Thu Sep 10 07:48:22.7795 2020  6  Stream: 5, StreamClient[0] with SessionID 0273321084
11  Thu Sep 10 07:48:22.7795 2020  6  Stream: 5, Parsing transport parameters...
12  Thu Sep 10 07:48:22.7797 2020  7  Stream: 5, Parsing transport parameters (Finished)
13  Thu Sep 10 07:48:22.7798 2020  6  Stream: 5, Updating frontend...
14  Thu Sep 10 07:48:22.7798 2020  6  Stream: 5, Updating PID filters...
15  Thu Sep 10 07:48:22.7801 2020  7  Stream: 5, Set filter PID: 0220
16  Thu Sep 10 07:48:22.7826 2020  7  Stream: 5, Updating frontend (Finished)
17  Thu Sep 10 07:48:22.7827 2020  7  RTSP/1.0 200 OK
18  Thu Sep 10 07:48:22.7827 2020  7  RTP-Info: url=rtsp://10.24.10.10/stream=5
19  Thu Sep 10 07:48:22.7827 2020  7  CSeq: 6
20  Thu Sep 10 07:48:22.7827 2020  7  Session: 0273321084
21  Thu Sep 10 07:48:22.7827 2020  7  Range: npt=0.000-
22  Thu Sep 10 07:48:22.7827 2020  7  --- LINE END ---
23  Thu Sep 10 07:48:22.7852 2020  7  RTSP Stream data from client None with IP 10.24.10.11 on Port 44739: PLAY rtsp://10.24.10.10:554/stream=5?addpids=230,210 RTSP/1.0
24  Thu Sep 10 07:48:22.7852 2020  7  CSeq: 7
25  Thu Sep 10 07:48:22.7852 2020  7  Session: 0273321084
26  Thu Sep 10 07:48:22.7852 2020  7  --- LINE END ---
27  Thu Sep 10 07:48:22.7854 2020  6  Found StreamID 5 - SessionID 0273321084
28  Thu Sep 10 07:48:22.7854 2020  6  Stream: 5, StreamClient[0] with SessionID 0273321084
29  Thu Sep 10 07:48:22.7855 2020  6  Stream: 5, Parsing transport parameters...
30  Thu Sep 10 07:48:22.7856 2020  7  Stream: 5, Parsing transport parameters (Finished)
31  Thu Sep 10 07:48:22.7857 2020  6  Stream: 5, Updating frontend...
32  Thu Sep 10 07:48:22.7858 2020  6  Stream: 5, Updating PID filters...
33  Thu Sep 10 07:48:22.7861 2020  7  Stream: 5, Set filter PID: 0210
34  Thu Sep 10 07:48:22.7863 2020  7  Stream: 5, Set filter PID: 0230
35  Thu Sep 10 07:48:22.7884 2020  7  Stream: 5, Updating frontend (Finished)
36  Thu Sep 10 07:48:22.7885 2020  7  RTSP/1.0 200 OK
37  Thu Sep 10 07:48:22.7885 2020  7  RTP-Info: url=rtsp://10.24.10.10/stream=5
38  Thu Sep 10 07:48:22.7885 2020  7  CSeq: 7
39  Thu Sep 10 07:48:22.7885 2020  7  Session: 0273321084
40  Thu Sep 10 07:48:22.7885 2020  7  Range: npt=0.000-
41  Thu Sep 10 07:48:22.7885 2020  7  --- LINE END ---
42  Thu Sep 10 07:48:22.8680 2020  7  RTSP Stream data from client None with IP 10.24.10.11 on Port 44739: PLAY rtsp://10.24.10.10:554/stream=5?delpids=0 RTSP/1.0
43  Thu Sep 10 07:48:22.8680 2020  7  CSeq: 8
44  Thu Sep 10 07:48:22.8680 2020  7  Session: 0273321084
45  Thu Sep 10 07:48:22.8680 2020  7  --- LINE END ---
46  Thu Sep 10 07:48:22.8683 2020  6  Found StreamID 5 - SessionID 0273321084
47  Thu Sep 10 07:48:22.8684 2020  6  Stream: 5, StreamClient[0] with SessionID 0273321084
48  Thu Sep 10 07:48:22.8684 2020  6  Stream: 5, Parsing transport parameters...
49  Thu Sep 10 07:48:22.8686 2020  7  Stream: 5, Parsing transport parameters (Finished)
50  Thu Sep 10 07:48:22.8686 2020  6  Stream: 5, Updating frontend...
51  Thu Sep 10 07:48:22.8687 2020  6  Stream: 5, Updating PID filters...
52  Thu Sep 10 07:48:22.8688 2020  7  Stream: 5, Remove filter PID: 0000 - Packet Count: 0
53  Thu Sep 10 07:48:22.8711 2020  7  Stream: 5, Updating frontend (Finished)
54  Thu Sep 10 07:48:22.8711 2020  7  RTSP/1.0 200 OK
55  Thu Sep 10 07:48:22.8711 2020  7  RTP-Info: url=rtsp://10.24.10.10/stream=5
56  Thu Sep 10 07:48:22.8711 2020  7  CSeq: 8
57  Thu Sep 10 07:48:22.8711 2020  7  Session: 0273321084
58  Thu Sep 10 07:48:22.8711 2020  7  Range: npt=0.000-
59  Thu Sep 10 07:48:22.8711 2020  7  --- LINE END ---
60  Thu Sep 10 07:48:22.8734 2020  7  RTSP Stream data from client None with IP 10.24.10.11 on Port 44739: PLAY rtsp://10.24.10.10:554/stream=5?addpids=0 RTSP/1.0
61  Thu Sep 10 07:48:22.8734 2020  7  CSeq: 9
62  Thu Sep 10 07:48:22.8734 2020  7  Session: 0273321084
63  Thu Sep 10 07:48:22.8734 2020  7  --- LINE END ---
64  Thu Sep 10 07:48:22.8737 2020  6  Found StreamID 5 - SessionID 0273321084
65  Thu Sep 10 07:48:22.8737 2020  6  Stream: 5, StreamClient[0] with SessionID 0273321084
66  Thu Sep 10 07:48:22.8738 2020  6  Stream: 5, Parsing transport parameters...
67  Thu Sep 10 07:48:22.8739 2020  7  Stream: 5, Parsing transport parameters (Finished)
68  Thu Sep 10 07:48:22.8741 2020  6  Stream: 5, Updating frontend...
69  Thu Sep 10 07:48:22.8742 2020  6  Stream: 5, Updating PID filters...
70  Thu Sep 10 07:48:22.8748 2020  7  Stream: 5, Set filter PID: 0000
71  Thu Sep 10 07:48:22.8778 2020  7  Stream: 5, Updating frontend (Finished)
72  Thu Sep 10 07:48:22.8778 2020  7  RTSP/1.0 200 OK
[... 302 lines cut ...]
375  Thu Sep 10 07:49:07.8756 2020  7  Session: 0273321084
376  Thu Sep 10 07:49:07.8756 2020  7  Range: npt=0.000-
377  Thu Sep 10 07:49:07.8756 2020  7  --- LINE END ---
[... cut ...]
392  Thu Sep 10 07:49:07.8809 2020  7  Stream: 5, Remove filter PID: 5000 - Packet Count: 0
393  Thu Sep 10 07:49:07.8819 2020  7  Stream: 5, Updating frontend (Finished)
394  Thu Sep 10 07:49:07.8819 2020  7  RTSP/1.0 200 OK
395  Thu Sep 10 07:49:07.8819 2020  7  RTP-Info: url=rtsp://10.24.10.10/stream=5
396  Thu Sep 10 07:49:07.8819 2020  7  CSeq: 25
397  Thu Sep 10 07:49:07.8819 2020  7  Session: 0273321084
398  Thu Sep 10 07:49:07.8819 2020  7  Range: npt=0.000-
399  Thu Sep 10 07:49:07.8819 2020  7  --- LINE END ---

neo7530, 2020-09-10:
And some log after a reboot.

0  Thu Sep 10 07:59:12.3857 2020  6  Stream: 5, PMT - Stream Type: 11  ES PID: 2171  ES-Length: 014
1  Thu Sep 10 07:59:12.3858 2020  6  Stream: 5, PMT - Stream Type: 12  ES PID: 2176  ES-Length: 003
2  Thu Sep 10 07:59:12.3858 2020  6  Stream: 5, PMT - Stream Type: 27  ES PID: 5311  ES-Length: 003
3  Thu Sep 10 07:59:12.3859 2020  6  Stream: 5, PMT - Stream Type: 06  ES PID: 5312  ES-Length: 014
4  Thu Sep 10 07:59:12.3859 2020  6  Stream: 5, PMT - Stream Type: 06  ES PID: 5313  ES-Length: 021
5  Thu Sep 10 07:59:12.3859 2020  6  Stream: 5, PMT - Stream Type: 06  ES PID: 5314  ES-Length: 010
6  Thu Sep 10 07:59:12.3860 2020  6  Stream: 5, PMT - Stream Type: 06  ES PID: 5315  ES-Length: 013
7  Thu Sep 10 07:59:12.3860 2020  6  Stream: 5, PMT - PID 5310: sectionLength: 136  tableDataSize: 188  secNr: 0  lastSecNr: 0  currSecNr: 0
8  Thu Sep 10 07:59:12.3861 2020  6  Stream: 5, PMT - Section Length: 136  Prog NR: 10351  Version: 197  secNr: 0  lastSecNr: 0  PCR-PID: 5311  Program Length: 0  CRC: 0xD28382F1
9  Thu Sep 10 07:59:12.3861 2020  6  Stream: 5, PMT - Stream Type: 05  ES PID: 0677  ES-Length: 005
10  Thu Sep 10 07:59:12.3862 2020  6  Stream: 5, PMT - Stream Type: 11  ES PID: 2171  ES-Length: 014
11  Thu Sep 10 07:59:12.3862 2020  6  Stream: 5, PMT - Stream Type: 12  ES PID: 2176  ES-Length: 003
12  Thu Sep 10 07:59:12.3862 2020  6  Stream: 5, PMT - Stream Type: 27  ES PID: 5311  ES-Length: 003
13  Thu Sep 10 07:59:12.3863 2020  6  Stream: 5, PMT - Stream Type: 06  ES PID: 5312  ES-Length: 014
14  Thu Sep 10 07:59:12.3864 2020  6  Stream: 5, PMT - Stream Type: 06  ES PID: 5313  ES-Length: 021
15  Thu Sep 10 07:59:12.3864 2020  6  Stream: 5, PMT - Stream Type: 06  ES PID: 5314  ES-Length: 010
16  Thu Sep 10 07:59:12.3864 2020  6  Stream: 5, PMT - Stream Type: 06  ES PID: 5315  ES-Length: 013
17  Thu Sep 10 07:59:12.3865 2020  6  Stream: 5, PMT - PID 5310: sectionLength: 136  tableDataSize: 188  secNr: 0  lastSecNr: 0  currSecNr: 0
18  Thu Sep 10 07:59:12.3865 2020  6  Stream: 5, PMT - Section Length: 136  Prog NR: 10351  Version: 197  secNr: 0  lastSecNr: 0  PCR-PID: 5311  Program Length: 0  CRC: 0xD28382F1
19  Thu Sep 10 07:59:12.3866 2020  6  Stream: 5, PMT - Stream Type: 05  ES PID: 0677  ES-Length: 005
20  Thu Sep 10 07:59:12.3866 2020  6  Stream: 5, PMT - Stream Type: 11  ES PID: 2171  ES-Length: 014
21  Thu Sep 10 07:59:12.3867 2020  6  Stream: 5, PMT - Stream Type: 12  ES PID: 2176  ES-Length: 003
22  Thu Sep 10 07:59:12.3867 2020  6  Stream: 5, PMT - Stream Type: 27  ES PID: 5311  ES-Length: 003
23  Thu Sep 10 07:59:12.3868 2020  6  Stream: 5, PMT - Stream Type: 06  ES PID: 5312  ES-Length: 014
24  Thu Sep 10 07:59:12.3868 2020  6  Stream: 5, PMT - Stream Type: 06  ES PID: 5313  ES-Length: 021
25  Thu Sep 10 07:59:12.3869 2020  6  Stream: 5, PMT - Stream Type: 06  ES PID: 5314  ES-Length: 010
26  Thu Sep 10 07:59:12.3869 2020  6  Stream: 5, PMT - Stream Type: 06  ES PID: 5315  ES-Length: 013
27  Thu Sep 10 07:59:12.3870 2020  6  Stream: 5, PMT - PID 5310: sectionLength: 136  tableDataSize: 188  secNr: 0  lastSecNr: 0  currSecNr: 0
28  Thu Sep 10 07:59:12.3870 2020  6  Stream: 5, PMT - Section Length: 136  Prog NR: 10351  Version: 197  secNr: 0  lastSecNr: 0  PCR-PID: 5311  Program Length: 0  CRC: 0xD28382F1
29  Thu Sep 10 07:59:12.3871 2020  6  Stream: 5, PMT - Stream Type: 05  ES PID: 0677  ES-Length: 005
30  Thu Sep 10 07:59:12.3871 2020  6  Stream: 5, PMT - Stream Type: 11  ES PID: 2171  ES-Length: 014
31  Thu Sep 10 07:59:12.3872 2020  6  Stream: 5, PMT - Stream Type: 12  ES PID: 2176  ES-Length: 003
32  Thu Sep 10 07:59:12.3872 2020  6  Stream: 5, PMT - Stream Type: 27  ES PID: 5311  ES-Length: 003
33  Thu Sep 10 07:59:12.3872 2020  6  Stream: 5, PMT - Stream Type: 06  ES PID: 5312  ES-Length: 014
34  Thu Sep 10 07:59:12.3873 2020  6  Stream: 5, PMT - Stream Type: 06  ES PID: 5313  ES-Length: 021
35  Thu Sep 10 07:59:12.3873 2020  6  Stream: 5, PMT - Stream Type: 06  ES PID: 5314  ES-Length: 010
36  Thu Sep 10 07:59:12.3874 2020  6  Stream: 5, PMT - Stream Type: 06  ES PID: 5315  ES-Length: 013
37  Thu Sep 10 07:59:12.3874 2020  6  Stream: 5, PMT - PID 5310: sectionLength: 136  tableDataSize: 188  secNr: 0  lastSecNr: 0  currSecNr: 0
38  Thu Sep 10 07:59:12.3875 2020  6  Stream: 5, PMT - Section Length: 136  Prog NR: 10351  Version: 197  secNr: 0  lastSecNr: 0  PCR-PID: 5311  Program Length: 0  CRC: 0xD28382F1
39  Thu Sep 10 07:59:12.3875 2020  6  Stream: 5, PMT - Stream Type: 05  ES PID: 0677  ES-Length: 005
40  Thu Sep 10 07:59:12.3876 2020  6  Stream: 5, PMT - Stream Type: 11  ES PID: 2171  ES-Length: 014
41  Thu Sep 10 07:59:12.3876 2020  6  Stream: 5, PMT - Stream Type: 12  ES PID: 2176  ES-Length: 003
42  Thu Sep 10 07:59:12.3876 2020  6  Stream: 5, PMT - Stream Type: 27  ES PID: 5311  ES-Length: 003
43  Thu Sep 10 07:59:12.3877 2020  6  Stream: 5, PMT - Stream Type: 06  ES PID: 5312  ES-Length: 014
44  Thu Sep 10 07:59:12.3877 2020  6  Stream: 5, PMT - Stream Type: 06  ES PID: 5313  ES-Length: 021
45  Thu Sep 10 07:59:12.3878 2020  6  Stream: 5, PMT - Stream Type: 06  ES PID: 5314  ES-Length: 010
46  Thu Sep 10 07:59:12.3878 2020  6  Stream: 5, PMT - Stream Type: 06  ES PID: 5315  ES-Length: 013
47  Thu Sep 10 07:59:12.3879 2020  6  Stream: 5, PMT - PID 5310: sectionLength: 136  tableDataSize: 188  secNr: 0  lastSecNr: 0  currSecNr: 0
[... cut ...]
392  Thu Sep 10 07:59:36.8218 2020  7  Stream: 5, Remove filter PID: 2171 - Packet Count: 1671
393  Thu Sep 10 07:59:36.8235 2020  7  Stream: 5, Updating frontend (Finished)
394  Thu Sep 10 07:59:36.8236 2020  7  RTSP/1.0 200 OK
395  Thu Sep 10 07:59:36.8236 2020  7  RTP-Info: url=rtsp://10.24.10.10/stream=5
396  Thu Sep 10 07:59:36.8236 2020  7  CSeq: 27
397  Thu Sep 10 07:59:36.8236 2020  7  Session: 0254007563
398  Thu Sep 10 07:59:36.8236 2020  7  Range: npt=0.000-
399  Thu Sep 10 07:59:36.8236 2020  7  --- LINE END ---

Barracuda09, 2020-09-10:
@neo7530 the complete log will also be in syslog. Could you attach it here as an file?

Thanks for your testing and time so far.

neo7530, 2020-09-10:
where is the syslog for satpi?

Barracuda09, 2020-09-10:
@neo7530 

Did you disable enigma from starting? (disable it in enigma.sh from starting for example)

It is doing a lot. Like tuning and locking to the signal and requesting PIDS but there are no packets received. I have seen this with 'older' drivers from Vu+ but the current version I use it seems to be working for me. I do not use enigma at all I use the Vu+ only headless. So only as SAT>IP server.

Maybe enigma is changing something that i am not aware of.

neo7530, 2020-09-10:
On the Server box? No, Enigma is running. I can try to disable Enigma from starting and see if anything changes.

Barracuda09, 2020-09-10:
Yes you could try that. Just put the line were enigma is starting in
comment in enigma.sh

neo7530, 2020-09-10:
Don't have an enigma.sh on my box....

Barracuda09, 2020-09-10:
Maybe it's called enigma2.sh.. I'm not really sure. But there is an script
somewhere

neo7530, 2020-09-10:
Yea, found it. but it didn't Change anything. The Problem is the SAT>IP Client, not the Server. This works like a charme. DVBViewer works without any problems.

Barracuda09, 2020-09-10:
Mmm... Ok.. it could be that the sat>up client has to short timeouts an
delete pids to early before any packets are received.

But after an reboot it works also with client?

neo7530, 2020-09-10:
Yep, it works until i Change the channel…

Barracuda09, 2020-09-10:
Could you make a syslog of this situation?

Barracuda09, 2022-09-23:
This has been resolved in the meantime.

closed 2022-09-23 by Barracuda09