Barracuda09/SATPI #93: SSDP announcement is empty

grasduinen, 2020-09-27:
The VLC player show the SATPI server, but not the complete SSDP announcement. The SSDP fields are empty.
If I open the SatPI.xml file,  is show that it is not well formed. Maybe the mash up of XML causes the incomplete SSDP announcement? 
[SATIP_SSDP.pdf](https://github.com/Barracuda09/SATPI/files/5288524/SATIP_SSDP.pdf)

Barracuda09, 2020-09-27:
Hi @grasduinen 

The SatPI.xml is not the description file, it is used for the Web Interface.

The file you are looking for is desc.xml:
http://ip.of.the.server:8875/desc.xml

Barracuda09, 2020-09-27:
Hi @grasduinen 

I don't think you will find more information as shown in the PDF

grasduinen, 2020-09-27:
> Hi @grasduinen
> 
> I don't think you will find more information as shown in the PDF

How can I refer to the location of channellist.m3u ? It now shows an default value : http://www.satip.info/Playlists/auto.m3u

Barracuda09, 2020-09-27:
Hi @grasduinen 

Well it seems there is something not going right with VLC and parsing the satip:X_SATIPM3U

grasduinen, 2020-09-27:
@Barracuda09 
I've replaced the %1 - %5 in desc.xml with fixed paths, now the port 8875 is used. In the m3u file the %1 is also not including the rtsp 8554 port.
Now VLC is reading the channellist;
[image]

Barracuda09, 2020-09-28:
@grasduinen 

Could you please explain a bit more what you changes?

Or attach the desc.xml you use now here?

The m3 I file will currently only replace %1 with IP address. This I can change but have to parse which protocol is used (http or rrsp) to add correct port number.

Thanks for you report an time

grasduinen, 2020-09-28:
@Barracuda09 
Testing your last code compiled on an Raspberry with DVB-T2 HAT ;
I've adjusted the following part in desc.xml;

<presentationURL>http://192.168.147.250**:8875**/index.html</presentationURL>

<satip:X_SATIPCAP xmlns:satip="urn:ses-com:satip">**DVBT2-1**</satip:X_SATIPCAP>

<satip:X_SATIPM3U xmlns:satip="urn:ses-com:satip">http://192.168.147.250**:8875**/channellist.m3u</satip:X_SATIPM3U>

My channellist.m3u ;

#EXTM3U

#EXTINF:0,1. NPO 1
rtsp://192.168.147.250**:8554**/?freq=482&bw=8&msys=dvbt2&tmode=8k&mtype=64qam&gi=14&fec=12&pids=0,1,16,17,18,7010,7011,7012,7013

#EXTINF:0,2. NPO 2
rtsp://192.168.147.250**:8554**/?freq=482&bw=8&msys=dvbt2&tmode=8k&mtype=64qam&gi=14&fec=12&pids=0,1,16,17,18,7020,7021,7022,7023

#EXTINF:0,3. NPO 3
rtsp://192.168.147.250**:8554**/?freq=482&bw=8&msys=dvbt2&tmode=8k&mtype=64qam&gi=14&fec=12&pids=0,1,16,17,18,7030,7031,7032,7033

Barracuda09, 2020-09-28:
@grasduinen 

I have tried with VLC 3.0.11 which here works with the 'default' desc.xml. But there was some weird in the in the UPnP preferences which I had to set to 'ServerList' and back to 'Auto' and now it finds it.

The M3U file I have to add the port number of http or rtsp. I have two options:

1.  rtsp://**%1**:**%2**/?....  %1 will be replaced by IP address and %2 will be replaced by rtsp port
    http://**%1**:**%3**/?....  %1 will be replaced by IP address and %3 will be replaced by http port
2. or make it more automatic by parsing and checking the protocol that is used etc.

grasduinen, 2020-09-28:
I'm on the same VLC version, changing to serverlist did the trick. Location is now pointing to the channellist (and channels expand)!
Option 2 sounds more solid, but both would be fine for me.

Do you have an suggestion on the Panasonic TV, why it's not finding the channels?

Barracuda09, 2020-09-28:
I do not own a Panasonic TV but here is an suggestion you can try  https://github.com/Barracuda09/SATPI/issues/62#issuecomment-470699255

9000h, 2020-10-26:
for unicast satip the Panasonic need to do a channel scan on for the delivery system, you cannot provide a channel list via the server and you cannot mix delivery systems.

lars18th, 2020-10-27:
Hi @9000h ,

> [...] you cannot mix delivery systems.

That's the reason to use the "channel-remaping" functionality. I'm using it to remap DVB-T muxes to some DVB-S frequencies.

I hope @Barracuda09 will fix soon the Child-PIPE bugs, as I want to remap some HLS inputs too!
Regards.

grasduinen, 2020-10-28:
> Hi @9000h ,
> 
> > [...] you cannot mix delivery systems.
> 
> That's the reason to use the "channel-remaping" functionality. I'm using it to remap DVB-T muxes to some DVB-S frequencies.
> 
> I hope @Barracuda09 will fix soon the Child-PIPE bugs, as I want to remap some HLS inputs too!
> Regards.

Could you  post an example of your HLS mapping setup? Thanks!

9000h, 2020-10-28:
Hi,
I do not use SATPI at all, sorry.
CU

Barracuda09, 2022-09-23:
This should already be implemented.

closed 2022-09-23 by Barracuda09