Barracuda09/SATPI #96: BUG: Blocking UI when using Child PIPE - TS Reader

lars18th, 2020-10-03:
Hi @Barracuda09 ,

When requesting a channel using a regular SAT>IP client, that is served with the "Child PIPE - TS Reader" front end, in this case the HTTP UI has missing data (empty table) when selecting "Frontend Overview" and "Frontend". And just when closing the session, this returns to work. This behaviour is not observed using "TS Reader" and/or "Streamer" inputs, so I feel is a blocking problem with this specific input.

Please, can you revise this?
Thank you!

lars18th, 2020-10-13:
Has anyone else experienced this?

Barracuda09, 2020-10-13:
@lars18th 

Yes I have seen this as well with testing this. I haven't figured out what it is yet.

Barracuda09, 2020-10-15:
Hi @lars18th 

When using the wget in ChildPIPE reader the SatPI.xml has become invalid. So it the contains illegal characters.

lars18th, 2020-10-18:
Hi @Barracuda09 ,

> When using the wget in ChildPIPE reader the SatPI.xml has become invalid. So it the contains illegal characters.

OK. Then if you know now where is the problem I hope you can fix it at some point.

Thank you for your hard work on this project! :+1: 
Regards.

Barracuda09, 2020-10-18:
Hi @lars18th 

Yes it is here:
https://github.com/Barracuda09/SATPI/blob/f33d58c3ab3f3fc708418047a2b7dae43af8013f/src/input/childpipe/TSReaderData.cpp#L43-L45

`pathname` has illegal characters. I only have to think about a proper fix because it can happen with any string in the xml file.

Barracuda09, 2020-10-27:
Hi @lars18th 

I have made a fix for this in branch V1.6.2, Please check it out

lars18th, 2020-11-02:
Hi @Barracuda09 ,

This seems to be fixed too! :+1: 

However, one additional suggestion: When using the ChildPIPE, TS Reader or Steramer _with_ the mappings.m3u (from my point of view the most natural mode to use them), it will be interesting to print the "Frequency" used in the UI. You agree?

Regards.

Barracuda09, 2020-11-09:
Reopen

lars18th, 2020-11-18:
Hi @Barracuda09 ,

In my opinion this issue can be closed. You agree?

commit 955669d8 (2020-10-27): Fix illegal characters in XML files to fix #96
closed 2020-11-18 by Barracuda09