Barracuda09/SATPI #97: Question: Has the file/pipe TS Reader a limited bandiwth?

lars18th, 2020-10-03:
Hi @Barracuda09 ,

I do this test: using an external tool I'm restreaming a .ts file to a multicast address. The file has a bitrate of around 24Mbps. Then when I configure the Streamer front end with this multicast address, and I receive the full-TS (pids=all) with a SAT>IP client all works as expected. However, when I use the `msys=file` _and_ with `msys=childpipe` too, the same TS is served, but the bitrate is reduced to ~10Mbps (and the result has glitches to replay).

So, the question is: You have limited the bandwith/speed of the "TS Reader" inputs?
If yes, then please, can you provide a method to increase the bandwith?

Regards.

lars18th, 2020-10-03:
Hi @Barracuda09 ,

I found a workaround, but I don't think it can be a definitive solution. So, almost I want to comment it here:

- I changed the value of the waiting time from `1000` to `150` on these two lines:
https://github.com/Barracuda09/SATPI/blob/f33d58c3ab3f3fc708418047a2b7dae43af8013f/src/input/file/TSReader.cpp#L107
https://github.com/Barracuda09/SATPI/blob/f33d58c3ab3f3fc708418047a2b7dae43af8013f/src/input/childpipe/TSReader.cpp#L107

So, almost in my case, this increases the bandwith. And we can be sure that the problem (the limit) exists.

Any idea to solve the problem?
Regards.

lars18th, 2020-11-26:
Hi @Barracuda09 ,

I suspect it will be necessary to use a configurable value for this timeout. So, please can you add it to the config? Thank you!

Barracuda09, 2020-11-26:
Yes I can add it, but it should normally be updated by PCR of the stream.

lars18th, 2020-11-26:
> Yes I can add it, but it should normally be updated by PCR of the stream.

Ok, I don't know this. Then, perhaps the most simple is to reduce even more the _default_ value. Please, try with 500ms (or 250ms tha is more aproximated to the typical interval between PCR marks).

Barracuda09, 2020-11-26:
@lars18th 

I already lowered this to 150ms in a previouse fix.

lars18th, 2020-11-26:
> I already lowered this to 150ms in a previouse fix.

OK. So I close this issue.

closed 2020-11-26 by lars18th