Barracuda09/SATPI #98: Cross mappings don't work

lars18th, 2020-10-06:
Hi @Barracuda09 ,

See this `mapping.m3u` example:
```
#EXTINF:-1 satip-freq="12470", SOURCE
rtsp://%1/?msys=childpipe&exec="wget%20-qO- ... '"
#EXTINF:-1 satip-freq="12722", ALIAS
rtsp://%1/?freq=12470&msys=dvbs2
```

It seems to not work. But, why not? The second remap links to the first.
I want to use multiple frequencies for the same target.

Barracuda09, 2020-10-27:
Hi @lars18th 

I don't think I have thought of this or understand correctly what the problem is.

lars18th, 2020-10-28:
Hi @Barracuda09 ,

> I don't think I have thought of this or understand correctly what the problem is.

Ok, then let to explain it more:

- The objective is to chain a mapping. So if you map A freq. to B, then you also map B to C. As a result, if you request A you obtain C.
- Why we need this? Think on this use case: you have a source injected using the childpipe input. And you wan to assign this source to a DVB-T mapping and to a DVB-S mapping. Then in this case, you have _two_ different frequencies (virtual mappings) that point to the same input. However, if you "copy&paste" the input, you will obtain two independent calls to the childpipe input. However, if you map the childpipe to one of the mappings, and the other mapping points to the first mapping, then you will point to the same source _only one time_!
- In my original example when a client requests the frequency "12470" this will be translated to the childpipe input. So, what I want is that any request to "12722" will be also translated to the childpipe... with an initial translation to "12470".

In resum: Please when computing a mapping translation, after exit, check if the target requires to be translated too.

I hope it will be more clear now.
Regards.

lars18th, 2021-03-05:
Hi @Barracuda09 ,

You feel this will easy to implement or it's stupid? My objective is to have the same childpipe exported over different sources. Imagine that you have one remap using childpipe (even valid for DVB-C/T too) that you have to include in more than one orbital positions. In this case instead of "copy&paste" the entire command, you can define it one time, and the "link it in the other frequencies".

Please, you feel is complex, that say it. Then will close this request. It can work with a copy&paste. Even it will be more clean (for the user) to support such linking mappings.

Regards.

Barracuda09, 2021-03-05:
Hi @lars18th 

Well to be fair, I haven't really thought about this yet.