Barracuda09/SATPI #100: Suggestion: Autorestart functionality

lars18th, 2020-11-03:
Hi @Barracuda09 ,

When using the v1.6.2 with the last fixes of the ChildPipe frontend, I see time to time a "Segmentation fault". At time, I can't see where the error is (however, I'll try to debug it, so don't care about this now). But I suggest to implement some auto-restarting functionality to overcome such unkown errors. In my environment the SATPI acts as another SAT>IP server, so it is used by another SAT>IP server running as the frontend. So for real SAT>IP clients is not a problem if the SATPI process restarts at some point (the user perception is a faulty signal, that will be restored after some seconds).

So, please, can you implement something like this?
https://stackoverflow.com/questions/3703227/self-restart-program-on-segfault-under-linux

I hope you agree to do it. And if you want to implement this, don't forget to print something about the running state before restarting. This will be very useful to discover the root cause of th error.

Thank you very much for this good project!
Regards.

Barracuda09, 2020-11-04:
Hi @lars18th 

Yes it could be possible to catch the SIGSEGV and 'restart' SatPI, so I see what I can do. But I rather know the cause of this "Segmentation fault"

lars18th, 2020-11-08:
Hi @Barracuda09 ,

> But I rather know the cause of this "Segmentation fault"

Me too! However, to know when it appears I need more info in the log about the cause. So, if you can implement some catch code then you can print something about the current status and restart the process. And this will be very useful.

Futhermore, I'm still waiting for the merge of the last improvements in the branch 1.6.2. Please, merge them and after we can try to investigate about this "segmentation fault".

Regards.

Barracuda09, 2020-11-27:
Hi @lars18th 

I have implemented some backtrace functionalaty it will create an file in `/tmp/satpi.bt` with addresses of the stack trace.
You can use `satpi --backtrace /tmp/satpi.bt` to recall the line numbers etc. SatPI will use addr2line to do this, so it should be available on the target system.

The restart is something i am hesitant to implement within SatPI because when a SEGV occured the memory is probably corrupt at one point. So it is better that the process stops so it can be cleaned.

Barracuda09, 2020-11-27:
Hi @lars18th 

One comment about backtace: It will only create a file in an DEBUG build