Barracuda09/SATPI #102: Suggestion: ChildPIPE "postcmd"

lars18th, 2020-11-12:
Hi @Barracuda09 ,

I want to execute _one_ command _after_ the ChildPIPE process is closed. So, I ask if you can add something like this: `?msys=childpipe&exec="/bin/my-binary"&postexec="/bin/cmd-close.sh"`.

As you can notice, it's easy to execute some command before (using a simple ` && ` or ` ; `), but impossible to execute anything after closing the child process. So, please consider this.

Futhermore, perhaps you want to add this `postexec` to the `streamer` too. In this case, then not forget to include some `preexec` to execute something before starting to read.

I feel than with this change it will possible to feed the SatPI with complex sources.
Thank you!

lars18th, 2020-11-16:
Hi @Barracuda09 ,

I rewrite the request to a more _clear_ and useful request:

- First, I want to describe the objective in my environment: I have one HDHomeRun tuner in my network. And I can drive this external tuner using commands with the stock `hdhomerun_config` tool. Using this tool I can tune, set filters and set the target (UDP or RTP unicast/multicast address); and obviously I can stop the streaming. Then I want to use SATPI with the CHILDPIPE input _and_ with the STREAMER inputs too. As you know, the MPEG2-TS can be recived using Files, Pipes and Sockets. And the current implementation of the SATPI supports the three inputs. However, the _current_ control is very limited. I explain why...
- The CHILDPIPE input is good for sources like one HTTP server. The reason is because the streaming starts with the connection request, and stops when the socket is closed. Then it can be combined without troubles with a subprocess connected using Pipes. Futhermore, this input can be run too with any external process that generates one TS stream.
- However, when using the STREAMER input, we don't have at time any method to _trigger_ the start or stop of the streaming. Even if this streaming mode is more efficient than the Pipe subprocess. Futhermore, with this input it can be also possible to adjust the pid filtering if the source supports it. So I suggest to add three new commands to the STREAMER input (and optionally to the CHILDPIPE input too).

The idea is this:
- `tune-cmd`: This will be called when the input is opened to enable the streaming.
- `filter-cmd`: This will be called every time the pid filter list is updated passing the list of pids as a parameter.
- `stop-cmd`: This will be called when the input is closed, or every time the input is destroyed (to overcome phantom errors).

So the idea is to support something like this:
`rtsp://%1/?msys=streamer&uri="udp@239.0.0.1:1234"&tune-cmd="/opt/script-cmd.sh"&filter-cmd="/opt/script-filter.sh"&stop-cmd="/opt/script-stop.sh"`

Then the start algorithm will be:
- When the input is selected (opened) do:
 - - Execute the `tune-cmd` and IF it not returns _Success_ (exit 0) or timeout pass then:
 - - - Kill the `tune-cmd` and close the input with an error.
 - - ELSE start the STREAMER input listening in the URI address.

And similar for the filter command and the stop command.

What you think about this idea?

lars18th, 2020-11-17:
Hi @Barracuda09 ,

Before you start to implement this (only if you want to do it), I'm experimenting with the `trap` command inside the shell scripts. So, perhaps it will be sufficient to make more advanced scripts to overcome the cleanup problem.

I'ill publish my results here when I am finished.
Regards.

lars18th, 2020-11-18:
Hi @Barracuda09 ,

Good news! I created a script file that does what is expected. So no changes in SATPI are required.
Perhaps you want to add this file to the repository as one CHILDPIPE input script example:

childpipe-example.sh
```
#!/bin/sh

trap cleanup TERM INT KILL EXIT HUP QUIT STOP ABRT;

## Set here your variables
CHANNEL=$1
## End Set
pid=0

preexec() {
  ## Execute here your initialization commands
  tool_config select $CHANNEL && \
  tool_config filter all
  ## End Execute
}

runcmd() {
  ## Execute here your your running command (with PIPEs) in background
  tool_config stream - &
  ## End Execute
  pid=$!
}

cleanup() {
  if [ -z $pid ] || [ "$pid" -eq 0 ] ; then
    exit 1
  else
    kill -9 $pid
    ## Execute here your cleanup commands
    tool_config stop
    ## End Execute
  fi
  exit 0
}

preexec
runcmd
if [ -z $pid ] || [ "$pid" -eq 0 ] ; then
  exit 1
else
  wait $pid
fi
```

As you can see, only lines between `##` require to be changed. The example `tool_config` is here to illustrate the use (in my environment I use `hdhomerun_config` with multiple commands).

I feel this script will be useful for a lot of uses!
Regards.

lars18th, 2021-01-11:
Hi @Barracuda09 ,

Still wating to include this in the documentation, or as an example. I suggest to update the `mapping.m3u` with one example using this script. You agree? You want a PR with one example?

Barracuda09, 2021-01-11:
Hi @lars18th 

I am not 100% sure how you would like to present this. So if you can, please make an PR and I look at it.

lars18th, 2021-03-12:
Hi @Barracuda09 ,

I close this issue as it's fixed.
Regards.

closed 2021-03-12 by lars18th