Barracuda09/SATPI #103: Suggestion: Virtual frontends in "Frontend Overview"

lars18th, 2020-11-18:
Hi @Barracuda09 ,

I commented something related to this here: https://github.com/Barracuda09/SATPI/issues/96#issuecomment-720631887

However, I want to suggest a more complete suggestion. This is the current page "Frontend Overview":
```
#   Name                    Owner       # Packets   Frequency (kHz)     Delivery System     Network Name    Channel Name
0   TS Reader               0.0.0.0     0                                                   Not Found       Not Found
1   Streamer                0.0.0.0     0                                                   Not Found       Not Found
3   Child PIPE - TS Reader  192.168.1.6 11111                                               NET-1           PROGRAM-1
```

And this is my suggestion:
```
#   Name                    Owner       # Packets   Frequency (kHz)     Delivery System     Network Name    Channel Name
0   TS Reader               0.0.0.0     0                                                   Not Found       Not Found
1   Streamer                0.0.0.0     0                                                   Not Found       Not Found
3   Child PIPE - TS Reader  192.168.1.6 11111       10744               wget -qO- http://   NET-1           PROGRAM-1
```

As you can see the change is quite simple: Instead of showing empty values in the columns "Frequency" and "Delivery System", you can add valuable information. The values come from:
- Frequency: This is the "mapping" value used. If no mapping is done, then you can write "_none_" indicating that the mapping is not in use (so it's a direct request).
- Delivery System: Here you can print the _same_ info that at time is printed in the section "Frontend/Frontend-3/Frontend Info/Path"... this info is very useful and basic. So I suggest to print it here.

Finally, I want to add the suggestion commented here: https://github.com/Barracuda09/SATPI/issues/101#issue-741535210
The concept is very simple: Print in the LOG a message when the _FIRST_ TS packet arrives. This will be very useful to debug the virtual frontends. I know that you print the number of packets in the web UI, but this line will be very useful to review the LOG file. So, please consider it.

I hope you want to add these small, but useful, changes.
Regards.

lars18th, 2020-11-18:
Hi @Barracuda09 ,

And regarding the physical tuners, I suggest to add the mapping too: instead of printing only the target frequency, you can print something like `10744 >> 458`. This will indicate that the target frequency is 458 (DVB-C) with a mapping from 10744 (DVB-S2).

You agree?

lars18th, 2020-11-18:
Hi @Barracuda09 ,

And inside the "Stream Info" section in the "Frontend Info" why not print a row with the `PID list` requested by the user?

Thank you for your support.
Regards.

lars18th, 2020-11-26:
Hi @issue ,

Please, can you implement this? I see that you implemented my request of #88, so I hope you want to do it.

Thank you all for this interesting SAT>IP server implementation with a lot of enhanced functionalities!

lars18th, 2020-11-26:
Hi @Barracuda09 ,

> The concept is very simple: Print in the LOG a message when the _FIRST_ TS packet arrives. This will be very useful to debug the virtual frontends. I know that you print the number of packets in the web UI, but this line will be very useful to review the LOG file. So, please consider it.

If you can, please consider to add this _simple_ enhancement now. It will be very useful to debug the use of the virtual inputs.
Thank you!

lars18th, 2020-12-14:
Hi @Barracuda09 ,

Thank you to incorporate part of my request in last revisions. The Path for the ChildPIPE frontend is great!
However, you will implement the *translated* frequency in the Frequency column, please?

Regards.

lars18th, 2021-01-11:
Hi @Barracuda09 ,

Thank you for last updates! :+1: 

However, before closing this issue, please can you comment if this request has sense, or it's complex to implement it?

> However, you will implement the translated frequency in the Frequency column, please?

Regards.

Barracuda09, 2021-01-11:
Hi @lars18th 

Thanks for you input. About this I have to look into this. The rest of the changes, I made were already in the 'SatPI.xml'. So very easy to implement with changes in the WebIF.

lars18th, 2021-02-22:
Hi @Barracuda09 ,

If you have sufficient time, please don't forget to include the "re-mapping frequency" in the table. This info is very useful when using PIPE-remappings. :wink: 

Thank you!

lars18th, 2021-03-05:
Hi @Barracuda09 ,

Thank you to implement it. It works like a charm. So I close the issue.

closed 2021-03-05 by lars18th