Barracuda09/SATPI #105: clang: multiple conversions from switch condition type 'std::atomic<State>' to an integral or enumeration type

decke, 2020-11-26:
With clang 10.0.1 I get the following compile error:

```
c++ -c -O2 -pipe -I/usr/local/include -fstack-protector-strong -fno-strict-aliasing   -I ./src -std=c++11 -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DHAS_NP_FUNCTIONS src/base/Thread.cpp -o obj/base/Thread.o
src/base/Thread.cpp:176:5: error: multiple conversions from switch condition type 'std::atomic<State>' to an integral or enumeration type
                                switch (_state) {
                                ^       ~~~~~~
/usr/include/c++/v1/atomic:1489:5: note: conversion to enumeration type 'base::Thread::State'
    operator _Tp() const volatile _NOEXCEPT {return load();}
    ^
/usr/include/c++/v1/atomic:1491:5: note: conversion to enumeration type 'base::Thread::State'
    operator _Tp() const _NOEXCEPT          {return load();}
    ^
```

Looking up the details of that issue I found the following stackoverflow question:
https://stackoverflow.com/questions/25143860/implicit-conversion-from-class-to-enumeration-type-in-switch-conditional

which indicates that it should work with `-std=c++1y` and I can confirm this worked for me.

Barracuda09, 2020-11-26:
Hi,

Does it work with Clang and std=c++11 like this:
`switch (_state.load()) {`

I prefere not to use beyond c++11 for now

decke, 2020-11-26:
Yes, that also seems to work fine. The problem exists two times in the code:

```
src/base/Thread.cpp:176
src/output/StreamThreadBase.cpp:75
```

Barracuda09, 2020-11-26:
Or use a different compile flags for clang?

decke, 2020-11-26:
Sorry, I don't understand your idea. Your proposed fix using `switch (_state.load()) {` works fine with clang and `std=c++11` so no need to change any compiler flags.

closed 2020-11-28 by decke