Barracuda09/SATPI #106: Tuners enumerated in the wrong order

fe31nz, 2020-12-05:
I am trying SATPI on a PC that has two 8 tuner cards (TBS6909 DVB-S/S2 and TBS6902 DVB-T/T2/C).  With 16 tuners (currently numbered 0-4,7-17 for historical reasons), it looks like SATPI is enumerating them in alphanumeric order, so they go 0,1,10,11,12,13,14,15,16,17,2,3,4.  This is terribly confusing when you are trying work out which tuner is which - I could only work it out by looking in the SATPI.xml file.  It would be much better if the tuners were enumerated in numeric order: 0,1,2,3,4,6,7,8,9,10,11,12,13,14,15,16,17.

It would also be really useful if the "Frontend Overview" web page listed the adapter number (/dev/dvb/frontend<x>).  That would help where the tuners are not consecutively numbered, as with my missing 5 and 6.

Barracuda09, 2020-12-05:
Hi @fe31nz 

Thanks for giving this a try.

Is it really enumerating wrong or showing wrong in web-if? So what is the path showing in web-if "Frontend xx" is that accordingly numbered? 

Regards,

Marc

fe31nz, 2020-12-05:
The web-if "Frontend 0" is /dev/dvb/frontend0, 1 is 1, 2 is 10, 3 is 11, 4 is 12, ... 8 is 16, 9 is 17, 10 is 2, 11 is 3, ..., 15 is 9.

Barracuda09, 2020-12-05:
Thanks @fe31nz 

https://github.com/Barracuda09/SATPI/blob/4249cdc1313c37a9a307e40ee2cf9446f005ef23/src/input/dvb/Frontend.cpp#L118-L120

you could try to change `alphasort `to `versionsort`

fe31nz, 2020-12-06:
Thank you, that fixes it.

Barracuda09, 2020-12-11:
Hi @fe31nz 

I implemented the suggested changes.

And @issue implemented this suggestion:
> It would also be really useful if the "Frontend Overview" web page listed the adapter number (/dev/dvb/frontend).

<br>
<br>
Thanks for giving this a try, and please verify and close this issue if you like.

closed 2021-01-11 by Barracuda09