Barracuda09/SATPI #107: Support this SAT>IP Server the SAT>IP norm EN 50585:2014?

420Dope, 2020-12-26:
Title says it all. I can't find any information if this SAT>IP Server supports the SAT>IP Norm EN 50585:2014?
So you can be safe if you buy a new tv or other devices which is certified as a SAT>IP client it works 100%.

Barracuda09, 2020-12-27:
Hi @420Dope 

To start, thanks for your interest in SatPI.

Well in the start it was only this norm https://www.satip.info/sites/satip/files/resource/satip_specification_version_1_2_2.pdf
But my guess is that it did not change by very much during this change that the protocol is standardized as CENELEC EN50585.

Last of all, I can not give you a 100% guarantee, I hope you can understand that. I will do my best to resolve any issues users report, or I myself will come across.

Regards,

Marc

420Dope, 2020-12-27:
Thank you very much for your answer.
This is good to know that you have followed the early standard protokoll. It also think that the protokoll is not changed during the standardization.

Its a cool project. Thank you for your hard work!

Best regards and stay safe.

Barracuda09, 2021-02-22:
Moved to discussions

closed 2021-02-22 by Barracuda09