Barracuda09/SATPI #109: Suggestion: Software frontends security

lars18th, 2021-01-11:
Hi @Barracuda09 ,

The current implementation of the SATPI project is very good. However, I have one suggestion to almost provide a basic protection regarding hacking users:

- With the current implementation, if the server has enabled the PIPE frontend, it's possible to execute _any_ shell command in the server. And this can be considered a big security hole. Futhermore, with FILE and TS READER frontends too some bad uses can be exploited. So I suggest a simple method to protect them.

The idea is quite simple: Provide a method to only enable such frontends using the remapping functionality. So, if the process receives an external SAT>IP request to these frontends, then the request is rejected. And only if the request is comming from the `mappings.m3u` then it will be processed.

What you think about this protection strategy?
Regards.

Barracuda09, 2021-01-11:
Hi @lars18th 

Yes I agree these are (big) security issue, if you use SatPI outside your own home. However the 'Child PIPE - TS Reader' frontend is only enabled with an command-line parameter. 

So what you mean is only use translation from freq to one of these frontends?

lars18th, 2021-01-11:
Hi @Barracuda09 ,

> So what you mean is only use translation from freq to one of these frontends?

Yes, but with a more generic/simple way. My suggestion is this:

- You don't need to change anything, except adding a new command line parameter called `--enable-unsecure-frontends`. When enabling this option the SATPI process uses the current behaviour: if enabled, the software frontends can be remotely used. But, when this option is not enabled (the new default behaviour), then _all_ SAT>IP requests that include these frontends (even when they are enabled) are rejected. The only exception is when the request is comming from a remapping (i.e. inside the `mapping.m3u` you include something like `#EXTINF:-1 satip-freq="200" ... rtsp://%1/?msys=file&uri=`

The code then only requires a simple check. And you only need to know if the URI is from a user (a direct SAT>IP request) or from a translation (after applying the `mapping.m3u`). Futhermore, note that this will be independent of enabling/disabling the ChildPIPE frontend. So it's sufficient secure as you block all remote calls, and only the user can configure his pre-configured sources.

I hope you want to implement this (simple), or any other perhaps more powerful control. In any case, I feel this will be enough for most users.
Regards.

lars18th, 2021-01-27:
Hi @Barracuda09 ,

What you think about my previous description? You think this will be easy to implement? What I want is to protect my SAT>IP server but continue using the good remmaping functionality.

Thank you for continue supporting this robust SAT>IP server! :+1:

Barracuda09, 2021-01-28:
Hi @lars18th ,

Yes I will implement this in the near future. But sorry, I have unfortunately less time to spent on this at this point in time.

I have to look if it is easy to implement this.

lars18th, 2021-02-22:
Hi @Barracuda09 ,

Just to be sure: After https://github.com/Barracuda09/SATPI/commit/ee20639cfcbdf7fb9c7e68b10d033de9d059c1f7 if I *don't* set the new parameter `--enable-unsecure-frontends` but I set the `--childpipe`, then the PIPE forntend will work when using the `mapping.m3u` file?

If so, thank you so much! :+1:

Barracuda09, 2021-02-22:
@lars18th 

if you `SatPI --help` it should be there?

lars18th, 2021-02-22:
> @lars18th
> 
> if you `SatPI --help` it should be there?

Yes, but this is not the question. I see the source code. My request if this is the _expected_ behaviour. :stuck_out_tongue_winking_eye:

Barracuda09, 2021-02-22:
Yes sorry misunderstood you. You are correct that would be the intention.

lars18th, 2021-02-22:
Hi @Barracuda09 ,

I compiled and run it, but this doesn't work! I have in the `mapping.m3u` the remap to use the PIPE when requesting some frequency. That's working. However, if I request from external, for example with `http://satpi-server:8554/?msys=childpipe&exec="..."` then it continues to work. For launching the SATPI I'm using only the parameter `--childpipe` and not the new `--enable-unsecure-frontends` (that I understand that is off by default).

Or the parameters was implemented as `--disable-unsecure-frontends` and I neet to set it?

Barracuda09, 2021-02-22:
No you should ONLY set  `--enable-unsecure-frontends` to enable direct request like `http://satpi-server:8554/?msys=childpipe&exec="..."`

Sorry, I must say I did not test it myself. I thought it should work. I have to check this.

lars18th, 2021-02-22:
Sorry @Barracuda09 ,

I dot a mistake! I start to run it, without killing the previous service. Now I'm testing the current version, but I feel it has a bug too! If I start with `--enable-unsecure-frontends` the program doesn't starts. Please, check it.

Barracuda09, 2021-02-22:
Hi @lars18th 

Did you enable the _Transformation Info_ in frontend page
- _Transformation enable_  -> checked
- _Advertise as_ -> DVB-S2

The start problem is fixed with a1f4d3628d8cbd5656bc102000b5b3d09deb40cf

lars18th, 2021-02-22:
Hi @Barracuda09 ,

> The start problem is fixed with a1f4d36

Yes, now the satpi accepts the parameter. Thank you!

> Did you enable the Transformation Info in frontend page

Sure! And now, using the parameters `--childpipe --enable-unsecure-frontends` the translation works. But using only `--childpipe` it doesn't work. So the functionality is blocking, at time, the mapping. So please, can you fix it?

Regards.

Barracuda09, 2021-02-22:
@lars18th 

Please try again with 41c2dfdac09d1d8a412c66c39508847cb9d387e7

It should be fixed now, tested my self this time.

lars18th, 2021-02-22:
Yes, now is working!
So I close this issue as it's resolved.
Thank you @Barracuda09 !

closed 2021-02-22 by lars18th