Barracuda09/SATPI #110: GBQuad STB - satpi causing kernel crash

maedula, 2021-02-01:
Hi all,

have any of you ever noticed that satpi is causing a kernel crash? This is happening to me and I have no clue why. I can install satpi on my gbquad STB and everything will work fine. But once I reboot the system gets stuck and dmesg indicates a kernel crash caused by satpi. It seems like the crash is caused by a function call within dvbapi and while it is trying to retrieve information from my DVB-C tuner SP8221L. Any help to figure out the issue is appreciated. Thanks a lot!

Best Regards,
maedula

[   20.667000] CPU 1 Unable to handle kernel paging request at virtual address 00000000, epc == e0b25770, ra == e0b258d0
[   20.678000] Oops[#1]:
[   20.678000] CPU: 1 PID: 528 Comm: satpi Tainted: P           O    4.8.3 #1
[   20.678000] task: cd289880 task.stack: ccf64000
[   20.678000] $ 0   : 00000000 10008b00 e0b25770 00001b4c
[   20.678000] $ 4   : 00000000 000000f8 00000001 ccf65b00
[   20.678000] $ 8   : 00000010 fffffffd cda7f000 7fede8a0
[   20.678000] $12   : 00000015 76c410b8 00000000 7fede8a8
[   20.678000] $16   : 00000001 00000000 e1200000 00000006
[   20.678000] $20   : 000000f8 00000000 cced5900 e1277b98
[   20.678000] $24   : 00000000 76dec40c
[   20.678000] $28   : ccf64000 ccf65af0 7fede7e8 e0b258d0
[   20.678000] Hi    : 00001063
[   20.678000] Lo    : 0006e30b
[   20.678000] epc   : e0b25770 L1_ReadBytes+0x0/0x60 [dvb]
[   20.678000] ra    : e0b258d0 L1_ReadRegister+0x78/0x194 [dvb]
[   20.678000] Status: 10008b03 KERNEL EXL IE
[   20.678000] Cause : 00800008 (ExcCode 02)
[   20.678000] BadVA : 00000000
[   20.678000] PrId  : 00025a11 (Broadcom BMIPS5000)
[   20.678000] Modules linked in: ipv6 boxmodel(PO) xfs libcrc32c dvb(O)
[   20.678000] Process satpi (pid: 528, threadinfo=ccf64000, task=cd289880, tls=7714e4a0)
[   20.678000] Stack : ccf65ce8 8007eda8 ccf4f000 8005b5a8 ccf1ff00 00000000 00000001 ccf65b90
[   20.678000]    00000000 ccf65b90 ccf65dc8 00000000 00000000 e0b2965c 808f2080 80843414
[   20.678000]    80930000 cfc19d00 00000000 8005045c e1277b98 e0b1eabc 00000000 800505e0
[   20.678000]    8191ee00 80050490 ccf4f000 80576a14 8191ee00 00000000 cd2e7c00 ccf65dc8
[   20.678000]    cd2e7c00 ccf65dc8 ccf1ff80 80578754 8191ee00 8097fe00 808f23c8 ccf65dd8
[   20.678000]    ...
[   20.678000] Call Trace:
[   20.678000] [<e0b25770>] L1_ReadBytes+0x0/0x60 [dvb]
[   20.680000] [<e0b258d0>] L1_ReadRegister+0x78/0x194 [dvb]
[   20.680000] [<e0b2965c>] SP8221L_get_frontend+0x64/0xd0 [dvb]
[   20.680000] [<e0b1eabc>] dev_dmx_get_fops+0x1360/0x5b54 [dvb]
[   20.680000]
[   20.680000]
[   20.680000] Code: 2739558c  03200008  01003821 <8c830000> 240800c8  90640000  10880003  00001021  03e00008
[   20.680000] ---[ end trace 4b4fc0b736aea978 ]---

Barracuda09, 2021-02-01:
Hi @maedula 

Thanks for your feedback.
Is there a file created, something like  _/tmp/satpi.bt_

Regards,

Marc

maedula, 2021-02-01:
Hi @Barracuda09 

no...there is unfortunately no such file existing in tmp. I also tried the build from [https://github.com/Barracuda09/SATPI/issues/57#issuecomment-719750804](url) causing the same crash but no satpi.bt.

Best Regards,
maedula

Barracuda09, 2021-02-01:
But the system crashed and is halted or is is still running. And can you check if SatPI is still running?

maedula, 2021-02-01:
It is still running and I can connect to it via SSH. But the Enigma2 subsystem does not seem to boot up. TV is stuck in the initial OpenATV bootscreen. SatPI is not running.

Barracuda09, 2021-02-01:
Was SatPI running? you can set logging to syslog (could you check that)

maedula, 2021-02-01:
Syslog is enabled. I can see the loglines after initial installation. After reboot you just see the kernel crash related to satpi
[messages.txt](https://github.com/Barracuda09/SATPI/files/5903260/messages.txt)

Barracuda09, 2021-02-01:
I do not see any logs from SatPI in syslog. Did you enable it in SatPI

[image]

maedula, 2021-02-01:
Correct...you do not see any. But the flag is enabled and I see the log entries after initial installation. Just not after the reboot for which I shared with you the messages.txt

Barracuda09, 2021-02-01:
If you do a power cycle it is okey?

maedula, 2021-02-01:
Do you mean a physical switch off and on the STB? Up to now I only used command `reboot`. But I do not expect any other results though. I can retry if you like

Barracuda09, 2021-02-01:
Yes please switch off and on again.

maedula, 2021-02-01:
Removed power plug...waited 10 seconds. Plugged it back in. Same issue

Barracuda09, 2021-02-01:
How did you install it at all then?

maedula, 2021-02-01:
my latest Openatv image v4.6 has its own package management tool opkg. I installed it using its default repo. As mentioned before I also swapped out all files against your debug built for mips E2 with same result.

Barracuda09, 2021-02-01:
What if you start satpi again with _--no-daemon_ ?

maedula, 2021-02-01:
I have tried a few things more and it seems to be related to the fact that the process is added to autostart via init.d. If I remove autostart and start satpi manually once the system is up and running, satpi works well and no kernel crash is caused.

Barracuda09, 2021-02-02:
So SatPI is started to early?

maedula, 2021-02-02:
That´s probably the conclusion. Even though I do not know what previous service it is waiting for. I have worked around this now by adding SatPI to a cron job waiting 120 seconds after boot before being started. That seems to work well and I have removed the init.d start scripts auto-installed by opkg. Thanks a lot for your support and my apologies for all that noise. It might be good to know what services need to be available for SatPI for not causing the kernel crash.

closed 2021-02-02 by maedula