Barracuda09/SATPI #111: TROUBLE: Slow streaming with high bandwith channels

lars18th, 2021-02-17:
Hi @Barracuda09 ,

This is not directly related to #97. But I see this when streaming from PIPE one UHD channel (~25Mbps). In this case the output has many CC errors and the total bitrate is around ~6Mbps.

So the question is: Is the _output_ bandwith limited in any sense?

lars18th, 2021-02-17:
Hi @Barracuda09 ,

I detected where is the problem. So, I want to share it with you:

- This line https://github.com/Barracuda09/SATPI/blob/232e5b6d8bcb6b94adf72cfc7fb7f2dad8b6e64f/src/input/childpipe/TSReader.cpp#L96 enforces to use PCR timestamps with PIPE inputs. However, the input perhaps have incorrect PCR values. And in this case, the code doesn't work.

At this point, we have three options:
1. Disable the PCR check in PIPE inputs (a simple `if(0)` in this line fixes the problem).
2. Include one flag for PIPE inputs to indicate if the user wants or not to use PCR checks for this input (like `?msys=childpipe&exec="..."&nopcr`).
3. Improve the code of handling PCR timestamps with PIPE inputs.

What you prefer?
Please comment about it.

lars18th, 2021-02-17:
Just to note: When using PIPE inputs, the stream can be readed from two different sources: from disk (or similar) and from network (or similar). The first implies that you need to read based on PCR. And the second implies to read at full speed, as it's an streaming. So the flag to select to use PCR or not, seems the best option from my point of view.

Barracuda09, 2021-02-18:
@lars18th Thanks again for you support.

This could be an solution:

This could be added after line 63
```
		const int pcrtime = StringConverter::getIntParameter(msg, method, "pcrtime=");
		if (pcrtime != -1) {
			_pcrtime = pcrtime;
		}
```
The return of _pcrtime could be added after line 91 somewere:
```
		int TSReaderData::getPCRTime() const {
			base::MutexLock lock(_mutex);
			return _pcrtime;
		}
```

https://github.com/Barracuda09/SATPI/blob/135dc39d3681c8fd8d64b0924a30bf42657c8179/src/input/childpipe/TSReaderData.cpp#L53-L65

When:
-  _pcrtime = 0 it should use PCR (Or if not specified).
-  _pcrtime > 0 it could be used for timing.

lars18th, 2021-02-18:
Hi @Barracuda09 ,

So you're suggesting to let the user the option for selecting to use PCRs or not, right? Then I agree with you. I hope you can implement it when you've sufficient time. Until that I use the simple `if(0)`workaround.

Thank you for your response.
Regards.

lars18th, 2021-02-22:
Hi @Barracuda09 ,

I see your commit https://github.com/Barracuda09/SATPI/commit/20ee7d80ce72a2267ae71a0eb66a86abf4344009 but I don't know how to use it. Plase, can you explain more?

Thank you!

Barracuda09, 2021-02-22:
@lars18th 

It should work like option 2 in only the param is called 'pcrtimer'

> At this point, we have three options:
>
>     1. Disable the PCR check in PIPE inputs (a simple `if(0)` in this line fixes the problem).
>
>     2. Include one flag for PIPE inputs to indicate if the user [...]

When:
- pcrtimer=0  means the PCR from stream is used
- pcrtimer>0  means this timer + 150 us is used

lars18th, 2021-02-22:
OK,

So the correct syntax is `?msys=childpipe&exec="..."&pcrtimer=0` for use the PCR from stream (the default value I suspect, equal to until now), and `?msys=childpipe&exec="..."&pcrtimer=10` to use 160us as wait time. Correct? Then to use at full speed I need to use `&pcrtimer=1` to obtain similar to the current `if(0)` workaround, right?

Good job then! :+1:

Barracuda09, 2021-02-22:
Yes, that would be the intention

lars18th, 2021-02-22:
OK. Tested and working. :+1: 

So I suggest to close this issue, but after it will be documented. Please remember to indicate to recommend `&pcrtimer=1` for no waiting.

Barracuda09, 2021-02-22:
@lars18th 

I have tried to document it here in the wiki [05.-Explanation-of-Virtual-tuners](https://github.com/Barracuda09/SATPI/wiki/05.-Explanation-of-Virtual-tuners)

I am not an expert in documenting or explaining this very good. Hope this helps somewhat.

lars18th, 2021-02-22:
Just to comment: I recommend 
```
add &pcrtimer=1 to command of ChildPIPE to force the use of time base read (Time-interval is 150 us + 1; where the "1" can be changed to any number that represents the increment in microseconds)"
```

Regards.

Barracuda09, 2021-02-23:
@lars18th, I have added you suggestion, and i will close this one

closed 2021-02-23 by Barracuda09