Barracuda09/SATPI #115: BUG: LOG prints "Stream" instead of "FrontEnd"

lars18th, 2021-02-22:
Hi @Barracuda09 ,

See this part of the LOG:
```
  Mon Feb 22 14:19:11.6781 2021  7  RTSP/1.0 503 Service Unavailable
  Mon Feb 22 14:19:11.6780 2021  3  Found no Stream/Client of interest!
  Mon Feb 22 14:19:11.6780 2021  6  Stream: 3, Not capable of handling msys=dvbs2
  Mon Feb 22 14:19:11.6780 2021  6  Stream: 2, Not capable of handling msys=dvbs2
  Mon Feb 22 14:19:11.6780 2021  6  Stream: 1, Not capable of handling msys=dvbs2
  Mon Feb 22 14:19:11.6779 2021  6  Stream: 0, New session but this stream is not enabled, skipping...
  Mon Feb 22 14:19:11.6778 2021  6  Found StreamID x - SessionID: 0276486420
  Mon Feb 22 14:19:11.6685 2021  7  CSeq: 1
```

I feel that the correct print is to indicate "FrontEnd" and not "Stream", as the check is regarding the TUNER, and not the STREAM.
I'm right?

Barracuda09, 2021-02-24:
@lars18th 

Yes, you are probably right

lars18th, 2021-02-24:
Hi @Barracuda09 ,

What I see is that you call "Stream" to an internal source, so far a "Tuner" or "Virtual Tuner". I don't know why this, but that's true in the source code. So, in case you don't want to change it, at least please use a friendlier name in the LOG.

Regards.

Barracuda09, 2022-09-22:
Should be fixed in the meantime.

closed 2022-09-22 by Barracuda09