Barracuda09/SATPI #116: Possibility to add more ChildPIPE frontends

Barracuda09, 2021-02-24:
Add with option `--childpipe` a number of how many ChildPIPE frontends are enumerated

lars18th, 2021-03-05:
Hi @Barracuda09 ,

Why you want to enumerate them?

Barracuda09, 2021-03-05:
@lars18th 

That is how they are added. So now only 1 is added at the moment

https://github.com/Barracuda09/SATPI/blob/e0b638609c456581d3b2983965ea3ad43867d422/src/StreamManager.cpp#L59-L80

lars18th, 2021-03-05:
Ahh! So the idea is to provide more than streams over the childpipe in parallel, right?
Great idea! :+1:

Barracuda09, 2021-03-05:
Yes to have more the 1 ChildPIPE frontend with with argument `--childpipe` to add there a number like `--childpipe 3` for 3 ChildPIPE frontends

Barracuda09, 2021-03-05:
@lars18th 

Could you check last commits?

Thanks

lars18th, 2021-03-05:
Hi @Barracuda09 ,

> Could you check last commits?

Tested. And it works like a charm. Now I can stream 2 different pipes at time.

However, I've discovered that SATPI can't share streams. For example, if I do two client connections requesting the same childpipe (using remaping) then two _different_ frontends are open. Why not share the frontends? It's a waste of resources.

Barracuda09, 2021-03-06:
No, sharing of frontends with the same "channel/parameters" is something I have not really implemented (yet).

lars18th, 2021-03-06:
OK. I suggest to close this issue as the functionality is already implemented.

Thank you|

Barracuda09, 2021-03-06:
Done

closed 2021-03-06 by Barracuda09