Barracuda09/SATPI #117: Configuration for Cband

glockmaster, 2021-02-27:
Hi,

I tried all on Satpi with Raspberry pi 4 and i cant make C-band work like 55.5W.
Any tips?

Barracuda09, 2021-02-27:
Hi @glockmaster 

First of all, thanks for trying SatPI.

Which setup do you use, like LNB type (freq.) maybe switches etc.

glockmaster, 2021-02-27:
My Setup is C-Band LNBF. Its my first time trying such an incredible new world of satellite with this tool. My Frontend0 only shows the range of 95000hz to 215000hz, i dont find a place to change those values to correct, plus on LNB switch im using 5150 | 0 | 0 as LOF in order 1st to 3rd.
I Can connect with progdvb wich i own a license but i cant find any channels na d the log says "Failed to update frontend".
I would love to use this tool with Cband and not only astra 19.2E or 28.2E.

Barracuda09, 2021-02-28:
> My Frontend0 only shows the range of 95000hz to 215000hz

This is not adjustable, this is an property of the tuner. 

I am no expert, but you could try this:

Monoblock  without switch use:
   `Lof High (MHz)`  => 0
   `Lof Low (MHz)` => LNB Freq.
   `Lof Switch (MHz)` => 0

C-Band Monoblock  without switch use:
   `Lof High (MHz)`  => LNB Freq.
   `Lof Low (MHz)` => LNB Freq.
   `Lof Switch (MHz)` => 0

closed 2021-03-06 by Barracuda09