Barracuda09/SATPI #118: Suggestion: Put a link of the XML in the UI

lars18th, 2021-03-04:
Hi @Barracuda09 ,

Perhaps this sounds stupid, but every SAT>IP server uses a diffent path, and it's not possible to see the SSDP messages without some tools. So I suggest to add a simple link to the XML file in some part of the UI. Then it will be more easy to read it, or to copy the address. I know that in SATPI is `/desc.xml` but including the full URL will be useful.

You agree?

Barracuda09, 2021-03-04:
Hi @lars18th 

Here you have the Description file. Do you means something like this? or really the file content?

[image]

lars18th, 2021-03-05:
> Here you have the Description file. Do you means something like this? or really the file content?

I want the full link! So, at the left you can write something like:  "Satip Description XML (link)"
So when you "copy" the link then you can paste it directly where you need it. Or if you click on it, then the page is loaded. You don't need to print it!

You accept it?
It' simple but useful.

Barracuda09, 2021-03-05:
@lars18th 

Could you check last commit, if this is what you mean?

lars18th, 2021-03-05:
> Could you check last commit, if this is what you mean?

Perfect! Thank you @Barracuda09 .

closed 2021-03-05 by lars18th