Barracuda09/SATPI #119: Suggestion: Frontend Overview streaming URL

lars18th, 2021-03-05:
Hi @Barracuda09 ,

This perhaps sound stupid to you, but it's very useful for testing and very simple to implement. The suggestion is to add in the "Frontend Overview" page a link of the current streaming. For example, if some tuner is in use then the URL will provide something like "rtsp://satpi-address://8554?src=4&freq=...&pids=all". The values are the parameters currently running, and the pids the current pid list. With this, it will very easy to "copy" this URL and paste it in some client (for example VLC, in GUI or CLI) to test the same request. It's very practical. And you can provide the RTSP and the HTTP versions if you want.

I hope you agree with this suggestion.
Regards.

Barracuda09, 2021-05-15:
This is at the moment only implemented for DVB-C and S2 requests

[image]

lars18th, 2021-05-15:
Thank you @Barracuda09.
However, I can't check it as I only use the Virtual Frontends _plus_ the Remapping. So, I hope you'll update it soon to support both.

Barracuda09, 2021-05-15:
Here is the Tooltip update:
[image]