Barracuda09/SATPI #122: Client port in UI

lars18th, 2021-03-07:
Hi @Barracuda09 ,

I don't want to request for stupid and non-useful changes. However, some small changes in the UI can help _much_ when doing testing or debug (to the SATPI or to a SAT>IP client).

So, another missing information in the UI is the RTP port used by the client. In the Frontend Overview (and in the detailed view of the FE), the "Owner" is the address of the client. Great! But then can you please add the RTP port too? In fact the **target port**, as when using the RTSP protocol it will be the RTP (UDP base port), and when using the HTTP protocol is the TCP client port.

With this information will be more easy to know the port for example to launch a simple tcpdump.
You want to implement it, please?
Thank you!

lars18th, 2021-04-08:
Hi @Barracuda09 ,

I hope you can implement this. It seems easy; and powerful for _testing_.
So, please can you add the TARGET PORT of the client in the UI?
Thank you!

lars18th, 2021-05-10:
Hi @Barracuda09 ,

To be more precise: In the "Frontend Info", in the section "Stream Info", when you write the "Owner", the idea is to print not only the IP address of the client, but the pair of RTP UDP ports when streaming as well. Example: `Owner:    192.168.1.51:43002-43003`.

Thank you!

lars18th, 2021-05-15:
Hi @Barracuda09 ,

I see the changes in the commit https://github.com/Barracuda09/SATPI/commit/cf539a4e4721f37f2d3f68cf71fe4b67b4c2bd03. However, after pull the last version and recompile, I don't see this in the web page. Only a new row "Stream URL" appears (even the link is wrong as it every time points to "http://satpi/frontend.html"). However, when I check for the "page source" the new lines for them are here:
```
page += addTableLineEntry("RTP Port", xmlDoc, streamID + "rtpPort");
page += addTableLineEntry("RTCP Port", xmlDoc, streamID + "rtcpPort");
page += addTableLineEntry("HTTP Port", xmlDoc, streamID + "httpPort");
```

You know what happens?

Barracuda09, 2021-05-15:
Hi @lars18th 

- The "Stream URL" only is implemented for now, when using DVB-C/S2. So not for Child-PIPE/Streamer inputs.
- I am not sure why you don't see the lines, here is how I see it:

[image]

lars18th, 2021-05-15:
Yes, that's what I expected. However, in my case I only see the rows (but in the source of the page the code it's here):
- Stream URL
- Owner
- Session ID
- User-Agent
- RTP packet count
- RTP streamed (MB)

Please, remember that I've enabled ONLY virtual tuners: 
- Frontend 0: TS Reader
- Frontend 1: Streamer
- Frontend 2: Child PIPE - TS Reader
- Frontend 3: Child PIPE - TS Reader

Could this have any influence?

lars18th, 2021-05-15:
My Chrome prints this in the console:
```
DevTools failed to load SourceMap: Could not load content for http://192.168.1.35:8080/assets/js/bootstrap.min.js.map: HTTP error: status code 404, net::ERR_HTTP_RESPONSE_CODE_FAILURE
DevTools failed to load SourceMap: Could not load content for http://192.168.1.35:8080/assets/css/bootstrap.min.css.map: HTTP error: status code 404, net::ERR_HTTP_RESPONSE_CODE_FAILURE
```
Any idea?

Barracuda09, 2021-05-15:
The example picture I gave was with Child PIPE.

Barracuda09, 2021-05-15:
> My Chrome prints this in the console:

I am not sure if this is the problem, as I have it as well

Barracuda09, 2021-05-15:
Here you can get them and try it
https://github.com/twbs/bootstrap/releases/download/v4.5.3/bootstrap-4.5.3-dist.zip

lars18th, 2021-05-15:
Hi @Barracuda09 ,

Problem fixed. **It's not related to the bootstrap**. However, I've updated it.

Before close this issue (now fixed/implemented) I request to enlarge a lot the default vertical size of the Frontend Info tab. Then we don't need to use the scroll as now just for a very small points the scroll is activated.

Barracuda09, 2021-05-15:
@lars18th 

Yes, or maybe just have Bootstrap resize it (But I am not an expert in Web Development/Design I have managed to get this far, and with help from @issue )