Barracuda09/SATPI #123: Request: Use correct RTCP description messages with REMAP

lars18th, 2021-03-07:
Hi @Barracuda09 ,

When using the remap functionality the RTCP messages describing the tuner status are (partially) incorrect. In fact, if you request a DVB-S2 frequency and then you remap it to a TSreader, then you receive the data of the second. Then instead of sending the data composed from this function:
https://github.com/Barracuda09/SATPI/blob/6832414b529494634ec4185f868cdf8c4f40ac8e/src/input/dvb/FrontendData.cpp#L341
you receive the calculated inside this other function:
https://github.com/Barracuda09/SATPI/blob/6832414b529494634ec4185f868cdf8c4f40ac8e/src/input/file/TSReaderData.cpp#L69

And this can create troubles with clients. So I suggest a simple way to send the correct message. When you do a remap, store the original mtype and send the message based on this information.

I hope you'll want to consider this modification. Even without a high priority since not all clients interpret the messages completely to fully work.
Regards.

lars18th, 2021-03-09:
Hi @Barracuda09 ,

Some clients refuse to work because the values doesn't match. Please, take note that to check that the tunning is working, the client can check the requested frequency. So, I suggest to create a "fake" response for each modulation type and change only the value of the frequency based on the original remap (and perhaps the polarisation too). For example:

- DVB-S2: `ver=1.0;src=<srcID>;tuner=<feID>,240,1,15,<freq>,<polarisation>,dvbs,,,,,;pids=`
- DVB-T: `ver=1.1;tuner=<feID>,240,1,15,<freq>,,dvbt,,,,,,,;pids=`
- DVB-C: `ver=1.2;tuner=<feID>,240,1,15,<freq>,,dvbc,,,,,,;pids=`

I feel you can improve this function to include the remmaping: https://github.com/Barracuda09/SATPI/blob/6832414b529494634ec4185f868cdf8c4f40ac8e/src/input/dvb/FrontendData.cpp#L341
And use it with _all_ type of inputs (including the `ver=1.5` that you've added for software frontends).

Barracuda09, 2021-03-09:
Hi @lars18th 

It should have worked like this. So if you have a DVB-S2 mapping to "ChildPIPE" is should give a DVB-S2 describe string.

I have to look into this why this is not occurring.

lars18th, 2021-03-10:
Hi @Barracuda09 ,

Please, check it. As in fact it doesn't work as expected.
And remember to centralize the `ver=1.5` code inside the `FrontendData` class. As at time, the code is repeated multiple times inside the subclasses of software frontends.

Thank you to review it!

lars18th, 2021-03-13:
Hi @Barracuda09 ,

Almost the error is when remapping to a childpipe. So, perhaps the error is not when doing remaps from physical sources.

Regards.

lars18th, 2021-03-13:
Hi @Barracuda09 ,

I see that this is identical to #76. So please, close one of these issues as they're duplicated.

lars18th, 2021-04-08:
Hi @Barracuda09 ,

Any update on this?

lars18th, 2021-05-10:
This is related with #76 (but it isn't the same).

lars18th, 2022-01-11:
Hi @Barracuda09 ,

You'll want to fix this?

lars18th, 2022-11-16:
Hi @Barracuda09 ,

Please when you will commit my PR #179 close this issue. All troubles will be fixed.

Regards.