Barracuda09/SATPI #124: src/decrypt/dvbapi/Client.cpp:229] Connecting to OSCam Server failed

BaltasarParreira, 2021-03-11:
I Know there is already a few explanations about using OSCAM here, but I setup everything using enter to save and so on, but still can't connect to may local OSCAM where SATPI is running.
Always only getting:
src/decrypt/dvbapi/Client.cpp:229] Connecting to OSCam Server failed

Is this still working with the latest OSCAM and SATPI?

Thanks.

Barracuda09, 2021-03-11:
Hi @BaltasarParreira 

First of all thanks for your feedback.

Well I don't use the cutting edge version of OSCam (_I use version R11272_). I know the API has changed over time and probably not working in combination with SatPI at the moment.

Regards,
Marc

BaltasarParreira, 2021-03-11:
Ok, going to install that version and see if that's the problem.
Anyway any specific oscam.conf parameters aparte the usual:
[dvbapi]
enabled                       = 1
au                            = 1
pmt_mode                      = 2
listen_port                   = 9999
delayer                       = 60
read_sdt                      = 1
write_sdt_prov                = 1

You answer so quickly that I was editing the previous post, but I ask again anyway regarding those parameters I put there.  ;)

Thanks.

BaltasarParreira, 2021-03-11:
Sorry not working, I'm running now the OSCam r11272 but still getting same error.

My SATPI version is: SatPI version: 1.5.356~g1f7aa3e Enigma

Did you made any changes that could solve this problem after this version?

Thanks.

BaltasarParreira, 2021-03-11:
Answering myself, I just compile the latest SATPI, now running: 1.5.409~g45892cc

But same problem, can't connect to OSCAM.
Don't really know what I can do more so waiting for your feed back please.

Thanks.

Barracuda09, 2021-03-11:
I have these as API
[dvbapi]
enabled  = 1
au           = 1
pmt_mode = 4
listen_port  = 15012
user            = TEST
read_sdt     = 2
boxtype     = pc

BaltasarParreira, 2021-03-11:
Ok and did you create the user "TEST"?

Barracuda09, 2021-03-11:
Yes I did

BaltasarParreira, 2021-03-11:
Ok, still nothing.

So my scenario is, a Gigablue UHD UE 4K receiver running OSCAM and SATPI inside.
IS this ok for the SATPI configuration?
[image]

Barracuda09, 2021-03-11:
I have this:

[image]

BaltasarParreira, 2021-03-11:
Nothing... really strange since I have everything identical.
This is my oscam.version giving by the interface:

```
Unix starttime: 1615499023
Starttime:      11.03.2021 21:43:43
Version:        oscam-1.20-unstable_svn-r11272
Compiler:       arm-newnigma2-linux-gnueabihf
Box type:       gigablue (generic)
PID:            2701
TempDir:        /tmp/.oscam
ConfigDir:      /etc/tuxbox/config/oscam-stable/
WebifPort:      83

Web interface support:                   yes
LiveLog support:                         yes
jQuery support intern:                   yes
Touch interface support:                 no
SSL support:                             no
DVB API support:                         yes
DVB API with AZBOX support:              no
DVB API with MCA support:                no
DVB API with COOLAPI support:            no
DVB API with COOLAPI2 support:           no
DVB API with STAPI support:              no
DVB API with STAPI5 support:             no
DVB API read-sdt charsets:               yes
Irdeto guessing:                         yes
Anti-cascading support:                  no
Debug mode:                              yes
Monitor:                                 yes
Loadbalancing support:                   yes
Cache exchange support:                  no
CW Cycle Check support:                  no
LCD support:                             no
LED support:                             no
Clockfix with realtime clock:            no
IPv6 support:                            no

camd 3.3x:                               no
camd 3.5 UDP:                            yes
camd 3.5 TCP:                            yes
newcamd:                                 yes
CCcam:                                   yes
CCcam share:                             yes
gbox:                                    yes
radegast:                                no
scam:                                    yes
serial:                                  no
constant CW:                             no
Pandora:                                 no
ghttp:                                   no

Reader support:                          yes

Nagra:                                   yes
Irdeto:                                  yes
Conax:                                   yes
Cryptoworks:                             yes
Seca:                                    yes
Viaccess:                                yes
NDS Videoguard:                          yes
DRE Crypt:                               yes
TONGFANG:                                yes
Bulcrypt:                                yes
Griffin:                                 yes
DGCrypt:                                 yes

cardreader_phoenix:                      yes
cardreader_drecas:                       no
cardreader_internal_azbox:               no
cardreader_internal_coolapi:             no
cardreader_internal_coolapi2:            no
cardreader_internal_sci:                 yes
cardreader_sc8in1:                       no
cardreader_mp35:                         no
cardreader_smargo:                       no
cardreader_pcsc:                         no
cardreader_smartreader:                  no
cardreader_db2com:                       no
cardreader_stapi:                        no
cardreader_stapi5:                       no
cardreader_stinger:                      yes
```

Do you have any other protocol or something that is missing in mine?

Barracuda09, 2021-03-11:
You did restart OSCam I assume.

BaltasarParreira, 2021-03-11:
Yes I did.
I'm starting SATPI only like this:
satpi --http-path /usr/share/satpi/web

Do you use any other parameter?

Barracuda09, 2021-03-11:
I will convert this to a discussion

closed 2021-03-11 by Barracuda09