Barracuda09/SATPI #126: SATPI and Raspberry Pi TV HAT

petrkutalek, 2021-03-24:
I wanted to use SATPI V1.6.2 with Raspberry Pi TV HAT (Sony CXD2880 based).

I was not successful at all, the only result I am able to get is this error:
```
…
Found StreamID x - SessionID: 0299846454
Stream: 0, StreamClient[0] with SessionID 0299846454 for dvbt
Stream: 0, Parsing transport parameters...
Stream: 0, New frequency requested, clearing old channel data...
Stream: 0, Parsing transport parameters (Finished)
Stream: 0, Found Streaming type: HTTP
Stream: 0, Updating frontend...
Stream: 0, Opened /dev/dvb/adapter0/frontend0 fd: 14
Stream: 0, Start tuning process for DVB-T(2)...
Stream: 0, Waiting on lock...
Stream: 0, Not locked yet   (FE status 0x0)...
Stream: 0, Not locked yet   (FE status 0x0)...
Stream: 0, Not locked yet   (FE status 0x0)...
Stream: 0, Not locked yet   (FE status 0x0)...
Stream: 0, Updating frontend (Failed)
…
```
and
```
…
Mar 24 22:09:34 DVB-T2 kernel: cxd2880: cxd2880_set_frontend: sys:16 freq:514000000 bw:8
Mar 24 22:09:35 DVB-T2 kernel: cxd2880: cxd2880_set_frontend: tune result 0
Mar 24 22:10:30 DVB-T2 kernel: cxd2880: cxd2880_set_frontend: sys:16 freq:514000000 bw:8
Mar 24 22:10:31 DVB-T2 kernel: cxd2880: cxd2880_set_frontend: tune result 0
Mar 24 22:16:07 DVB-T2 kernel: cxd2880: cxd2880_set_frontend: sys:16 freq:514000000 bw:8
Mar 24 22:16:08 DVB-T2 kernel: cxd2880: cxd2880_set_frontend: tune result 0
Mar 24 22:16:52 DVB-T2 kernel: cxd2880: cxd2880_set_frontend: sys:16 freq:514000000 bw:8
Mar 24 22:16:53 DVB-T2 kernel: cxd2880: cxd2880_set_frontend: tune result 0
Mar 24 22:17:29 DVB-T2 kernel: cxd2880: cxd2880_set_frontend: sys:16 freq:570000000 bw:8
Mar 24 22:17:30 DVB-T2 kernel: cxd2880: cxd2880_set_frontend: tune result 0
Mar 24 22:19:09 DVB-T2 kernel: cxd2880: cxd2880_set_frontend: sys:16 freq:570000000 bw:8
Mar 24 22:19:10 DVB-T2 kernel: cxd2880: cxd2880_set_frontend: tune result 0
Mar 24 22:20:06 DVB-T2 kernel: cxd2880: cxd2880_set_frontend: sys:16 freq:570000000 bw:8
Mar 24 22:20:07 DVB-T2 kernel: cxd2880: cxd2880_set_frontend: tune result 0
…
```

I would like to ask you - does anyone use SATPI with CXD2880 (RPi TV HAT) with any success?

Barracuda09, 2021-03-24:
Hi @petrkutalek 

Thanks for giving this a try.

I have used it with a Pi and an DVB-T2 usb stick with no issues.

But I have some questions:
- Could you provide a more complete log (With tuning parameters and requests)
- Which client are you using

Regards

robbieb43, 2021-03-25:
I also have been successfully using this combination (two instances) for some time now. Works very well.
At least on one of these I am using v1.5.270 if that helps.  I would not overlook the physical aspects as getting a decent stable coax connection to the device is a challenge - I made a support that stopped the weight of the cable pulling on the mini plug. Looks like your device is recognised so that’s a start.  I also found dvbscan and it’s variants useful in the set up phase to ensure your device is seeing signal.

robbieb43, 2021-03-25:
Sample request url  (BBC 1 HD off secondary transmitter in South of UK)
http://192.168.1.163:8875/?freq=546&bw=8&msys=dvbt2&tmode=32K&mtype=256qam&gi=1128&fec=23&pids=0,1,6600,6601,6602,6606,7319,7302"

petrkutalek, 2021-03-25:
Thank you @robbieb43!! I like this project @Barracuda09, I will be happy to share all details, of course. Please just give me some time.

Barracuda09, 2021-04-03:
Hi @petrkutalek and @robbieb43 

Could you try the last commit and see if that makes any difference?

Thanks,

Marc

robbieb43, 2021-04-03:
Hi Marc

I am a bit of a git ‘numpty’ but assuming with ‘SatPI version: 1.5.414~g9d9fa22’ I have compiled the right version, all looks good.

robbieb43, 2021-04-03:
To be on the safe side I have now downloaded the zip of the last comic (which gives version unknown) and this too is working well. 
  Sat Apr  3 13:43:56.2703 2021  6  --- Starting SatPI version: -unknown ---
  Sat Apr  3 13:43:56.2707 2021  6  Number of processors online: 4
  Sat Apr  3 13:43:56.2708 2021  6  Default network buffer size: 80 KBytes
  Sat Apr  3 13:43:56.2713 2021  6  eth0: 192.168.0.23 [dc:a6:32:0e:58:e5]
  Sat Apr  3 13:43:56.2718 2021  6  Current DVB_API_VERSION: 5.11
  Sat Apr  3 13:43:56.2718 2021  6  Enumerating all devices...
  Sat Apr  3 13:43:56.2718 2021  6  Detecting frontends in: /dev/dvb
  Sat Apr  3 13:43:56.2741 2021  6  Frontend Name: Sony CXD2880
  Sat Apr  3 13:43:56.2742 2021  6  Frontend Type: Terrestrial (DVB-T)
  Sat Apr  3 13:43:56.2742 2021  6  Frontend Type: Terrestrial (DVB-T2)
  Sat Apr  3 13:43:56.2742 2021  6  Frontend Freq: 174000000 Hz to 862000000 Hz
  Sat Apr  3 13:43:56.2742 2021  6  Frontend srat: 0 symbols/s to 0 symbols/s
  Sat Apr  3 13:43:56.2744 2021  6  Frontends found: 1
  Sat Apr  3 13:43:56.2744 2021  6  Setting up TS Reader using path: .
  Sat Apr  3 13:43:56.2759 2021  6  Setting up TS Streamer
  Sat Apr  3 13:43:56.2800 2021  6  Setting HTTP Port to: 8875
  Sat Apr  3 13:43:56.2800 2021  6  Setting RTSP Port to: 554
  Sat Apr  3 13:43:56.2800 2021  6  Setting WEB Path to: ./web
  Sat Apr  3 13:43:56.2801 2021  6  Setting App Data Path to: .
  Sat Apr  3 13:43:56.2840 2021  6  Setting up HTTP server
  Sat Apr  3 13:43:56.2841 2021  6  Setting up RTSP server
  Sat Apr  3 13:43:56.2874 2021  6  Setting up SSDP server with BOOTID: 2  annouce interval: 60 Sec
  Sat Apr  3 13:43:56.8517 2021  6  HTTP Connection from 192.168.0.29 Port 60082 with fd: 8
  Sat Apr  3 13:43:56.8520 2021  7  Client: 192.168.0.29 requested desc.xml
  Sat Apr  3 13:44:27.0215 2021  6  Root Device Client 192.168.0.29 : tries to discover the network, sending reply back
  Sat Apr  3 13:44:27.0218 2021  6  Root Device Client 192.168.0.29 : tries to discover the network, sending reply back
  Sat Apr  3 13:44:35.5519 2021  6  HTTP Connection from 192.168.0.62 Port 58729 with fd: 9
  Sat Apr  3 13:44:35.5660 2021  6  HTTP Connection from 192.168.0.62 Port 58730 with fd: 10
  Sat Apr  3 13:44:35.5661 2021  6  HTTP Connection from 192.168.0.62 Port 58731 with fd: 11
  Sat Apr  3 13:44:35.5702 2021  6  HTTP Connection from 192.168.0.62 Port 58732 with fd: 12
  Sat Apr  3 13:45:09.7222 2021  6  HTTP Client 192.168.0.62:58729 Connection closed with fd: 9
  Sat Apr  3 13:45:09.7223 2021  6  HTTP Client 192.168.0.62:58730 Connection closed with fd: 10
  Sat Apr  3 13:45:09.7224 2021  6  HTTP Client 192.168.0.62:58731 Connection closed with fd: 11
  Sat Apr  3 13:45:09.7224 2021  6  HTTP Client 192.168.0.62:58732 Connection closed with fd: 12
  Sat Apr  3 13:45:19.0099 2021  6  Root Device Client 192.168.0.29 : tries to discover the network, sending reply back
  Sat Apr  3 13:45:19.0102 2021  6  Root Device Client 192.168.0.29 : tries to discover the network, sending reply back
  Sat Apr  3 13:45:22.2816 2021  6  HTTP Connection from 192.168.0.23 Port 50436 with fd: 9
  Sat Apr  3 13:45:22.2817 2021  7  HTTP Stream data from client None with IP 192.168.0.23 on Port 50436: GET /?freq=530&bw=8&msys=dvbt&tmode=8k&mtype=64qam&gi=14&fec=12&pids=0,16,300,301,302,303,304,311,115 HTTP/1.1
  Sat Apr  3 13:45:22.2817 2021  7  User-Agent: Lavf/58.20.100
  Sat Apr  3 13:45:22.2817 2021  7  Accept: */*
  Sat Apr  3 13:45:22.2817 2021  7  Range: bytes=0-
  Sat Apr  3 13:45:22.2817 2021  7  Connection: close
  Sat Apr  3 13:45:22.2817 2021  7  Host: 192.168.0.23:8875
  Sat Apr  3 13:45:22.2817 2021  7  Icy-MetaData: 1
  Sat Apr  3 13:45:22.2817 2021  7  --- LINE END ---
  Sat Apr  3 13:45:22.2819 2021  6  Found StreamID x - SessionID: 0267214254
  Sat Apr  3 13:45:22.2819 2021  6  Stream: 0, StreamClient[0] with SessionID 0267214254 for dvbt
  Sat Apr  3 13:45:22.2819 2021  6  Stream: 0, Parsing transport parameters...
  Sat Apr  3 13:45:22.2819 2021  6  Stream: 0, New frequency requested, clearing old channel data...
  Sat Apr  3 13:45:22.2821 2021  7  Stream: 0, Parsing transport parameters (Finished)
  Sat Apr  3 13:45:22.2821 2021  7  Stream: 0, Found Streaming type: HTTP
  Sat Apr  3 13:45:22.2821 2021  6  Stream: 0, Updating frontend...
  Sat Apr  3 13:45:22.4336 2021  6  Stream: 0, Opened /dev/dvb/adapter0/frontend0 for Read/Write with fd: 10
  Sat Apr  3 13:45:22.4587 2021  6  Stream: 0, Start tuning process for DVB-T(2)...
  Sat Apr  3 13:45:22.4587 2021  6  Stream: 0, Waiting on lock...
  Sat Apr  3 13:45:23.4802 2021  6  Stream: 0, Not locked yet   (FE status 0x0)...
  Sat Apr  3 13:45:23.5404 2021  6  Stream: 0, Not locked yet   (FE status 0x0)...
  Sat Apr  3 13:45:23.6006 2021  6  Stream: 0, Not locked yet   (FE status 0x0)...
  Sat Apr  3 13:45:23.6608 2021  6  Stream: 0, Not locked yet   (FE status 0x0)...
  Sat Apr  3 13:45:23.7218 2021  6  Stream: 0, Tuned and locked (FE status 0x1F)
  Sat Apr  3 13:45:23.7219 2021  6  Stream: 0, Updating PID filters...
  Sat Apr  3 13:45:23.7283 2021  6  Stream: 0, Set DMX buffer size to 18874368 Bytes
  Sat Apr  3 13:45:23.7286 2021  6  Stream: 0, Opened /dev/dvb/adapter0/demux0 fd: 11
  Sat Apr  3 13:45:23.7286 2021  7  Stream: 0, Set filter PID: 0000
  Sat Apr  3 13:45:23.7288 2021  7  Stream: 0, Set filter PID: 0001
  Sat Apr  3 13:45:23.7289 2021  7  Stream: 0, Set filter PID: 0016
[... cut ...]
  Sat Apr  3 13:45:23.7299 2021  7  Stream: 0, Set filter PID: 0302
  Sat Apr  3 13:45:23.7301 2021  7  Stream: 0, Set filter PID: 0303
  Sat Apr  3 13:45:23.7302 2021  7  Stream: 0, Set filter PID: 0304
  Sat Apr  3 13:45:23.7303 2021  7  Stream: 0, Set filter PID: 0311
  Sat Apr  3 13:45:23.7336 2021  7  Stream: 0, Updating frontend (Finished)
  Sat Apr  3 13:45:23.7337 2021  6  Stream: 0, HTTP set network buffer size: 25637 KBytes
  Sat Apr  3 13:45:23.7340 2021  6  Stream: 0, Start HTTP stream to 192.168.0.23:50436

petrkutalek, 2021-04-05:
I have just tested version `SatPI version: 1.5.415~g8fe87fc`. 

`./satpi --app-data-path . --http-path web --http-port 8875 --rtsp-port 8554 --no-daemon`
```
  Mon Apr  5 12:42:36.0777 2021  6  --- Starting SatPI version: 1.5.415~g8fe87fc ---
  Mon Apr  5 12:42:36.0786 2021  6  Number of processors online: 1
  Mon Apr  5 12:42:36.0788 2021  6  Default network buffer size: 88 KBytes
  Mon Apr  5 12:42:36.0797 2021  6  usb0: 192.168.219.1 [d8:ce:3a:06:84:69]
  Mon Apr  5 12:42:36.0803 2021  6  Current DVB_API_VERSION: 5.11
  Mon Apr  5 12:42:36.0804 2021  6  Enumerating all devices...
  Mon Apr  5 12:42:36.0805 2021  6  Detecting frontends in: /dev/dvb
  Mon Apr  5 12:42:36.0842 2021  6  Frontend Name: Sony CXD2880
  Mon Apr  5 12:42:36.0843 2021  6  Frontend Type: Terrestrial (DVB-T)
  Mon Apr  5 12:42:36.0844 2021  6  Frontend Type: Terrestrial (DVB-T2)
  Mon Apr  5 12:42:36.0844 2021  6  Frontend Freq: 174000000 Hz to 862000000 Hz
  Mon Apr  5 12:42:36.0845 2021  6  Frontend srat: 0 symbols/s to 0 symbols/s
  Mon Apr  5 12:42:36.0849 2021  6  Frontends found: 1
  Mon Apr  5 12:42:36.0849 2021  6  Setting up TS Reader using path: .
  Mon Apr  5 12:42:36.0867 2021  6  Setting up TS Streamer
  Mon Apr  5 12:42:36.0938 2021  6  Setting HTTP Port to: 8875
  Mon Apr  5 12:42:36.0939 2021  6  Setting RTSP Port to: 8554
  Mon Apr  5 12:42:36.0940 2021  6  Setting WEB Path to: web
  Mon Apr  5 12:42:36.0941 2021  6  Setting App Data Path to: .
  Mon Apr  5 12:42:36.1135 2021  6  Setting up RTSP server
  Mon Apr  5 12:42:36.1139 2021  6  Setting up HTTP server
  Mon Apr  5 12:42:36.1250 2021  6  Setting up SSDP server with BOOTID: 4  annouce interval: 60 Sec
  Mon Apr  5 12:42:42.9845 2021  6  HTTP Connection from 192.168.219.2 Port 61980 with fd: 8
  Mon Apr  5 12:42:43.1706 2021  6  HTTP Connection from 192.168.219.2 Port 61982 with fd: 9
  Mon Apr  5 12:42:43.1725 2021  6  HTTP Connection from 192.168.219.2 Port 61984 with fd: 10
  Mon Apr  5 12:42:43.1878 2021  6  HTTP Connection from 192.168.219.2 Port 61986 with fd: 11
  Mon Apr  5 12:42:43.1907 2021  6  HTTP Connection from 192.168.219.2 Port 61988 with fd: 12
  Mon Apr  5 12:42:43.2180 2021  6  HTTP Connection from 192.168.219.2 Port 61990 with fd: 13
  Mon Apr  5 12:42:54.7479 2021  6  HTTP Connection from 192.168.219.2 Port 61992 with fd: 14
  Mon Apr  5 12:42:54.7483 2021  7  HTTP Stream data from client None with IP 192.168.219.2 on Port 61992: GET /?freq=514&msys=dvbt2&pids=0,2110,2120,2121,2122,2123,2130,2150,2160 HTTP/1.1
  Mon Apr  5 12:42:54.7483 2021  7  Host: 192.168.219.1:8875
  Mon Apr  5 12:42:54.7483 2021  7  User-Agent: curl/7.76.0
  Mon Apr  5 12:42:54.7483 2021  7  Accept: */*
  Mon Apr  5 12:42:54.7483 2021  7  Accept-Encoding: deflate, gzip, zstd
  Mon Apr  5 12:42:54.7483 2021  7  --- LINE END ---
  Mon Apr  5 12:42:54.7494 2021  6  Found StreamID x - SessionID: 0247229088
  Mon Apr  5 12:42:54.7496 2021  6  Stream: 0, StreamClient[0] with SessionID 0247229088 for dvbt2
  Mon Apr  5 12:42:54.7497 2021  6  Stream: 0, Parsing transport parameters...
  Mon Apr  5 12:42:54.7499 2021  6  Stream: 0, New frequency requested, clearing old channel data...
  Mon Apr  5 12:42:54.7523 2021  7  Stream: 0, Parsing transport parameters (Finished)
  Mon Apr  5 12:42:54.7524 2021  7  Stream: 0, Found Streaming type: HTTP
  Mon Apr  5 12:42:54.7525 2021  6  Stream: 0, Updating frontend...
  Mon Apr  5 12:42:54.9121 2021  6  Stream: 0, Opened /dev/dvb/adapter0/frontend0 for Read/Write with fd: 15
  Mon Apr  5 12:42:54.9374 2021  6  Stream: 0, Start tuning process for DVB-T(2)...
  Mon Apr  5 12:42:54.9389 2021  6  Stream: 0, Waiting on lock...
  Mon Apr  5 12:42:54.9390 2021  6  Stream: 0, Not locked yet   (FE status 0x0)...
  Mon Apr  5 12:42:55.9746 2021  6  Stream: 0, Not locked yet   (FE status 0x0)...
  Mon Apr  5 12:42:56.0350 2021  6  Stream: 0, Not locked yet   (FE status 0x0)...
  Mon Apr  5 12:42:56.0955 2021  6  Stream: 0, Not locked yet   (FE status 0x0)...
  Mon Apr  5 12:42:56.1560 2021  6  Stream: 0, Not locked yet   (FE status 0x0)...
  Mon Apr  5 12:42:56.2165 2021  6  Stream: 0, Not locked yet   (FE status 0x0)...
  Mon Apr  5 12:42:56.2771 2021  6  Stream: 0, Not locked yet   (FE status 0x0)...
  Mon Apr  5 12:42:56.3376 2021  6  Stream: 0, Not locked yet   (FE status 0x0)...
  Mon Apr  5 12:42:56.3980 2021  6  Stream: 0, Not locked yet   (FE status 0x0)...
❌Mon Apr  5 12:42:56.4585 2021  6  Stream: 0, Not locked yet   (FE status 0x0)...
  Mon Apr  5 12:42:56.5188 2021  6  Stream: 0, Updating frontend (Failed)
  Mon Apr  5 12:42:56.5192 2021  7  HTTP/1.1 200 OK
  Mon Apr  5 12:42:56.5192 2021  7  Server: SatPI WebServer v0.1
  Mon Apr  5 12:42:56.5192 2021  7  Location: 
  Mon Apr  5 12:42:56.5192 2021  7  CSeq: 0
  Mon Apr  5 12:42:56.5192 2021  7  cache-control: no-cache
  Mon Apr  5 12:42:56.5192 2021  7  Content-Type: video/MP2T
  Mon Apr  5 12:42:56.5192 2021  7  --- LINE END ---
  Mon Apr  5 12:43:04.6941 2021  6  HTTP Client 192.168.219.2:61992 Connection closed with fd: 14
  Mon Apr  5 12:43:05.0346 2021  6  Stream: 0, Watchdog kicked in for StreamClient[0] with SessionID 0247229088
  Mon Apr  5 12:43:05.0348 2021  6  Stream: 0, Teardown StreamClient[0] with SessionID 0247229088
[... 85 lines cut ...]
  Mon Apr  5 12:43:10.2924 2021  6  Stream: 0,  serviceID: 0x410B - 16651  EIT: 0xFF  NetworkName: CESKA TELEVIZE  ChannelName: CRo RADIOZURNAL T2
[... cut ...]
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 91.6M    0 91.6M    0     0   565k      0 --:--:--  0:02:46 --:--:--  523k

```
I am happy it works (most of the time). Thanks a lot @robbieb43! 🎉 Nevertheless, as you can see, sometimes it does not tune and lock the FE. I have a perfect signal quality here, all TV report 100 % strength and 100 % quality, even `dvblast` works all right.

Barracuda09, 2021-04-05:
@petrkutalek 

Thanks for trying. I see indeed it thanks a long time to lock the FE. The default checks are set to 10 and can be increased to 15
 
[image]

petrkutalek, 2021-04-05:
🤔 I will try to modify _Wait On Lock Retry_ parameter and let you know.

petrkutalek, 2021-04-10:
Well, only once/twice I had to retry to request tuning process. The value 15 seams OK. However if you could allow to select up to 20, it would be nice. Thanks a lot. Well done.

Barracuda09, 2021-04-10:
@petrkutalek 

Thanks for giving this a try! I can increase it, but I thinks the signal is maybe not as good to the TV HAT if it takes that many tries.

Barracuda09, 2022-09-22:
@petrkutalek 

Is this still an issue, or can this be closed.