Barracuda09/SATPI #127: No picture with Sundtek SkyTV Ultimate Dual

nikolauzi11, 2021-03-28:
Hello,
I would like to install a SATPI server in a VM. For this I have a Sundtek SkyTV Ultimate Dual USB Tuner passed through to the VM. SATPI is running, web interface is there, but I get no picture. I have a simple DiseqC 1.0 switch and for testing I use DVBviewer Lite. Can someone please help me to solve the problem?

[log.txt](https://github.com/Barracuda09/SATPI/files/6217725/log.txt)

Thanks

Barracuda09, 2021-03-29:
Hi @nikolauzi11 

Thanks for giving this a try

It is strange, because it locks to a signal. (Why is DiSEqC send that many times?)
```
Sun Mar 28 18:28:00.0486 2021  6  Stream: 0, Opened /dev/dvb/adapter0/frontend0 for Read/Write with fd: 11
  Sun Mar 28 18:28:00.0487 2021  6  Stream: 0, Start tuning process for DVB-S(2)...
  Sun Mar 28 18:28:00.0487 2021  6  Stream: 0, Opened /dev/dvb/adapter0/frontend0 for Writing DiSEqC command with fd: 11
  Sun Mar 28 18:28:00.0928 2021  6  Stream: 0, Sending DiSEqC: [e0] [10] [38] [f3] - DiSEqC Src: 0
  Sun Mar 28 18:28:00.5807 2021  6  Stream: 0, Sending DiSEqC: [e1] [10] [38] [f3] - DiSEqC Src: 0
  Sun Mar 28 18:28:01.0465 2021  6  Stream: 0, Sending DiSEqC: [e1] [10] [38] [f3] - DiSEqC Src: 0
  Sun Mar 28 18:28:01.5138 2021  6  Stream: 0, Sending DiSEqC: [e1] [10] [38] [f3] - DiSEqC Src: 0
  Sun Mar 28 18:28:01.9991 2021  6  Stream: 0, Sending DiSEqC: [e1] [10] [38] [f3] - DiSEqC Src: 0
  Sun Mar 28 18:28:02.4641 2021  6  Stream: 0, Sending DiSEqC: [e1] [10] [38] [f3] - DiSEqC Src: 0
  Sun Mar 28 18:28:02.9462 2021  6  Stream: 0, Sending DiSEqC: [e1] [10] [38] [f3] - DiSEqC Src: 0
  Sun Mar 28 18:28:03.4162 2021  6  Stream: 0, Sending DiSEqC: [e1] [10] [38] [f3] - DiSEqC Src: 0
  Sun Mar 28 18:28:03.8973 2021  6  Stream: 0, Sending DiSEqC: [e1] [10] [38] [f3] - DiSEqC Src: 0
  Sun Mar 28 18:28:04.3853 2021  6  Stream: 0, Sending DiSEqC: [e1] [10] [38] [f3] - DiSEqC Src: 0
  Sun Mar 28 18:28:04.8254 2021  7  Stream: 0, Set Properties: Frequency 1237000
  Sun Mar 28 18:28:05.2407 2021  6  Stream: 0, Waiting on lock...
  Sun Mar 28 18:28:05.2603 2021  6  Stream: 0, Not locked yet   (FE status 0x0)...
  Sun Mar 28 18:28:05.5880 2021  6  Stream: 0, Tuned and locked (FE status 0x1F)
  Sun Mar 28 18:28:05.5883 2021  6  Stream: 0, Updating PID filters...
  Sun Mar 28 18:28:05.5921 2021  6  Stream: 0, Set DMX buffer size to 18874368 Bytes
  Sun Mar 28 18:28:05.6214 2021  6  Stream: 0, Opened /dev/dvb/adapter0/demux0 fd: 13
  Sun Mar 28 18:28:05.6216 2021  7  Stream: 0, Set filter PID: 0000
  Sun Mar 28 18:28:05.6217 2021  7  Stream: 0, Set filter PID: 0018
```
But is does not receive data on the requested PIDS
```
Sun Mar 28 18:28:13.4584 2021  6  HTTP Connection from 192.168.112.61 Port 59840 with fd: 18
  Sun Mar 28 18:28:13.4660 2021  6  HTTP Connection from 192.168.112.61 Port 59841 with fd: 19
  Sun Mar 28 18:28:13.4664 2021  6  HTTP Connection from 192.168.112.61 Port 59842 with fd: 20
  Sun Mar 28 18:28:18.4675 2021  6  HTTP Client 192.168.112.61:59842 Connection closed with fd: 20
  Sun Mar 28 18:28:47.2379 2021  7  RTSP Stream data from client None with IP 192.168.112.61 on Port 59763: PLAY rtsp://192.168.112.66:554/stream=0?delpids=101,106,0,100,104 RTSP/1.0
  Sun Mar 28 18:28:47.2379 2021  7  CSeq: 10
  Sun Mar 28 18:28:47.2379 2021  7  Session: 0272220533
  Sun Mar 28 18:28:47.2379 2021  7  --- LINE END ---
  Sun Mar 28 18:28:47.2379 2021  6  Found StreamID 0 - SessionID 0272220533
  Sun Mar 28 18:28:47.2379 2021  6  Stream: 0, StreamClient[0] with SessionID 0272220533
  Sun Mar 28 18:28:47.2379 2021  6  Stream: 0, Parsing transport parameters...
  Sun Mar 28 18:28:47.2379 2021  7  Stream: 0, Parsing transport parameters (Finished)
  Sun Mar 28 18:28:47.2379 2021  6  Stream: 0, Updating frontend...
  Sun Mar 28 18:28:47.2380 2021  6  Stream: 0, Updating PID filters...
  Sun Mar 28 18:28:47.2380 2021  7  Stream: 0, Remove filter PID: 0000 - Packet Count: 0
  Sun Mar 28 18:28:47.2380 2021  7  Stream: 0, Remove filter PID: 0100 - Packet Count: 0
  Sun Mar 28 18:28:47.2380 2021  7  Stream: 0, Remove filter PID: 0101 - Packet Count: 0
  Sun Mar 28 18:28:47.2380 2021  7  Stream: 0, Remove filter PID: 0104 - Packet Count: 0
  Sun Mar 28 18:28:47.2380 2021  7  Stream: 0, Remove filter PID: 0106 - Packet Count: 0
  Sun Mar 28 18:28:47.2385 2021  7  Stream: 0, Updating frontend (Finished)
```

Sorry, I can not tell way or were the problem is at the moment

nikolauzi11, 2021-03-29:
Hello @Barracuda09 

Thanks for looking at it.

> Sorry, I can not tell way or were the problem is at the moment

I'm sorry to hear that.
Is there anything I can do to help solve the problem?

Barracuda09, 2021-03-29:
Hi @nikolauzi11 

Could you look at the 'Frontend Overview' to see if there are packets send?
[image]

Could you try VLC and construct an valid channel request?

Barracuda09, 2021-03-29:
Hi @nikolauzi11 

Could you try with new commit? 65cad5b

nikolauzi11, 2021-03-30:
Hi @Barracuda09 

I tried your last commit, but unfortunately it does not work yet.
I used VLC with following stream address:

[rtsp://satip/?src=1&freq=11494&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=0,17,18,5100,5101,5102,5104](url)

The packet counter remains at 0.

`  Tue Mar 30 12:25:08.9048 2021  6  --- Starting SatPI version: 1.5.410~g65cad5b ---
  Tue Mar 30 12:25:08.9048 2021  6  Number of processors online: 1
  Tue Mar 30 12:25:08.9048 2021  6  Default network buffer size: 104 KBytes
  Tue Mar 30 12:25:08.9049 2021  6  ens18: 192.168.112.66 [86:03:51:15:4f:89]
  Tue Mar 30 12:25:08.9049 2021  6  Current DVB_API_VERSION: 5.11
  Tue Mar 30 12:25:08.9049 2021  6  Enumerating all devices...
  Tue Mar 30 12:25:08.9049 2021  6  Detecting frontends in: /dev/dvb
  Tue Mar 30 12:25:08.9093 2021  6  Frontend Name: Sundtek DVB-S/S2 (VIII)
  Tue Mar 30 12:25:08.9094 2021  6  Frontend Type: Satellite (DVB-S2)
  Tue Mar 30 12:25:08.9094 2021  6  Frontend Type: Satellite (DVB-S)
  Tue Mar 30 12:25:08.9094 2021  6  Frontend Freq: 925000 Hz to 2175000 Hz
  Tue Mar 30 12:25:08.9094 2021  6  Frontend srat: 1000000 symbols/s to 45000000 symbols/s
  Tue Mar 30 12:25:08.9121 2021  6  Frontend Name: Sundtek DVB-S/S2 (VIII)
  Tue Mar 30 12:25:08.9122 2021  6  Frontend Type: Satellite (DVB-S2)
  Tue Mar 30 12:25:08.9122 2021  6  Frontend Type: Satellite (DVB-S)
  Tue Mar 30 12:25:08.9122 2021  6  Frontend Freq: 925000 Hz to 2175000 Hz
  Tue Mar 30 12:25:08.9122 2021  6  Frontend srat: 1000000 symbols/s to 45000000 symbols/s
  Tue Mar 30 12:25:08.9122 2021  6  Frontends found: 2
  Tue Mar 30 12:25:08.9122 2021  6  Setting up TS Reader using path: /var/lib/satpi
  Tue Mar 30 12:25:08.9123 2021  6  Setting up TS Streamer
  Tue Mar 30 12:25:08.9131 2021  6  Setting HTTP Port to: 8875
  Tue Mar 30 12:25:08.9131 2021  6  Setting RTSP Port to: 554
  Tue Mar 30 12:25:08.9131 2021  6  Setting WEB Path to: /usr/share/satpi/web
  Tue Mar 30 12:25:08.9131 2021  6  Setting App Data Path to: /var/lib/satpi
  Tue Mar 30 12:25:08.9146 2021  6  Setting up SSDP server with BOOTID: 15  annouce interval: 60 Sec
  Tue Mar 30 12:25:08.9148 2021  6  Setting up RTSP server
  Tue Mar 30 12:25:08.9148 2021  6  Setting up HTTP server
  Tue Mar 30 12:25:09.0987 2021  6  HTTP Connection from 192.168.112.61 Port 52055 with fd: 8
  Tue Mar 30 12:25:09.3439 2021  6  HTTP Connection from 192.168.112.61 Port 52057 with fd: 9
  Tue Mar 30 12:25:10.2339 2021  6  HTTP Connection from 192.168.112.61 Port 52062 with fd: 10
  Tue Mar 30 12:25:10.2343 2021  7  Client: 192.168.112.61 requested desc.xml
  Tue Mar 30 12:25:11.9226 2021  6  HTTP Connection from 192.168.112.20 Port 51407 with fd: 11
  Tue Mar 30 12:25:11.9227 2021  7  Client: 192.168.112.20 requested desc.xml
  Tue Mar 30 12:25:11.9594 2021  6  HTTP Connection from 192.168.112.61 Port 52067 with fd: 12
  Tue Mar 30 12:25:11.9596 2021  7  Client: 192.168.112.61 requested desc.xml
  Tue Mar 30 12:25:11.9718 2021  6  HTTP Connection from 192.168.112.68 Port 62533 with fd: 13
  Tue Mar 30 12:25:11.9722 2021  7  Client: 192.168.112.68 requested desc.xml
  Tue Mar 30 12:25:14.3510 2021  6  HTTP Client 192.168.112.61:52057 Connection closed with fd: 9
  Tue Mar 30 12:25:25.6249 2021  6  RTSP Connection from 192.168.112.61 Port 52099 with fd: 9
  Tue Mar 30 12:25:25.6249 2021  7  RTSP Stream data from client None with IP 192.168.112.61 on Port 52099: OPTIONS rtsp://satip:554/?src=1&freq=11494&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=0,17,18,5100,5101,5102,5104 RTSP/1.0
  Tue Mar 30 12:25:25.6249 2021  7  CSeq: 2
  Tue Mar 30 12:25:25.6249 2021  7  User-Agent: LibVLC/3.0.12 (LIVE555 Streaming Media v2016.11.28)
  Tue Mar 30 12:25:25.6249 2021  7  --- LINE END ---
  Tue Mar 30 12:25:25.6249 2021  7  RTSP/1.0 200 OK
  Tue Mar 30 12:25:25.6249 2021  7  CSeq: 2
  Tue Mar 30 12:25:25.6249 2021  7  Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
  Tue Mar 30 12:25:25.6249 2021  7  --- LINE END ---
  Tue Mar 30 12:25:25.6310 2021  7  RTSP Stream data from client None with IP 192.168.112.61 on Port 52099: DESCRIBE rtsp://satip:554/?src=1&freq=11494&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=0,17,18,5100,5101,5102,5104 RTSP/1.0
  Tue Mar 30 12:25:25.6310 2021  7  CSeq: 3
  Tue Mar 30 12:25:25.6310 2021  7  User-Agent: LibVLC/3.0.12 (LIVE555 Streaming Media v2016.11.28)
  Tue Mar 30 12:25:25.6310 2021  7  Accept: application/sdp
  Tue Mar 30 12:25:25.6310 2021  7  --- LINE END ---
  Tue Mar 30 12:25:25.6310 2021  7  RTSP/1.0 404 Not Found
  Tue Mar 30 12:25:25.6310 2021  7  CSeq: 3
  Tue Mar 30 12:25:25.6310 2021  7  Content-Type: application/sdp
  Tue Mar 30 12:25:25.6310 2021  7  Content-Base: rtsp://192.168.112.66/
  Tue Mar 30 12:25:25.6310 2021  7  Content-Length: 0
  Tue Mar 30 12:25:25.6310 2021  7  --- LINE END ---
  Tue Mar 30 12:25:25.6339 2021  6  RTSP Client 192.168.112.61:52099 Connection closed with fd: 9
  Tue Mar 30 12:25:25.6365 2021  6  RTSP Connection from 192.168.112.61 Port 52100 with fd: 9
  Tue Mar 30 12:25:25.6442 2021  7  RTSP Stream data from client None with IP 192.168.112.61 on Port 52100: SETUP rtsp://satip:554/?src=1&freq=11494&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=0,17,18,5100,5101,5102,5104 RTSP/1.0
[... 59 lines cut ...]
  Tue Mar 30 12:25:29.2925 2021  6  Stream: 0, StreamClient[0] with SessionID 0269270754
[... cut ...]
  Tue Mar 30 12:25:29.6938 2021  7  --- LINE END ---
  Tue Mar 30 12:25:29.8495 2021  6  RTSP Client 192.168.112.61:52100 Connection closed with fd: 9
  Tue Mar 30 12:26:10.2602 2021  6  HTTP Client 192.168.112.61:52062 Connection closed with fd: 10
  Tue Mar 30 12:26:11.9518 2021  6  HTTP Client 192.168.112.20:51407 Connection closed with fd: 11
  Tue Mar 30 12:26:11.9738 2021  6  HTTP Client 192.168.112.61:52067 Connection closed with fd: 12
  Tue Mar 30 12:26:11.9894 2021  6  HTTP Client 192.168.112.68:62533 Connection closed with fd: 13
`

Barracuda09, 2021-03-30:
Hi @nikolauzi11 

Ok thanks for trying, could you try it outside a VM?

nikolauzi11, 2021-03-30:
Hi @Barracuda09 

Thanks for your effort.
I will try on physical hardware, but I will need a day or two.

nikolauzi11, 2021-03-31:
Hi @Barracuda09 

The result is exactly the same on physical hardware.
I used an older Dell PC.

Barracuda09, 2021-03-31:
Hello @nikolauzi11 

Many thanks for trying and your effort as well. 

I am still puzzled by why it is not receiving PID data, as is it seems to tune correctly. Maybe it has to do with how the PIDS are requested in SatPI. I will try to setup my Raspberry Pi with DVB-C/T2 stick again, and see what happens with that. I have SatPI running at the moment only on Vu+ and GigaBlue receivers for both DVB-C and DVB-S2

https://github.com/Barracuda09/SATPI/blob/65cad5b44893502226bbd40ca0c3bb61dafda419/src/input/dvb/Frontend.cpp#L734-L748

Barracuda09, 2021-04-01:
Hi @nikolauzi11 

I assume you have the latest drivers for the SundTek

nikolauzi11, 2021-04-01:
Hi @Barracuda09 

Yes, everything is up to date.
I just testet it on an Raspberry Pi 3 B+ with the same result.

Barracuda09, 2021-04-01:
@nikolauzi11 some questions
- does the Sundtek tool work (with scanning)?
- are both channels affected?

nikolauzi11, 2021-04-01:
Hi @Barracuda09 

- I have brought some time to get the Sundtek streaming server running, but yes it works. I have a stream in VLC.
- Yes, both channels are affected

Barracuda09, 2021-04-01:
Hi @nikolauzi11 

Thanks for your feedback. I have to try this out on my Raspberry with DVB-C/T2 stick and see what happens

Barracuda09, 2021-04-01:
Hi @nikolauzi11 

I think I messed up somewhere, it does not work here as well. An older version 1.5.183~geffd715 seems to work.

Have to check what I changed

Barracuda09, 2021-04-03:
Hi @nikolauzi11 

First of all thanks for your effort so far.

Could you please try out the last commit and see if it makes any difference?

Thanks

Marc

nikolauzi11, 2021-04-03:
Hi @Barracuda09 

> First of all thanks for your effort so far.

This is not a problem at all, I will be very happy to help you solve the problem.

However, the problem is unfortunately still there.

```  Sat Apr  3 17:51:54.1077 2021  6  --- Starting SatPI version: 1.5.414~g9d9fa22 ---
  Sat Apr  3 17:51:54.1077 2021  6  Number of processors online: 1
  Sat Apr  3 17:51:54.1077 2021  6  Default network buffer size: 104 KBytes
  Sat Apr  3 17:51:54.1078 2021  6  ens18: 192.168.112.66 [86:03:51:15:4f:89]
  Sat Apr  3 17:51:54.1078 2021  6  Current DVB_API_VERSION: 5.11
  Sat Apr  3 17:51:54.1078 2021  6  Enumerating all devices...
  Sat Apr  3 17:51:54.1078 2021  6  Detecting frontends in: /dev/dvb
  Sat Apr  3 17:51:55.1662 2021  6  Frontend Name: Sundtek DVB-S/S2 (VIII)
  Sat Apr  3 17:51:55.1663 2021  6  Frontend Type: Satellite (DVB-S2)
  Sat Apr  3 17:51:55.1663 2021  6  Frontend Type: Satellite (DVB-S)
  Sat Apr  3 17:51:55.1663 2021  6  Frontend Freq: 925000 Hz to 2175000 Hz
  Sat Apr  3 17:51:55.1663 2021  6  Frontend srat: 1000000 symbols/s to 45000000 symbols/s
  Sat Apr  3 17:51:56.2275 2021  6  Frontend Name: Sundtek DVB-S/S2 (VIII)
  Sat Apr  3 17:51:56.2275 2021  6  Frontend Type: Satellite (DVB-S2)
  Sat Apr  3 17:51:56.2275 2021  6  Frontend Type: Satellite (DVB-S)
  Sat Apr  3 17:51:56.2275 2021  6  Frontend Freq: 925000 Hz to 2175000 Hz
  Sat Apr  3 17:51:56.2275 2021  6  Frontend srat: 1000000 symbols/s to 45000000 symbols/s
  Sat Apr  3 17:51:56.2276 2021  6  Frontends found: 2
  Sat Apr  3 17:51:56.2276 2021  6  Setting up TS Reader using path: /var/lib/satpi
  Sat Apr  3 17:51:56.2277 2021  6  Setting up TS Streamer
  Sat Apr  3 17:51:56.2286 2021  6  Setting HTTP Port to: 8875
  Sat Apr  3 17:51:56.2286 2021  6  Setting RTSP Port to: 554
  Sat Apr  3 17:51:56.2286 2021  6  Setting WEB Path to: /usr/share/satpi/web
  Sat Apr  3 17:51:56.2286 2021  6  Setting App Data Path to: /var/lib/satpi
  Sat Apr  3 17:51:56.2313 2021  6  Setting up SSDP server with BOOTID: 18  annouce interval: 60 Sec
  Sat Apr  3 17:51:56.2315 2021  6  Setting up RTSP server
  Sat Apr  3 17:51:56.2315 2021  6  Setting up HTTP server
  Sat Apr  3 17:51:56.2673 2021  6  HTTP Connection from 192.168.112.61 Port 55173 with fd: 8
  Sat Apr  3 17:51:56.2944 2021  6  HTTP Connection from 192.168.112.61 Port 55171 with fd: 9
  Sat Apr  3 17:51:56.5553 2021  6  HTTP Connection from 192.168.112.61 Port 55172 with fd: 10
  Sat Apr  3 17:52:01.2996 2021  6  HTTP Client 192.168.112.61:55171 Connection closed with fd: 9
  Sat Apr  3 17:52:02.3094 2021  6  HTTP Client 192.168.112.61:55172 Connection closed with fd: 10
  Sat Apr  3 17:52:04.3496 2021  6  HTTP Connection from 192.168.112.61 Port 55175 with fd: 9
  Sat Apr  3 17:52:04.3496 2021  6  HTTP Connection from 192.168.112.61 Port 55176 with fd: 10
  Sat Apr  3 17:52:04.3503 2021  6  HTTP Connection from 192.168.112.61 Port 55177 with fd: 11
  Sat Apr  3 17:52:04.3504 2021  6  HTTP Connection from 192.168.112.61 Port 55178 with fd: 12
  Sat Apr  3 17:52:04.3504 2021  6  HTTP Connection from 192.168.112.61 Port 55179 with fd: 13
  Sat Apr  3 17:52:33.5620 2021  6  RTSP Connection from 192.168.112.61 Port 55181 with fd: 14
  Sat Apr  3 17:52:33.5620 2021  7  RTSP Stream data from client None with IP 192.168.112.61 on Port 55181: OPTIONS rtsp://satip:554/?src=1&freq=11494&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=0,17,18,5100,5101,5102,5104 RTSP/1.0
  Sat Apr  3 17:52:33.5620 2021  7  CSeq: 2
  Sat Apr  3 17:52:33.5620 2021  7  User-Agent: LibVLC/3.0.12 (LIVE555 Streaming Media v2016.11.28)
  Sat Apr  3 17:52:33.5620 2021  7  --- LINE END ---
  Sat Apr  3 17:52:33.5621 2021  7  RTSP/1.0 200 OK
  Sat Apr  3 17:52:33.5621 2021  7  CSeq: 2
  Sat Apr  3 17:52:33.5621 2021  7  Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
  Sat Apr  3 17:52:33.5621 2021  7  --- LINE END ---
  Sat Apr  3 17:52:33.5673 2021  7  RTSP Stream data from client None with IP 192.168.112.61 on Port 55181: DESCRIBE rtsp://satip:554/?src=1&freq=11494&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=0,17,18,5100,5101,5102,5104 RTSP/1.0
  Sat Apr  3 17:52:33.5673 2021  7  CSeq: 3
  Sat Apr  3 17:52:33.5673 2021  7  User-Agent: LibVLC/3.0.12 (LIVE555 Streaming Media v2016.11.28)
  Sat Apr  3 17:52:33.5673 2021  7  Accept: application/sdp
  Sat Apr  3 17:52:33.5673 2021  7  --- LINE END ---
  Sat Apr  3 17:52:33.5674 2021  7  RTSP/1.0 404 Not Found
  Sat Apr  3 17:52:33.5674 2021  7  CSeq: 3
  Sat Apr  3 17:52:33.5674 2021  7  Content-Type: application/sdp
  Sat Apr  3 17:52:33.5674 2021  7  Content-Base: rtsp://192.168.112.66/
  Sat Apr  3 17:52:33.5674 2021  7  Content-Length: 0
  Sat Apr  3 17:52:33.5674 2021  7  --- LINE END ---
  Sat Apr  3 17:52:33.5709 2021  6  RTSP Client 192.168.112.61:55181 Connection closed with fd: 14
  Sat Apr  3 17:52:33.5750 2021  6  RTSP Connection from 192.168.112.61 Port 55182 with fd: 14
  Sat Apr  3 17:52:33.5895 2021  7  RTSP Stream data from client None with IP 192.168.112.61 on Port 55182: SETUP rtsp://satip:554/?src=1&freq=11494&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=0,17,18,5100,5101,5102,5104 RTSP/1.0
  Sat Apr  3 17:52:33.5895 2021  7  CSeq: 0
  Sat Apr  3 17:52:33.5895 2021  7  Transport: RTP/AVP;unicast;client_port=9206-9207
[... 52 lines cut ...]
  Sat Apr  3 17:52:38.3324 2021  7  CSeq: 2
  Sat Apr  3 17:52:38.3324 2021  7  Session: 0266233894
[... cut ...]
  Sat Apr  3 17:52:38.6332 2021  7  Stream: 0, Remove filter PID: 5104 - Packet Count: 0
  Sat Apr  3 17:52:38.6333 2021  6  Stream: 0, Closing /dev/dvb/adapter0/frontend0 fd: 16
  Sat Apr  3 17:52:38.6333 2021  6  Stream: 0, Closing /dev/dvb/adapter0/demux0 fd: 18
  Sat Apr  3 17:52:38.6334 2021  7  RTSP/1.0 200 OK
  Sat Apr  3 17:52:38.6334 2021  7  CSeq: 2
  Sat Apr  3 17:52:38.6334 2021  7  Session: 0266233894
  Sat Apr  3 17:52:38.6334 2021  7  --- LINE END ---
  Sat Apr  3 17:52:38.7898 2021  6  RTSP Client 192.168.112.61:55182 Connection closed with fd: 14
```

Barracuda09, 2021-04-03:
Hi @nikolauzi11 

Could you make a debug version? with:

`make debug`

nikolauzi11, 2021-04-03:
Hi @Barracuda09 

Sure:

```  Sat Apr  3 19:01:06.4114 2021  6  [                                 src/main.cpp:382] --- Starting SatPI version: 1.5.414~g9d9fa22 ---
  Sat Apr  3 19:01:06.4119 2021  6  [                                 src/main.cpp:383] Number of processors online: 1
  Sat Apr  3 19:01:06.4120 2021  6  [                                 src/main.cpp:384] Default network buffer size: 104 KBytes
  Sat Apr  3 19:01:06.4126 2021  6  [                        src/InterfaceAttr.cpp:130] ens18: 192.168.112.66 [86:03:51:15:4f:89]
  Sat Apr  3 19:01:06.4127 2021  6  [                        src/StreamManager.cpp:070] Current DVB_API_VERSION: 5.11
  Sat Apr  3 19:01:06.4128 2021  6  [                        src/StreamManager.cpp:071] Enumerating all devices...
  Sat Apr  3 19:01:06.4128 2021  6  [                   src/input/dvb/Frontend.cpp:175] Detecting frontends in: /dev/dvb
  Sat Apr  3 19:01:07.4678 2021  6  [                   src/input/dvb/Frontend.cpp:484] Frontend Name: Sundtek DVB-S/S2 (VIII)
  Sat Apr  3 19:01:07.4683 2021  6  [                   src/input/dvb/Frontend.cpp:561] Frontend Type: Satellite (DVB-S2)
  Sat Apr  3 19:01:07.4684 2021  6  [                   src/input/dvb/Frontend.cpp:557] Frontend Type: Satellite (DVB-S)
  Sat Apr  3 19:01:07.4684 2021  6  [                   src/input/dvb/Frontend.cpp:603] Frontend Freq: 925000 Hz to 2175000 Hz
  Sat Apr  3 19:01:07.4684 2021  6  [                   src/input/dvb/Frontend.cpp:604] Frontend srat: 1000000 symbols/s to 45000000 symbols/s
  Sat Apr  3 19:01:08.5269 2021  6  [                   src/input/dvb/Frontend.cpp:484] Frontend Name: Sundtek DVB-S/S2 (VIII)
  Sat Apr  3 19:01:08.5274 2021  6  [                   src/input/dvb/Frontend.cpp:561] Frontend Type: Satellite (DVB-S2)
  Sat Apr  3 19:01:08.5274 2021  6  [                   src/input/dvb/Frontend.cpp:557] Frontend Type: Satellite (DVB-S)
  Sat Apr  3 19:01:08.5274 2021  6  [                   src/input/dvb/Frontend.cpp:603] Frontend Freq: 925000 Hz to 2175000 Hz
  Sat Apr  3 19:01:08.5274 2021  6  [                   src/input/dvb/Frontend.cpp:604] Frontend srat: 1000000 symbols/s to 45000000 symbols/s
  Sat Apr  3 19:01:08.5275 2021  6  [                   src/input/dvb/Frontend.cpp:178] Frontends found: 2
  Sat Apr  3 19:01:08.5275 2021  6  [                  src/input/file/TSReader.cpp:056] Setting up TS Reader using path: /var/lib/satpi
  Sat Apr  3 19:01:08.5277 2021  6  [                src/input/stream/Streamer.cpp:059] Setting up TS Streamer
  Sat Apr  3 19:01:08.5354 2021  6  [                           src/Properties.cpp:075] Setting HTTP Port to: 8875
  Sat Apr  3 19:01:08.5356 2021  6  [                           src/Properties.cpp:080] Setting RTSP Port to: 554
  Sat Apr  3 19:01:08.5357 2021  6  [                           src/Properties.cpp:084] Setting WEB Path to: /usr/share/satpi/web
  Sat Apr  3 19:01:08.5357 2021  6  [                           src/Properties.cpp:088] Setting App Data Path to: /var/lib/satpi
  Sat Apr  3 19:01:08.5396 2021  6  [                           src/HttpServer.cpp:058] Setting up HTTP server
  Sat Apr  3 19:01:08.5404 2021  6  [                           src/RtspServer.cpp:046] Setting up RTSP server
  Sat Apr  3 19:01:08.5407 2021  6  [                     src/upnp/ssdp/Server.cpp:104] Setting up SSDP server with BOOTID: 19  annouce interval: 60 Sec
  Sat Apr  3 19:01:08.5898 2021  6  [                    src/socket/SocketAttr.cpp:137] HTTP Connection from 192.168.112.61 Port 57861 with fd: 8
  Sat Apr  3 19:01:08.8475 2021  6  [                    src/socket/SocketAttr.cpp:137] HTTP Connection from 192.168.112.61 Port 57860 with fd: 9
  Sat Apr  3 19:01:12.9084 2021  6  [                    src/socket/SocketAttr.cpp:137] RTSP Connection from 192.168.112.61 Port 57862 with fd: 10
  Sat Apr  3 19:01:12.9085 2021  7  [                          src/HttpcServer.cpp:153] RTSP Stream data from client None with IP 192.168.112.61 on Port 57862: OPTIONS rtsp://satip:554/?src=1&freq=11494&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=0,17,18,5100,5101,5102,5104 RTSP/1.0
  Sat Apr  3 19:01:12.9085 2021  7  CSeq: 2
  Sat Apr  3 19:01:12.9085 2021  7  User-Agent: LibVLC/3.0.12 (LIVE555 Streaming Media v2016.11.28)
  Sat Apr  3 19:01:12.9085 2021  7  --- LINE END ---
  Sat Apr  3 19:01:12.9085 2021  7  [                          src/HttpcServer.cpp:213] RTSP/1.0 200 OK
  Sat Apr  3 19:01:12.9085 2021  7  CSeq: 2
  Sat Apr  3 19:01:12.9085 2021  7  Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
  Sat Apr  3 19:01:12.9085 2021  7  --- LINE END ---
  Sat Apr  3 19:01:12.9147 2021  7  [                          src/HttpcServer.cpp:153] RTSP Stream data from client None with IP 192.168.112.61 on Port 57862: DESCRIBE rtsp://satip:554/?src=1&freq=11494&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=0,17,18,5100,5101,5102,5104 RTSP/1.0
  Sat Apr  3 19:01:12.9147 2021  7  CSeq: 3
  Sat Apr  3 19:01:12.9147 2021  7  User-Agent: LibVLC/3.0.12 (LIVE555 Streaming Media v2016.11.28)
  Sat Apr  3 19:01:12.9147 2021  7  Accept: application/sdp
  Sat Apr  3 19:01:12.9147 2021  7  --- LINE END ---
  Sat Apr  3 19:01:12.9148 2021  7  [                          src/HttpcServer.cpp:213] RTSP/1.0 404 Not Found
  Sat Apr  3 19:01:12.9148 2021  7  CSeq: 3
  Sat Apr  3 19:01:12.9148 2021  7  Content-Type: application/sdp
  Sat Apr  3 19:01:12.9148 2021  7  Content-Base: rtsp://192.168.112.66/
[... cut ...]
  Sat Apr  3 19:01:16.9163 2021  6  [                   src/input/dvb/Frontend.cpp:664] Stream: 0, Closing /dev/dvb/adapter0/demux0 fd: 13
  Sat Apr  3 19:01:16.9164 2021  7  [                          src/HttpcServer.cpp:213] RTSP/1.0 200 OK
  Sat Apr  3 19:01:16.9164 2021  7  CSeq: 2
  Sat Apr  3 19:01:16.9164 2021  7  Session: 0252133192
  Sat Apr  3 19:01:16.9164 2021  7  --- LINE END ---
  Sat Apr  3 19:01:17.0715 2021  6  [                     src/socket/TcpSocket.cpp:106] RTSP Client 192.168.112.61:57863 Connection closed with fd: 10
```

Barracuda09, 2021-04-03:
Hi @nikolauzi11 

Which Satellite and channel is this?

nikolauzi11, 2021-04-03:
Hi @Barracuda09 

It's ASTRA 19.2E, Das Erste HD.

Barracuda09, 2021-04-03:
Hi @nikolauzi11 

Could you add after L754 and L755
```
	if (::ioctl(_fd_dmx, DMX_START) != 0) {
		PERROR("DMX_START");
	}

```
https://github.com/Barracuda09/SATPI/blob/9d9fa2202f25ce74c832fe74e5ead4d6951537fd/src/input/dvb/Frontend.cpp#L745-L760

Barracuda09, 2021-04-03:
So for clarity, it should become this:

```
          struct dmx_pes_filter_params pesFilter;
          pesFilter.pid      = pid;
          pesFilter.input    = DMX_IN_FRONTEND;
          pesFilter.output   = DMX_OUT_TSDEMUX_TAP;
          pesFilter.pes_type = DMX_PES_OTHER;
          pesFilter.flags    = DMX_IMMEDIATE_START;
          if (::ioctl(_fd_dmx, DMX_SET_PES_FILTER, &pesFilter) != 0) {
          PERROR("Stream: %d, DMX_SET_PES_FILTER (PID %04d)", _streamID, 0);
              return;
          }
          /////////////////////////////////// Added
          if (::ioctl(_fd_dmx, DMX_START) != 0) {
               PERROR("DMX_START");
          }
          /////////////////////////////////// Added
          SI_LOG_INFO("Stream: %d, Opened %s fd: %d", _streamID, _path_to_dmx.c_str(), _fd_dmx);

```

nikolauzi11, 2021-04-03:
Hi @Barracuda09 

Just tested it, but still no picture:

```  Sat Apr  3 20:52:09.2759 2021  6  [                                 src/main.cpp:382] --- Starting SatPI version: 1.5.414~g9d9fa22 ---
  Sat Apr  3 20:52:09.2759 2021  6  [                                 src/main.cpp:383] Number of processors online: 1
  Sat Apr  3 20:52:09.2759 2021  6  [                                 src/main.cpp:384] Default network buffer size: 104 KBytes
  Sat Apr  3 20:52:09.2760 2021  6  [                        src/InterfaceAttr.cpp:130] ens18: 192.168.112.66 [86:03:51:15:4f:89]
  Sat Apr  3 20:52:09.2760 2021  6  [                        src/StreamManager.cpp:070] Current DVB_API_VERSION: 5.11
  Sat Apr  3 20:52:09.2760 2021  6  [                        src/StreamManager.cpp:071] Enumerating all devices...
  Sat Apr  3 20:52:09.2760 2021  6  [                   src/input/dvb/Frontend.cpp:175] Detecting frontends in: /dev/dvb
  Sat Apr  3 20:52:10.3340 2021  6  [                   src/input/dvb/Frontend.cpp:484] Frontend Name: Sundtek DVB-S/S2 (VIII)
  Sat Apr  3 20:52:10.3345 2021  6  [                   src/input/dvb/Frontend.cpp:561] Frontend Type: Satellite (DVB-S2)
  Sat Apr  3 20:52:10.3346 2021  6  [                   src/input/dvb/Frontend.cpp:557] Frontend Type: Satellite (DVB-S)
  Sat Apr  3 20:52:10.3346 2021  6  [                   src/input/dvb/Frontend.cpp:603] Frontend Freq: 925000 Hz to 2175000 Hz
  Sat Apr  3 20:52:10.3346 2021  6  [                   src/input/dvb/Frontend.cpp:604] Frontend srat: 1000000 symbols/s to 45000000 symbols/s
  Sat Apr  3 20:52:11.3942 2021  6  [                   src/input/dvb/Frontend.cpp:484] Frontend Name: Sundtek DVB-S/S2 (VIII)
  Sat Apr  3 20:52:11.3947 2021  6  [                   src/input/dvb/Frontend.cpp:561] Frontend Type: Satellite (DVB-S2)
  Sat Apr  3 20:52:11.3947 2021  6  [                   src/input/dvb/Frontend.cpp:557] Frontend Type: Satellite (DVB-S)
  Sat Apr  3 20:52:11.3947 2021  6  [                   src/input/dvb/Frontend.cpp:603] Frontend Freq: 925000 Hz to 2175000 Hz
  Sat Apr  3 20:52:11.3947 2021  6  [                   src/input/dvb/Frontend.cpp:604] Frontend srat: 1000000 symbols/s to 45000000 symbols/s
  Sat Apr  3 20:52:11.3947 2021  6  [                   src/input/dvb/Frontend.cpp:178] Frontends found: 2
  Sat Apr  3 20:52:11.3947 2021  6  [                  src/input/file/TSReader.cpp:056] Setting up TS Reader using path: /var/lib/satpi
  Sat Apr  3 20:52:11.3950 2021  6  [                src/input/stream/Streamer.cpp:059] Setting up TS Streamer
  Sat Apr  3 20:52:11.3994 2021  6  [                           src/Properties.cpp:075] Setting HTTP Port to: 8875
  Sat Apr  3 20:52:11.3996 2021  6  [                           src/Properties.cpp:080] Setting RTSP Port to: 554
  Sat Apr  3 20:52:11.3996 2021  6  [                           src/Properties.cpp:084] Setting WEB Path to: /usr/share/satpi/web
  Sat Apr  3 20:52:11.3997 2021  6  [                           src/Properties.cpp:088] Setting App Data Path to: /var/lib/satpi
  Sat Apr  3 20:52:11.4035 2021  6  [                           src/HttpServer.cpp:058] Setting up HTTP server
  Sat Apr  3 20:52:11.4045 2021  6  [                           src/RtspServer.cpp:046] Setting up RTSP server
  Sat Apr  3 20:52:11.4145 2021  6  [                     src/upnp/ssdp/Server.cpp:104] Setting up SSDP server with BOOTID: 23  annouce interval: 60 Sec
  Sat Apr  3 20:52:13.6303 2021  6  [                    src/socket/SocketAttr.cpp:137] HTTP Connection from 192.168.112.61 Port 51915 with fd: 8
  Sat Apr  3 20:52:13.6789 2021  6  [                    src/socket/SocketAttr.cpp:137] HTTP Connection from 192.168.112.61 Port 51916 with fd: 9
  Sat Apr  3 20:52:13.6793 2021  6  [                    src/socket/SocketAttr.cpp:137] HTTP Connection from 192.168.112.61 Port 51917 with fd: 10
  Sat Apr  3 20:52:13.6794 2021  6  [                    src/socket/SocketAttr.cpp:137] HTTP Connection from 192.168.112.61 Port 51918 with fd: 11
  Sat Apr  3 20:52:13.6794 2021  6  [                    src/socket/SocketAttr.cpp:137] HTTP Connection from 192.168.112.61 Port 51919 with fd: 12
  Sat Apr  3 20:52:13.6794 2021  6  [                    src/socket/SocketAttr.cpp:137] HTTP Connection from 192.168.112.61 Port 51920 with fd: 13
  Sat Apr  3 20:52:18.6888 2021  6  [                    src/socket/SocketAttr.cpp:137] RTSP Connection from 192.168.112.61 Port 51923 with fd: 14
  Sat Apr  3 20:52:18.7040 2021  7  [                          src/HttpcServer.cpp:153] RTSP Stream data from client None with IP 192.168.112.61 on Port 51923: OPTIONS rtsp://satip:554/?src=1&freq=11494&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=0,17,18,5100,5101,5102,5104 RTSP/1.0
  Sat Apr  3 20:52:18.7040 2021  7  CSeq: 2
  Sat Apr  3 20:52:18.7040 2021  7  User-Agent: LibVLC/3.0.12 (LIVE555 Streaming Media v2016.11.28)
  Sat Apr  3 20:52:18.7040 2021  7  --- LINE END ---
  Sat Apr  3 20:52:18.7041 2021  7  [                          src/HttpcServer.cpp:213] RTSP/1.0 200 OK
  Sat Apr  3 20:52:18.7041 2021  7  CSeq: 2
  Sat Apr  3 20:52:18.7041 2021  7  Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
  Sat Apr  3 20:52:18.7041 2021  7  --- LINE END ---
  Sat Apr  3 20:52:18.7067 2021  7  [                          src/HttpcServer.cpp:153] RTSP Stream data from client None with IP 192.168.112.61 on Port 51923: DESCRIBE rtsp://satip:554/?src=1&freq=11494&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec [... cut ...]
  Sat Apr  3 20:52:22.7375 2021  6  [                   src/input/dvb/Frontend.cpp:664] Stream: 0, Closing /dev/dvb/adapter0/demux0 fd: 18
  Sat Apr  3 20:52:22.7376 2021  7  [                          src/HttpcServer.cpp:213] RTSP/1.0 200 OK
  Sat Apr  3 20:52:22.7376 2021  7  CSeq: 2
  Sat Apr  3 20:52:22.7376 2021  7  Session: 0286532486
  Sat Apr  3 20:52:22.7376 2021  7  --- LINE END ---
  Sat Apr  3 20:52:22.9530 2021  6  [                     src/socket/TcpSocket.cpp:106] RTSP Client 192.168.112.61:51924 Connection closed with fd: 14
```

Barracuda09, 2021-04-03:
does dmesg (syslog) show any errors?

nikolauzi11, 2021-04-03:
Hi @Barracuda09 

Unfortunately not.

Barracuda09, 2021-04-04:
Hi @nikolauzi11 

One more thing to try, I have changed the handling of PID filter some time ago. Maybe the SundTek does not like this.

This checkout will revert if back before this change. I would appreciate if you could try this:
```
git branch -f devtmp 9c4b71d
git checkout devtmp

```

Regards,

Marc

nikolauzi11, 2021-04-04:
Hi @Barracuda09 

This version works with VLC :grin:
With the VDR I can test it only tomorrow.

nikolauzi11, 2021-04-05:
Hi @Barracuda09 

On the VDR I only get a very stuttering stream with this version.
But probably it's best to solve the original problem first.

Barracuda09, 2021-04-05:
Hi @nikolauzi11 

Thanks for testing, so it seems the Sundtek drivers do not like the way of setting the PES-Filters (PID).

Barracuda09, 2021-04-05:
Hi @nikolauzi11 

How do you start SatPI, in combination with sundtek mediaclient etc. 

`LD_PRELOAD=/path/to/libmediaclient.so satpi`

nikolauzi11, 2021-04-05:
Hi @Barracuda09 

I think the LD_PRELOAD thing is no longer necessary, since the driver installer creates an /etc/ld.so.preload with the line /opt/lib/libmediaclient.so.
But I already tested it with LD_PRELOAD too, it made no difference.

Barracuda09, 2021-04-05:
Is this interesting (or still used) `/var/log/mediasrv.log`

nikolauzi11, 2021-04-05:
Hi @Barracuda09 

Only entries regarding the IR remote, like this:
`2021-04-05 11:20:14 [659] RC: IR Event /dev/input/event4`

Barracuda09, 2021-04-05:
What if you use pids=all in VLC? (With current master branch)

nikolauzi11, 2021-04-05:
Hi @Barracuda09 

pids=all does not work either, do you need the log?

Barracuda09, 2021-04-05:
Do you start `mediasrv -v` for debug

Barracuda09, 2021-04-05:
Hi @nikolauzi11 

Many thanks for your effort so far and is much appreciated.

I have one more think to ask, could you add this (before L745):
```
	/////////////////////////////////// Added //////////////////////////
	int n = _streamID;
	if (::ioctl(_fd_dmx, DMX_SET_SOURCE, &n) != 0) {
		PERROR("DMX_SET_SOURCE");
	}
	SI_LOG_INFO("Set DMX_SET_SOURCE for frontend %d (Offset: %d)", _streamID, offset);
	/////////////////////////////////// Added //////////////////////////
	struct dmx_pes_filter_params pesFilter;
	pesFilter.pid      = pid;
	pesFilter.input    = DMX_IN_FRONTEND;
	pesFilter.output   = DMX_OUT_TSDEMUX_TAP;
	pesFilter.pes_type = DMX_PES_OTHER;
	pesFilter.flags    = DMX_IMMEDIATE_START;
	if (::ioctl(_fd_dmx, DMX_SET_PES_FILTER, &pesFilter) != 0) {
		PERROR("Stream: %d, DMX_SET_PES_FILTER (PID %04d)", _streamID, 0);
		return;
	}
	SI_LOG_INFO("Stream: %d, Opened %s fd: %d", _streamID, _path_to_dmx.c_str(), _fd_dmx);

```

https://github.com/Barracuda09/SATPI/blob/9d9fa2202f25ce74c832fe74e5ead4d6951537fd/src/input/dvb/Frontend.cpp#L745-L760

nikolauzi11, 2021-04-05:
Hi @Barracuda09 

> Do you start mediasrv -v for debug

mediasrv is somehow started by mediaclient, I don't know how to influence the start parameters.

> I have one more think to ask, could you add this (before L745):

I get this error when compiling:

```g++ -c  -I ./src -std=c++11 -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DHAS_NP_FUNCTIONS src/input/dvb/Frontend.cpp -o obj/input/dvb/Frontend.o
In file included from ./src/base/Mutex.h:23,
                 from ./src/base/XMLSupport.h:23,
                 from ./src/input/Device.h:24,
                 from ./src/input/dvb/Frontend.h:24,
                 from src/input/dvb/Frontend.cpp:20:
src/input/dvb/Frontend.cpp: In member function ‘void input::dvb::Frontend::openPid(int)’:
src/input/dvb/Frontend.cpp:750:85: error: ‘offset’ was not declared in this scope; did you mean ‘off_t’?
  750 |           SI_LOG_INFO("Set DMX_SET_SOURCE for frontend %d (Offset: %d)", _streamID, offset);
      |                                                                                     ^~~~~~
./src/Log.h:68:70: note: in definition of macro ‘SI_LOG_INFO’
   68 | #define SI_LOG_INFO(...)                       Log::applog(LOG_INFO, __VA_ARGS__)
      |                                                                      ^~~~~~~~~~~
make: *** [Makefile:200: obj/input/dvb/Frontend.o] Fehler 1```

Barracuda09, 2021-04-05:
Hi @nikolauzi11 

Sorry here is the correct code:

```
	/////////////////////////////////// Added
	int offset = 0;
	int n = _streamID;
	if (::ioctl(_fd_dmx, DMX_SET_SOURCE, &n) != 0) {
		PERROR("DMX_SET_SOURCE");
	}
	SI_LOG_INFO("Set DMX_SET_SOURCE for frontend %d (Offset: %d)", _streamID, offset);
	/////////////////////////////////// Added
	struct dmx_pes_filter_params pesFilter;
	pesFilter.pid      = pid;
	pesFilter.input    = DMX_IN_FRONTEND;
	pesFilter.output   = DMX_OUT_TSDEMUX_TAP;
	pesFilter.pes_type = DMX_PES_OTHER;
	pesFilter.flags    = DMX_IMMEDIATE_START;
	if (::ioctl(_fd_dmx, DMX_SET_PES_FILTER, &pesFilter) != 0) {
		PERROR("Stream: %d, DMX_SET_PES_FILTER (PID %04d)", _streamID, 0);
		return;
	}
	SI_LOG_INFO("Stream: %d, Opened %s fd: %d", _streamID, _path_to_dmx.c_str(), _fd_dmx);

```

nikolauzi11, 2021-04-05:
Hi @Barracuda09 

Here ist the log:

```  Mon Apr  5 22:16:36.8381 2021  6  --- Starting SatPI version: 1.5.415~g8fe87fc ---
  Mon Apr  5 22:16:36.8381 2021  6  Number of processors online: 1
  Mon Apr  5 22:16:36.8381 2021  6  Default network buffer size: 104 KBytes
  Mon Apr  5 22:16:36.8382 2021  6  ens18: 192.168.112.66 [86:03:51:15:4f:89]
  Mon Apr  5 22:16:36.8382 2021  6  Current DVB_API_VERSION: 5.11
  Mon Apr  5 22:16:36.8382 2021  6  Enumerating all devices...
  Mon Apr  5 22:16:36.8386 2021  6  Detecting frontends in: /dev/dvb
  Mon Apr  5 22:16:37.8961 2021  6  Frontend Name: Sundtek DVB-S/S2 (VIII)
  Mon Apr  5 22:16:37.8961 2021  6  Frontend Type: Satellite (DVB-S2)
  Mon Apr  5 22:16:37.8961 2021  6  Frontend Type: Satellite (DVB-S)
  Mon Apr  5 22:16:37.8961 2021  6  Frontend Freq: 925000 Hz to 2175000 Hz
  Mon Apr  5 22:16:37.8961 2021  6  Frontend srat: 1000000 symbols/s to 45000000 symbols/s
  Mon Apr  5 22:16:38.9671 2021  6  Frontend Name: Sundtek DVB-S/S2 (VIII)
  Mon Apr  5 22:16:38.9671 2021  6  Frontend Type: Satellite (DVB-S2)
  Mon Apr  5 22:16:38.9671 2021  6  Frontend Type: Satellite (DVB-S)
  Mon Apr  5 22:16:38.9671 2021  6  Frontend Freq: 925000 Hz to 2175000 Hz
  Mon Apr  5 22:16:38.9671 2021  6  Frontend srat: 1000000 symbols/s to 45000000 symbols/s
  Mon Apr  5 22:16:38.9672 2021  6  Frontends found: 2
  Mon Apr  5 22:16:38.9672 2021  6  Setting up TS Reader using path: .
  Mon Apr  5 22:16:38.9673 2021  6  Setting up TS Streamer
  Mon Apr  5 22:16:38.9797 2021  6  Setting HTTP Port to: 8875
  Mon Apr  5 22:16:38.9797 2021  6  Setting RTSP Port to: 554
  Mon Apr  5 22:16:38.9797 2021  6  Setting WEB Path to: ./web
  Mon Apr  5 22:16:38.9797 2021  6  Setting App Data Path to: .
  Mon Apr  5 22:16:38.9813 2021  6  Setting up SSDP server with BOOTID: 13  annouce interval: 60 Sec
  Mon Apr  5 22:16:38.9813 2021  6  Setting up RTSP server
  Mon Apr  5 22:16:38.9814 2021  6  Setting up HTTP server
  Mon Apr  5 22:16:39.0929 2021  6  HTTP Connection from 192.168.112.61 Port 50229 with fd: 8
  Mon Apr  5 22:16:39.3488 2021  6  HTTP Connection from 192.168.112.61 Port 50230 with fd: 9
  Mon Apr  5 22:16:44.3562 2021  6  HTTP Client 192.168.112.61:50230 Connection closed with fd: 9
  Mon Apr  5 22:16:46.5948 2021  6  RTSP Connection from 192.168.112.61 Port 50234 with fd: 9
  Mon Apr  5 22:16:46.5948 2021  7  RTSP Stream data from client None with IP 192.168.112.61 on Port 50234: OPTIONS rtsp://satip:554/?src=1&freq=11362&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=0,17,18,6100,6110,6120,6130 RTSP/1.0
  Mon Apr  5 22:16:46.5948 2021  7  CSeq: 2
  Mon Apr  5 22:16:46.5948 2021  7  User-Agent: LibVLC/3.0.12 (LIVE555 Streaming Media v2016.11.28)
  Mon Apr  5 22:16:46.5948 2021  7  --- LINE END ---
  Mon Apr  5 22:16:46.5948 2021  7  RTSP/1.0 200 OK
  Mon Apr  5 22:16:46.5948 2021  7  CSeq: 2
  Mon Apr  5 22:16:46.5948 2021  7  Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
  Mon Apr  5 22:16:46.5948 2021  7  --- LINE END ---
  Mon Apr  5 22:16:46.5976 2021  7  RTSP Stream data from client None with IP 192.168.112.61 on Port 50234: DESCRIBE rtsp://satip:554/?src=1&freq=11362&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=0,17,18,6100,6110,6120,6130 RTSP/1.0
  Mon Apr  5 22:16:46.5976 2021  7  CSeq: 3
  Mon Apr  5 22:16:46.5976 2021  7  User-Agent: LibVLC/3.0.12 (LIVE555 Streaming Media v2016.11.28)
  Mon Apr  5 22:16:46.5976 2021  7  Accept: application/sdp
  Mon Apr  5 22:16:46.5976 2021  7  --- LINE END ---
  Mon Apr  5 22:16:46.5976 2021  7  RTSP/1.0 404 Not Found
  Mon Apr  5 22:16:46.5976 2021  7  CSeq: 3
  Mon Apr  5 22:16:46.5976 2021  7  Content-Type: application/sdp
  Mon Apr  5 22:16:46.5976 2021  7  Content-Base: rtsp://192.168.112.66/
  Mon Apr  5 22:16:46.5976 2021  7  Content-Length: 0
  Mon Apr  5 22:16:46.5976 2021  7  --- LINE END ---
  Mon Apr  5 22:16:46.6015 2021  6  RTSP Client 192.168.112.61:50234 Connection closed with fd: 9
  Mon Apr  5 22:16:46.6039 2021  6  RTSP Connection from 192.168.112.61 Port 50235 with fd: 9
  Mon Apr  5 22:16:46.6086 2021  7  RTSP Stream data from client None with IP 192.168.112.61 on Port 50235: SETUP rtsp://satip:554/?src=1&freq=11362&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=0,17,18,6100,6110,6120,6130 RTSP/1.0
  Mon Apr  5 22:16:46.6086 2021  7  CSeq: 0
  Mon Apr  5 22:16:46.6086 2021  7  Transport: RTP/AVP;unicast;client_port=9318-9319
  Mon Apr  5 22:16:46.6086 2021  7  --- LINE END ---
  Mon Apr  5 22:16:46.6087 2021  6  Found StreamID x - SessionID: 0276808213
  Mon Apr  5 22:16:46.6087 2021  6  Stream: 0, StreamClient[0] with SessionID 0276808213 for dvbs2
  Mon Apr  5 22:16:46.6087 2021  6  Stream: 0, Parsing transport parameters...
  Mon Apr  5 22:16:46.6087 2021  6  Stream: 0, New frequency requested, clearing old channel data...
  Mon Apr  5 22:16:46.6088 2021  7  Stream: 0, Parsing transport parameters (Finished)
  Mon Apr  5 22:16:46.6088 2021  7  Stream: 0, Found Streaming type: RTSP Unicast
  Mon Apr  5 22:16:46.6089 2021  6  Stream: 0, Updating frontend...
  Mon Apr  5 22:16:46.7614 2021  6  Stream: 0, Opened /dev/dvb/adapter0/frontend0 for Read/Write with fd: 11
  Mon Apr  5 22:16:46.7865 2021  6  Stream: 0, Start tuning process for DVB-S(2)...
  Mon Apr  5 22:16:46.7866 2021  6  Stream: 0, Opened /dev/dvb/adapter0/frontend0 for Writing DiSEqC command with fd: 11
[... 42 lines cut ...]
  Mon Apr  5 22:16:50.1054 2021  7  CSeq: 2
[... cut ...]
  Mon Apr  5 22:16:50.6063 2021  7  Stream: 0, Remove filter PID: 6130 - Packet Count: 0
  Mon Apr  5 22:16:50.6064 2021  6  Stream: 0, Closing /dev/dvb/adapter0/frontend0 fd: 11
  Mon Apr  5 22:16:50.6064 2021  6  Stream: 0, Closing /dev/dvb/adapter0/demux0 fd: 13
  Mon Apr  5 22:16:50.6065 2021  7  RTSP/1.0 200 OK
  Mon Apr  5 22:16:50.6065 2021  7  CSeq: 2
  Mon Apr  5 22:16:50.6065 2021  7  Session: 0276808213
  Mon Apr  5 22:16:50.6065 2021  7  --- LINE END ---
  Mon Apr  5 22:16:50.7602 2021  6  RTSP Client 192.168.112.61:50235 Connection closed with fd: 9
```

nikolauzi11, 2021-04-05:
Hi @Barracuda09 

I found out how to increase the log level of mediasrv:

```2021-04-05 22:47:42 [949] RC: IR Event /dev/input/event4
2021-04-05 22:47:42 [960] RC: IR Event /dev/input/event4
2021-04-05 22:51:34 [1056] Starting up Multimedia Framework (Mar 29 2021 12:56:05)
2021-04-05 22:51:34 [1056] Semaphore exists, removing and reallocating it
2021-04-05 22:51:34 [1056] Detected DVB version.h
2021-04-05 22:51:34 [1056] API Version: 5.11
2021-04-05 22:51:34 [1056] Running on Linux (5.4.0-70-generic)
2021-04-05 22:51:34 [1056] Detected: GenuineIntel
2021-04-05 22:51:34 [1056] This CPU supports: MMX, SSE, SSE2
2021-04-05 22:51:34 [1056] scanning devices
2021-04-05 22:51:34 [1056] 
2021-04-05 22:51:34 [1056] Scanning for drivers
2021-04-05 22:51:34 [1056] found lib: libdrv_fm_dab_dabplus.so
2021-04-05 22:51:34 [1056] found lib: libdrv_tntdvb.so
2021-04-05 22:51:34 [1056] found lib: libdrv_net.so
2021-04-05 22:51:34 [1056] found lib: libdrv_fm.so
2021-04-05 22:51:34 [1056] found lib: libdrv_hikarisoft_isdb.so
2021-04-05 22:51:34 [1056] found lib: libdrv_ngTV.so
2021-04-05 22:51:34 [1056] found lib: libdrv_eLight.so
2021-04-05 22:51:34 [1056] found lib: libdrv_vivi.so
2021-04-05 22:51:34 [1056] found lib: libdrv_em28xx.so
2021-04-05 22:51:34 [1056] 
2021-04-05 22:51:34 [1056] Scanning for plugins
2021-04-05 22:51:34 [1056] found lib: libencoder_plugin.so
2021-04-05 22:51:34 [1056] MediaAVPlugin: MPEG Encoder
2021-04-05 22:51:34 [1056] Copyright: (C) Tpe 2011/2012
2021-04-05 22:51:34 [1056] 
2021-04-05 22:51:34 [1056] Starting up device manager
2021-04-05 22:51:34 [1057] Starting service processor (PID 1057)
2021-04-05 22:51:34 [1056] scanning devices done
2021-04-05 22:51:34 [1056] attached devices:
2021-04-05 22:51:34 [1057] Autostarting scripts in /etc/sundtek/
2021-04-05 22:51:34 [1057] executed all available items
2021-04-05 22:51:34 [1056] registering ID: 0
2021-04-05 22:51:34 [1056] Memory mapped transfers are not available
2021-04-05 22:51:34 [1056] Board config found: Sundtek SkyTV Dual 8 (eLight)
2021-04-05 22:51:34 [1056] This board provides 2 frontends
2021-04-05 22:51:34 [1056] 
2021-04-05 22:51:34 [1056] *********
2021-04-05 22:51:34 [1056] Signature: Sundtek
2021-04-05 22:51:34 [1056] Initializing SYSV-IPC
2021-04-05 22:51:34 [1056] Allocation using pg for type 65678
2021-04-05 22:51:34 [1056] Allocated Shared memory buffer: 0x7fe7eab0e000
2021-04-05 22:51:36 [1056] demodulator successfully attached (DVB_S/S2)!
2021-04-05 22:51:36 [1056] attaching second tuner...
2021-04-05 22:51:36 [1056] Allocation using pg for type 65678
2021-04-05 22:51:36 [1056] Allocated Shared memory buffer: 0x7fe7ea778000
2021-04-05 22:51:37 [1056] demodulator successfully attached (DVB_S/S2)!
2021-04-05 22:51:37 [1056] Configuring Dual LNB Power Supply
2021-04-05 22:51:37 [1056] Vertical Polarisation is set to 14.00 Volt
2021-04-05 22:51:37 [1056] Horizontal Polarisation is set to 19.10 Volt
2021-04-05 22:51:37 [1056] Enabling LNB status monitor
2021-04-05 22:51:37 [1056] Using Sundtek remote control layout
2021-04-05 22:51:37 [1056] 
2021-04-05 22:51:37 [1056] IR Setup
2021-04-05 22:51:37 [1056] Initializing Remote Control Support (interrupt)
2021-04-05 22:51:37 [1056] registered virtual: /dev/mediainput0
2021-04-05 22:51:37 [1056] registering: adapter0/0
2021-04-05 22:51:37 [1056] registered virtual: /dev/dvb/adapter0/frontend0
2021-04-05 22:51:37 [1056] registered virtual: /dev/dvb/adapter0/demux0
2021-04-05 22:51:37 [1056] registered virtual: /dev/dvb/adapter0/dvr0
2021-04-05 22:51:37 [1056] registering: adapter1/0
2021-04-05 22:51:37 [1056] registered virtual: /dev/dvb/adapter1/frontend0
2021-04-05 22:51:37 [1056] registered virtual: /dev/dvb/adapter1/demux0
2021-04-05 22:51:37 [1056] registered virtual: /dev/dvb/adapter1/dvr0
2021-04-05 22:51:37 [1056] Driver loaded within 3655 milliseconds
2021-04-05 22:51:37 [1069] RC: IR Event /dev/input/event4
2021-04-05 22:51:37 [1082] RC: IR Event /dev/input/event4
2021-04-05 22:51:41 [1056] Initial devicescan done, switching detection mode
2021-04-05 22:51:46 [1056] Disabling High Tone (22khz)
2021-04-05 22:51:46 [1056] Set Voltage 18V
2021-04-05 22:51:46 [1056] DISEQC> sending commands:
2021-04-05 22:51:46 [1056] DISEQC> e0
2021-04-05 22:51:46 [1056] DISEQC> 10
2021-04-05 22:51:46 [1056] DISEQC> 38
2021-04-05 22:51:46 [1056] DISEQC> f2
2021-04-05 22:51:46 [1056] DISEQC> done
2021-04-05 22:51:46 [1056] Disabling High Tone (22khz)
2021-04-05 22:51:46 [1056] Diseqc execution time: 98 ms
2021-04-05 22:51:46 [1056] Disabling High Tone (22khz)
2021-04-05 22:51:46 [1056] diseqc send burst Unmodulated (SEC_MINI_A)
2021-04-05 22:51:46 [1056] Disabling High Tone (22khz)
2021-04-05 22:51:46 [1056] Disabling High Tone (22khz)
2021-04-05 22:51:46 [1056] Setting Frequency: 1612000
2021-04-05 22:51:46 [1056] Setting DVB-S2
2021-04-05 22:51:46 [1056] Frequency: 1612
2021-04-05 22:51:46 [1056] Symbolrate: 22000
2021-04-05 22:51:47 [1056] Frontend has locked
2021-04-05 22:51:47 [1056] DMX_SET_SOURCE not implemented
2021-04-05 22:51:47 [1056] INIT_DTV: 0
2021-04-05 22:51:47 [1056] Starting transfer
2021-04-05 22:51:47 [1056] TS Sync byte not aligned, realigning stream (0 // 0 // 0 // FEID: 0)
2021-04-05 22:51:55 [1056] Enabling Standby
2021-04-05 22:52:05 [1056] Enabling Standby
```

Barracuda09, 2021-04-06:
Hi @nikolauzi11 

```
2021-04-05 22:51:47 [1056] Frontend has locked
2021-04-05 22:51:47 [1056] DMX_SET_SOURCE not implemented
2021-04-05 22:51:47 [1056] INIT_DTV: 0
2021-04-05 22:51:47 [1056] Starting transfer
2021-04-05 22:51:47 [1056] TS Sync byte not aligned, realigning stream (0 // 0 // 0 // FEID: 0)
2021-04-05 22:51:55 [1056] Enabling Standby
2021-04-05 22:52:05 [1056] Enabling Standby

```
Strange and interesting problem, but here is showing it is starting transfer and that 'TS Sync byte' is not aligned. But were is it tranfering to? Not to SatPI or were it is reading from _/dev/dvb/adapter0/demux0_
Could it be that there is maybe an bug in the driver? The driver (should) support DMX_ADD_PID according to:
https://support.sundtek.com/index.php?topic=1573.msg11636#msg11636 
So I would think it should work how SatPI is adding PES-Filters now. The older version of SatPI worked.

Or could it be something with hardware filters?

DMX_SET_SOURCE was an test, so that is not a problem I should think. And should not really be needed.

I have done in the beginning a test with a Sundtek DVB-C/T stick but do not own one myself.

nikolauzi11, 2021-04-06:
Hi @Barracuda09 

I would let you use my DVB-S2 stick for some time for testing.

Barracuda09, 2021-04-06:
Hi @nikolauzi11 

If it is not to much trouble, it would nice to test via remote access?

nikolauzi11, 2021-04-06:
Hi @Barracuda09 

No problem, I could set up SSH access to the SATPI machine.

Barracuda09, 2021-04-06:
Hi @nikolauzi11 

I have send an e-mail to the nice people of Sundtek, and they will check it out as well.

Barracuda09, 2021-04-07:
Hi @nikolauzi11 

Thanks for you testing and support so far. I do have one more request.
Could you contact Sundtek at:  _kontakt -at- sundtek.com_
with supject: _Test Version of driver for SatPI_

Then they will send you a test version for you to test, to see if the driver works in combination with SatPI.

Kind regards,

Marc

nikolauzi11, 2021-04-07:
Hi @Barracuda09 

Thanks for contacting Sundtek, I got the test driver.
The Sundtek people are really helpful.
I didn't get a stream in VLC, but I didn't get an error like usual.

```
  Wed Apr  7 13:18:59.5918 2021  6  --- Starting SatPI version: 1.5.415~g8fe87fc ---
  Wed Apr  7 13:18:59.5919 2021  6  Number of processors online: 1
  Wed Apr  7 13:18:59.5919 2021  6  Default network buffer size: 104 KBytes
  Wed Apr  7 13:18:59.5920 2021  6  ens18: 192.168.112.66 [86:03:51:15:4f:89]
  Wed Apr  7 13:18:59.5920 2021  6  Current DVB_API_VERSION: 5.11
  Wed Apr  7 13:18:59.5920 2021  6  Enumerating all devices...
  Wed Apr  7 13:18:59.5920 2021  6  Detecting frontends in: /dev/dvb
  Wed Apr  7 13:19:00.6532 2021  6  Frontend Name: Sundtek DVB-S/S2 (VIII)
  Wed Apr  7 13:19:00.6533 2021  6  Frontend Type: Satellite (DVB-S2)
  Wed Apr  7 13:19:00.6533 2021  6  Frontend Type: Satellite (DVB-S)
  Wed Apr  7 13:19:00.6533 2021  6  Frontend Freq: 925000 Hz to 2175000 Hz
  Wed Apr  7 13:19:00.6533 2021  6  Frontend srat: 1000000 symbols/s to 45000000 symbols/s
  Wed Apr  7 13:19:01.7092 2021  6  Frontend Name: Sundtek DVB-S/S2 (VIII)
  Wed Apr  7 13:19:01.7093 2021  6  Frontend Type: Satellite (DVB-S2)
  Wed Apr  7 13:19:01.7093 2021  6  Frontend Type: Satellite (DVB-S)
  Wed Apr  7 13:19:01.7093 2021  6  Frontend Freq: 925000 Hz to 2175000 Hz
  Wed Apr  7 13:19:01.7093 2021  6  Frontend srat: 1000000 symbols/s to 45000000 symbols/s
  Wed Apr  7 13:19:01.7093 2021  6  Frontends found: 2
  Wed Apr  7 13:19:01.7093 2021  6  Setting up TS Reader using path: .
  Wed Apr  7 13:19:01.7095 2021  6  Setting up TS Streamer
  Wed Apr  7 13:19:01.7103 2021  6  Setting HTTP Port to: 8875
  Wed Apr  7 13:19:01.7103 2021  6  Setting RTSP Port to: 554
  Wed Apr  7 13:19:01.7103 2021  6  Setting WEB Path to: ./web
  Wed Apr  7 13:19:01.7103 2021  6  Setting App Data Path to: .
  Wed Apr  7 13:19:01.7115 2021  6  Setting up RTSP server
  Wed Apr  7 13:19:01.7115 2021  6  Setting up HTTP server
  Wed Apr  7 13:19:01.7229 2021  6  Setting up SSDP server with BOOTID: 21  annouce interval: 60 Sec
  Wed Apr  7 13:19:01.7295 2021  6  HTTP Connection from 192.168.112.61 Port 57467 with fd: 8
  Wed Apr  7 13:19:01.9923 2021  6  HTTP Connection from 192.168.112.61 Port 57468 with fd: 9
  Wed Apr  7 13:19:02.2767 2021  6  HTTP Connection from 192.168.112.61 Port 57470 with fd: 10
  Wed Apr  7 13:19:02.2774 2021  7  Client: 192.168.112.61 requested desc.xml
  Wed Apr  7 13:19:02.2825 2021  6  HTTP Client 192.168.112.61:57470 Connection closed with fd: 10
  Wed Apr  7 13:19:02.8951 2021  6  HTTP Connection from 192.168.112.61 Port 57471 with fd: 10
  Wed Apr  7 13:19:02.8954 2021  7  Client: 192.168.112.61 requested desc.xml
  Wed Apr  7 13:19:02.9001 2021  6  HTTP Client 192.168.112.61:57471 Connection closed with fd: 10
  Wed Apr  7 13:19:07.0019 2021  6  HTTP Client 192.168.112.61:57468 Connection closed with fd: 9
  Wed Apr  7 13:19:08.6100 2021  6  RTSP Connection from 192.168.112.61 Port 57472 with fd: 9
  Wed Apr  7 13:19:08.6100 2021  7  RTSP Stream data from client None with IP 192.168.112.61 on Port 57472: OPTIONS rtsp://satip:554/?src=1&freq=11362&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=0,17,18,6100,6110,6120,6130 RTSP/1.0
  Wed Apr  7 13:19:08.6100 2021  7  CSeq: 2
  Wed Apr  7 13:19:08.6100 2021  7  User-Agent: LibVLC/3.0.12 (LIVE555 Streaming Media v2016.11.28)
  Wed Apr  7 13:19:08.6100 2021  7  --- LINE END ---
  Wed Apr  7 13:19:08.6100 2021  7  RTSP/1.0 200 OK
  Wed Apr  7 13:19:08.6100 2021  7  CSeq: 2
  Wed Apr  7 13:19:08.6100 2021  7  Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
  Wed Apr  7 13:19:08.6100 2021  7  --- LINE END ---
  Wed Apr  7 13:19:08.6124 2021  7  RTSP Stream data from client None with IP 192.168.112.61 on Port 57472: DESCRIBE rtsp://satip:554/?src=1&freq=11362&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=0,17,18,6100,6110,6120,6130 RTSP/1.0
  Wed Apr  7 13:19:08.6124 2021  7  CSeq: 3
  Wed Apr  7 13:19:08.6124 2021  7  User-Agent: LibVLC/3.0.12 (LIVE555 Streaming Media v2016.11.28)
  Wed Apr  7 13:19:08.6124 2021  7  Accept: application/sdp
  Wed Apr  7 13:19:08.6124 2021  7  --- LINE END ---
  Wed Apr  7 13:19:08.6124 2021  7  RTSP/1.0 404 Not Found
  Wed Apr  7 13:19:08.6124 2021  7  CSeq: 3
  Wed Apr  7 13:19:08.6124 2021  7  Content-Type: application/sdp
  Wed Apr  7 13:19:08.6124 2021  7  Content-Base: rtsp://192.168.112.66/
  Wed Apr  7 13:19:08.6124 2021  7  Content-Length: 0
  Wed Apr  7 13:19:08.6124 2021  7  --- LINE END ---
  Wed Apr  7 13:19:08.6153 2021  6  RTSP Client 192.168.112.61:57472 Connection closed with fd: 9
  Wed Apr  7 13:19:08.6177 2021  6  RTSP Connection from 192.168.112.61 Port 57473 with fd: 9
  Wed Apr  7 13:19:08.6231 2021  7  RTSP Stream data from client None with IP 192.168.112.61 on Port 57473: SETUP rtsp://satip:554/?src=1&freq=11362&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=0,17,18,6100,6110,6120,6130 RTSP/1.0
  Wed Apr  7 13:19:08.6231 2021  7  CSeq: 0
  Wed Apr  7 13:19:08.6231 2021  7  Transport: RTP/AVP;unicast;client_port=9260-9261
  Wed Apr  7 13:19:08.6231 2021  7  --- LINE END ---
  Wed Apr  7 13:19:08.6232 2021  6  Found StreamID x - SessionID: 0273402352
[... 151 lines cut ...]
2021-04-07 13:19:08 [973] Set Voltage 18V
2021-04-07 13:19:08 [973] DISEQC> sending commands:
2021-04-07 13:19:08 [973] DISEQC> e0
2021-04-07 13:19:08 [973] DISEQC> 10
[... cut ...]
2021-04-07 13:19:09 [973] Frequency: 1612
2021-04-07 13:19:09 [973] Symbolrate: 22000
2021-04-07 13:19:10 [973] Frontend has locked
2021-04-07 13:19:10 [973] DMX_SET_SOURCE not implemented
2021-04-07 13:19:10 [973] INIT_DTV: 0
2021-04-07 13:19:10 [973] Starting transfer
2021-04-07 13:19:10 [973] TS Sync byte not aligned, realigning stream (0 // 0 // d7 // FEID: 0)
2021-04-07 13:19:10 [973] TS Sync byte not aligned, realigning stream (192 // 0 // 73 // FEID: 0)
2021-04-07 13:19:16 [973] Enabling Standby
```

Barracuda09, 2021-04-07:
Hi @nikolauzi11 

I see that you are still using the test version of SatPI (with DMX_SET_SOURCE). It should not make to much difference, but on the save side just re-get master branch.

I see PID 0000 is now receiving data:
```
Wed Apr  7 13:19:10.1039 2021  7  --- LINE END ---
  Wed Apr  7 13:19:10.6678 2021  6  Stream: 0, PAT - PID 0000: sectionLength: 21  tableDataSize: 188  secNr: 0  lastSecNr: 0  currSecNr: 0
  Wed Apr  7 13:19:10.6678 2021  6  Stream: 0, PAT: Section Length: 21  TID: 1011  Version: 193  secNr: 0 lastSecNr: 0  CRC: 0x47179A1A
  Wed Apr  7 13:19:10.6678 2021  6  Stream: 0, PAT: Prog NR: 0x0000 - 00000  NIT PID: 0016
  Wed Apr  7 13:19:10.6678 2021  6  Stream: 0, PAT: Prog NR: 0x2B66 - 11110  PMT PID: 6100
  Wed Apr  7 13:19:10.6678 2021  6  Stream: 0, PAT: Prog NR: 0x2B7A - 11130  PMT PID: 6300
  Wed Apr  7 13:20:03.1066 2021  6  HTTP Connection from 192.168.112.61 Port 57485 with fd: 16
```

But is the rest of the PID receiving data?

nikolauzi11, 2021-04-07:
Hi @Barracuda09 

This time without DMX_SET_SOURCE:smiley:

```
  Wed Apr  7 13:47:49.3837 2021  6  --- Starting SatPI version: 1.5.415~g8fe87fc ---
  Wed Apr  7 13:47:49.3837 2021  6  Number of processors online: 1
  Wed Apr  7 13:47:49.3838 2021  6  Default network buffer size: 104 KBytes
  Wed Apr  7 13:47:49.3838 2021  6  ens18: 192.168.112.66 [86:03:51:15:4f:89]
  Wed Apr  7 13:47:49.3838 2021  6  Current DVB_API_VERSION: 5.11
  Wed Apr  7 13:47:49.3838 2021  6  Enumerating all devices...
  Wed Apr  7 13:47:49.3839 2021  6  Detecting frontends in: /dev/dvb
  Wed Apr  7 13:47:49.3861 2021  6  Frontend Name: Sundtek DVB-S/S2 (VIII)
  Wed Apr  7 13:47:49.3862 2021  6  Frontend Type: Satellite (DVB-S2)
  Wed Apr  7 13:47:49.3862 2021  6  Frontend Type: Satellite (DVB-S)
  Wed Apr  7 13:47:49.3862 2021  6  Frontend Freq: 925000 Hz to 2175000 Hz
  Wed Apr  7 13:47:49.3862 2021  6  Frontend srat: 1000000 symbols/s to 45000000 symbols/s
  Wed Apr  7 13:47:49.3891 2021  6  Frontend Name: Sundtek DVB-S/S2 (VIII)
  Wed Apr  7 13:47:49.3892 2021  6  Frontend Type: Satellite (DVB-S2)
  Wed Apr  7 13:47:49.3892 2021  6  Frontend Type: Satellite (DVB-S)
  Wed Apr  7 13:47:49.3892 2021  6  Frontend Freq: 925000 Hz to 2175000 Hz
  Wed Apr  7 13:47:49.3892 2021  6  Frontend srat: 1000000 symbols/s to 45000000 symbols/s
  Wed Apr  7 13:47:49.3892 2021  6  Frontends found: 2
  Wed Apr  7 13:47:49.3892 2021  6  Setting up TS Reader using path: .
  Wed Apr  7 13:47:49.3893 2021  6  Setting up TS Streamer
  Wed Apr  7 13:47:49.3901 2021  6  Setting HTTP Port to: 8875
  Wed Apr  7 13:47:49.3901 2021  6  Setting RTSP Port to: 554
  Wed Apr  7 13:47:49.3901 2021  6  Setting WEB Path to: ./web
  Wed Apr  7 13:47:49.3901 2021  6  Setting App Data Path to: .
  Wed Apr  7 13:47:49.3914 2021  6  Setting up RTSP server
  Wed Apr  7 13:47:49.3914 2021  6  Setting up HTTP server
  Wed Apr  7 13:47:49.4027 2021  6  Setting up SSDP server with BOOTID: 2  annouce interval: 60 Sec
  Wed Apr  7 13:47:49.5941 2021  6  HTTP Connection from 192.168.112.61 Port 58566 with fd: 8
  Wed Apr  7 13:47:49.6386 2021  6  HTTP Connection from 192.168.112.61 Port 58564 with fd: 9
  Wed Apr  7 13:47:49.8908 2021  6  HTTP Connection from 192.168.112.61 Port 58565 with fd: 10
  Wed Apr  7 13:47:50.0009 2021  6  HTTP Connection from 192.168.112.61 Port 58567 with fd: 11
  Wed Apr  7 13:47:50.0014 2021  7  Client: 192.168.112.61 requested desc.xml
  Wed Apr  7 13:47:50.0062 2021  6  HTTP Client 192.168.112.61:58567 Connection closed with fd: 11
  Wed Apr  7 13:47:50.3053 2021  6  HTTP Connection from 192.168.112.61 Port 58568 with fd: 11
  Wed Apr  7 13:47:50.3054 2021  7  Client: 192.168.112.61 requested desc.xml
  Wed Apr  7 13:47:50.3120 2021  6  HTTP Client 192.168.112.61:58568 Connection closed with fd: 11
  Wed Apr  7 13:47:50.6122 2021  6  HTTP Connection from 192.168.112.61 Port 58569 with fd: 11
  Wed Apr  7 13:47:50.6125 2021  7  Client: 192.168.112.61 requested desc.xml
  Wed Apr  7 13:47:50.6188 2021  6  HTTP Client 192.168.112.61:58569 Connection closed with fd: 11
  Wed Apr  7 13:47:54.6412 2021  6  HTTP Client 192.168.112.61:58564 Connection closed with fd: 9
  Wed Apr  7 13:47:55.6472 2021  6  HTTP Client 192.168.112.61:58565 Connection closed with fd: 10
  Wed Apr  7 13:48:05.2987 2021  6  RTSP Connection from 192.168.112.61 Port 58573 with fd: 9
  Wed Apr  7 13:48:05.2987 2021  7  RTSP Stream data from client None with IP 192.168.112.61 on Port 58573: OPTIONS rtsp://satip:554/?src=1&freq=11362&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=0,17,18,6100,6110,6120,6130 RTSP/1.0
  Wed Apr  7 13:48:05.2987 2021  7  CSeq: 2
  Wed Apr  7 13:48:05.2987 2021  7  User-Agent: LibVLC/3.0.12 (LIVE555 Streaming Media v2016.11.28)
  Wed Apr  7 13:48:05.2987 2021  7  --- LINE END ---
  Wed Apr  7 13:48:05.2988 2021  7  RTSP/1.0 200 OK
  Wed Apr  7 13:48:05.2988 2021  7  CSeq: 2
  Wed Apr  7 13:48:05.2988 2021  7  Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
  Wed Apr  7 13:48:05.2988 2021  7  --- LINE END ---
  Wed Apr  7 13:48:05.3036 2021  7  RTSP Stream data from client None with IP 192.168.112.61 on Port 58573: DESCRIBE rtsp://satip:554/?src=1&freq=11362&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=0,17,18,6100,6110,6120,6130 RTSP/1.0
  Wed Apr  7 13:48:05.3036 2021  7  CSeq: 3
  Wed Apr  7 13:48:05.3036 2021  7  User-Agent: LibVLC/3.0.12 (LIVE555 Streaming Media v2016.11.28)
  Wed Apr  7 13:48:05.3036 2021  7  Accept: application/sdp
  Wed Apr  7 13:48:05.3036 2021  7  --- LINE END ---
  Wed Apr  7 13:48:05.3036 2021  7  RTSP/1.0 404 Not Found
  Wed Apr  7 13:48:05.3036 2021  7  CSeq: 3
  Wed Apr  7 13:48:05.3036 2021  7  Content-Type: application/sdp
  Wed Apr  7 13:48:05.3036 2021  7  Content-Base: rtsp://192.168.112.66/
  Wed Apr  7 13:48:05.3036 2021  7  Content-Length: 0
  Wed Apr  7 13:48:05.3036 2021  7  --- LINE END ---
  Wed Apr  7 13:48:05.3084 2021  6  RTSP Client 192.168.112.61:58573 Connection closed with fd: 9
  Wed Apr  7 13:48:05.3110 2021  6  RTSP Connection from 192.168.112.61 Port 58574 with fd: 9
  Wed Apr  7 13:48:05.3146 2021  7  RTSP Stream data from client None with IP 192.168.112.61 on Port 58574: SETUP rtsp://satip:554/?src=1&freq=11362&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=0,17,18,6100,6110,6120,6130 RTSP/1.0
  Wed Apr  7 13:48:05.3146 2021  7  CSeq: 0
[... 135 lines cut ...]
2021-04-07 13:48:06 [1470] Set Voltage off
[... cut ...]
2021-04-07 13:48:06 [1470] Setting Frequency: 1612000
2021-04-07 13:48:06 [1470] Setting DVB-S2
2021-04-07 13:48:06 [1470] Frequency: 1612
2021-04-07 13:48:06 [1470] Symbolrate: 22000
2021-04-07 13:48:07 [1470] Frontend has locked
2021-04-07 13:48:07 [1470] INIT_DTV: 0
2021-04-07 13:48:07 [1470] Starting transfer
2021-04-07 13:48:07 [1470] TS Sync byte not aligned, realigning stream (0 // 0 // 3d // FEID: 0)
2021-04-07 13:48:07 [1470] TS Sync byte not aligned, realigning stream (279 // 0 // d6 // FEID: 0)
```

Barracuda09, 2021-04-07:
And if you stop the stream what are the PID showing?

nikolauzi11, 2021-04-07:
Hi @Barracuda09 
```
  Wed Apr  7 14:05:03.9523 2021  7  --- LINE END ---
  Wed Apr  7 14:05:04.5157 2021  6  Stream: 0, PAT - PID 0000: sectionLength: 21  tableDataSize: 188  secNr: 0  lastSecNr: 0  currSecNr: 0
  Wed Apr  7 14:05:04.5157 2021  6  Stream: 0, PAT: Section Length: 21  TID: 1011  Version: 193  secNr: 0 lastSecNr: 0  CRC: 0x47179A1A
  Wed Apr  7 14:05:04.5157 2021  6  Stream: 0, PAT: Prog NR: 0x0000 - 00000  NIT PID: 0016
  Wed Apr  7 14:05:04.5157 2021  6  Stream: 0, PAT: Prog NR: 0x2B66 - 11110  PMT PID: 6100
  Wed Apr  7 14:05:04.5157 2021  6  Stream: 0, PAT: Prog NR: 0x2B7A - 11130  PMT PID: 6300
  Wed Apr  7 14:05:06.1896 2021  6  HTTP Connection from 192.168.112.61 Port 58832 with fd: 16
  Wed Apr  7 14:05:06.1897 2021  7  Client: 192.168.112.61 requested desc.xml
  Wed Apr  7 14:05:06.1958 2021  6  HTTP Client 192.168.112.61:58832 Connection closed with fd: 16
  Wed Apr  7 14:05:06.4985 2021  6  HTTP Connection from 192.168.112.61 Port 58833 with fd: 16
  Wed Apr  7 14:05:06.4986 2021  7  Client: 192.168.112.61 requested desc.xml
  Wed Apr  7 14:05:06.5038 2021  6  HTTP Client 192.168.112.61:58833 Connection closed with fd: 16
  Wed Apr  7 14:05:24.1223 2021  7  RTSP Stream data from client None with IP 192.168.112.61 on Port 58831: TEARDOWN rtsp://satip:554/stream=0 RTSP/1.0
  Wed Apr  7 14:05:24.1223 2021  7  CSeq: 2
  Wed Apr  7 14:05:24.1223 2021  7  Session: 0270023721
  Wed Apr  7 14:05:24.1223 2021  7  --- LINE END ---
  Wed Apr  7 14:05:24.1223 2021  6  Found StreamID 0 - SessionID 0270023721
  Wed Apr  7 14:05:24.1223 2021  6  Stream: 0, StreamClient[0] with SessionID 0270023721
  Wed Apr  7 14:05:24.1223 2021  6  Stream: 0, Teardown StreamClient[0] with SessionID 0270023721
  Wed Apr  7 14:05:24.2224 2021  6  Stream: 0, Destroy RTP/UDP stream to 192.168.112.61:9086
  Wed Apr  7 14:05:24.5227 2021  6  Stream: 0, Destroy RTCP/UDP stream to 192.168.112.61:9087
  Wed Apr  7 14:05:24.5228 2021  7  Stream: 0, Remove filter PID: 0000 - Packet Count: 203
  Wed Apr  7 14:05:24.5228 2021  7  Stream: 0, Remove filter PID: 0001 - Packet Count: 0
  Wed Apr  7 14:05:24.5228 2021  7  Stream: 0, Remove filter PID: 0016 - Packet Count: 0
  Wed Apr  7 14:05:24.5228 2021  7  Stream: 0, Remove filter PID: 0017 - Packet Count: 0
  Wed Apr  7 14:05:24.5228 2021  7  Stream: 0, Remove filter PID: 0018 - Packet Count: 0
  Wed Apr  7 14:05:24.5231 2021  7  Stream: 0, Remove filter PID: 6100 - Packet Count: 0 - PMT
  Wed Apr  7 14:05:24.5231 2021  7  Stream: 0, Remove filter PID: 6110 - Packet Count: 0
  Wed Apr  7 14:05:24.5231 2021  7  Stream: 0, Remove filter PID: 6120 - Packet Count: 0
  Wed Apr  7 14:05:24.5231 2021  7  Stream: 0, Remove filter PID: 6130 - Packet Count: 0
  Wed Apr  7 14:05:24.5232 2021  6  Stream: 0, Closing /dev/dvb/adapter0/frontend0 fd: 11
  Wed Apr  7 14:05:24.5232 2021  6  Stream: 0, Closing /dev/dvb/adapter0/demux0 fd: 13
  Wed Apr  7 14:05:24.5233 2021  7  RTSP/1.0 200 OK
  Wed Apr  7 14:05:24.5233 2021  7  CSeq: 2
  Wed Apr  7 14:05:24.5233 2021  7  Session: 0270023721
  Wed Apr  7 14:05:24.5233 2021  7  --- LINE END ---```

Barracuda09, 2021-04-07:
Hi @nikolauzi11 

Thanks for your effort.

So it seems that at the moment only PID 0000 is receiving data.
And maybe the rest is not added (in the driver?) to the PES-filters with DMX_ADD_PID?

nikolauzi11, 2021-04-07:
Hi @Barracuda09 

Does this mean that Sundtek has to work on the driver again?

Barracuda09, 2021-04-07:
Hi @nikolauzi11

I'm afraid so, but I'm not 100% sure. The PID 0000 data is received by SatPI via the 'new' method. but for some reason the rest of the PIDS are not passed to SatPI

nikolauzi11, 2021-04-07:
Hi @Barracuda09 

Good news!
Sundtek support was just remote on my machine and fixed the DMX_ADD_PID feature. He said that it was forgotten because no one else was using it.
Anyway, SatPi now works with VLC and DMX_ADD_PID comes back in the official driver.
I still have to test with the VDR, but unfortunately I don't have time today.
I'll get back to you tomorrow.

Barracuda09, 2021-04-07:
Hi @nikolauzi11 

Great that they could fix it that quick.

Well I used the more 'used' (or historical) method before, that is why the old version worked. But this  method is way easier in setting up PES-Filters and reading data.

nikolauzi11, 2021-04-08:
Hi @Barracuda09 

Now that it works fine with VLC, with the VDR unfortunately it does not work yet.

```
  Thu Apr  8 11:34:19.5565 2021  7  RTSP/1.0 503 Service Unavailable
  Thu Apr  8 11:34:19.5565 2021  7  Server: SatPI WebServer v0.1
  Thu Apr  8 11:34:19.5565 2021  7  Location: 
  Thu Apr  8 11:34:19.5565 2021  7  CSeq: 2
  Thu Apr  8 11:34:19.5565 2021  7  cache-control: no-cache
  Thu Apr  8 11:34:19.5565 2021  7  Content-Type: video/MP2T
  Thu Apr  8 11:34:19.5565 2021  7  --- LINE END ---
  Thu Apr  8 11:34:19.5569 2021  6  RTSP Client 192.168.112.26:39078 Connection closed with fd: 10
  Thu Apr  8 11:34:19.8079 2021  6  RTSP Connection from 192.168.112.26 Port 39080 with fd: 10
  Thu Apr  8 11:34:19.8080 2021  7  RTSP Stream data from client None with IP 192.168.112.26 on Port 39080: OPTIONS rtsp://192.168.112.66/ RTSP/1.0
  Thu Apr  8 11:34:19.8080 2021  7  CSeq: 1
  Thu Apr  8 11:34:19.8080 2021  7  User-Agent: vdr-satip/2.4.1 (device 0)
  Thu Apr  8 11:34:19.8080 2021  7  --- LINE END ---
  Thu Apr  8 11:34:19.8080 2021  7  RTSP/1.0 200 OK
  Thu Apr  8 11:34:19.8080 2021  7  CSeq: 1
  Thu Apr  8 11:34:19.8080 2021  7  Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
  Thu Apr  8 11:34:19.8080 2021  7  --- LINE END ---
  Thu Apr  8 11:34:19.8085 2021  7  RTSP Stream data from client None with IP 192.168.112.26 on Port 39080: SETUP rtsp://192.168.112.66/?src=1&freq=11494&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&sr=22000&fec=23 RTSP/1.0
  Thu Apr  8 11:34:19.8085 2021  7  CSeq: 2
  Thu Apr  8 11:34:19.8085 2021  7  Transport: RTP/AVP;unicast;client_port=35032-35033
  Thu Apr  8 11:34:19.8085 2021  7  User-Agent: vdr-satip/2.4.1 (device 0)
  Thu Apr  8 11:34:19.8085 2021  7  --- LINE END ---
  Thu Apr  8 11:34:19.8085 2021  6  Found StreamID x - SessionID: 0267341282
  Thu Apr  8 11:34:19.8085 2021  6  Stream: 0, New session but this stream is in use, skipping...
  Thu Apr  8 11:34:19.8085 2021  6  Stream: 1, New session but this stream is in use, skipping...
  Thu Apr  8 11:34:19.8085 2021  6  Stream: 2, Not capable of handling msys=dvbs2
  Thu Apr  8 11:34:19.8085 2021  6  Stream: 3, Not capable of handling msys=dvbs2
  Thu Apr  8 11:34:19.8085 2021  3  Found no Stream/Client of interest!
  Thu Apr  8 11:34:19.8085 2021  7  RTSP/1.0 503 Service Unavailable
  Thu Apr  8 11:34:19.8085 2021  7  Server: SatPI WebServer v0.1
  Thu Apr  8 11:34:19.8085 2021  7  Location: 
  Thu Apr  8 11:34:19.8085 2021  7  CSeq: 2
  Thu Apr  8 11:34:19.8085 2021  7  cache-control: no-cache
  Thu Apr  8 11:34:19.8085 2021  7  Content-Type: video/MP2T
  Thu Apr  8 11:34:19.8085 2021  7  --- LINE END ---
  Thu Apr  8 11:34:19.8091 2021  6  RTSP Client 192.168.112.26:39080 Connection closed with fd: 10
  Thu Apr  8 11:34:20.0597 2021  6  RTSP Connection from 192.168.112.26 Port 39082 with fd: 10
  Thu Apr  8 11:34:20.0598 2021  7  RTSP Stream data from client None with IP 192.168.112.26 on Port 39082: OPTIONS rtsp://192.168.112.66/ RTSP/1.0
  Thu Apr  8 11:34:20.0598 2021  7  CSeq: 1
  Thu Apr  8 11:34:20.0598 2021  7  User-Agent: vdr-satip/2.4.1 (device 0)
  Thu Apr  8 11:34:20.0598 2021  7  --- LINE END ---
  Thu Apr  8 11:34:20.0598 2021  7  RTSP/1.0 200 OK
  Thu Apr  8 11:34:20.0598 2021  7  CSeq: 1
  Thu Apr  8 11:34:20.0598 2021  7  Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
  Thu Apr  8 11:34:20.0598 2021  7  --- LINE END ---
  Thu Apr  8 11:34:20.0603 2021  7  RTSP Stream data from client None with IP 192.168.112.26 on Port 39082: SETUP rtsp://192.168.112.66/?src=1&freq=11494&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&sr=22000&fec=23 RTSP/1.0
  Thu Apr  8 11:34:20.0603 2021  7  CSeq: 2
  Thu Apr  8 11:34:20.0603 2021  7  Transport: RTP/AVP;unicast;client_port=35032-35033
  Thu Apr  8 11:34:20.0603 2021  7  User-Agent: vdr-satip/2.4.1 (device 0)
  Thu Apr  8 11:34:20.0603 2021  7  --- LINE END ---
  Thu Apr  8 11:34:20.0603 2021  6  Found StreamID x - SessionID: 0271447621
  Thu Apr  8 11:34:20.0603 2021  6  Stream: 0, New session but this stream is in use, skipping...
  Thu Apr  8 11:34:20.0603 2021  6  Stream: 1, New session but this stream is in use, skipping...
  Thu Apr  8 11:34:20.0603 2021  6  Stream: 2, Not capable of handling msys=dvbs2
  Thu Apr  8 11:34:20.0603 2021  6  Stream: 3, Not capable of handling msys=dvbs2
  Thu Apr  8 11:34:20.0603 2021  3  Found no Stream/Client of interest!
  Thu Apr  8 11:34:20.0603 2021  7  RTSP/1.0 503 Service Unavailable
  Thu Apr  8 11:34:20.0603 2021  7  Server: SatPI WebServer v0.1
  Thu Apr  8 11:34:20.0603 2021  7  Location: 
  Thu Apr  8 11:34:20.0603 2021  7  CSeq: 2
  Thu Apr  8 11:34:20.0603 2021  7  cache-control: no-cache
  Thu Apr  8 11:34:20.0603 2021  7  Content-Type: video/MP2T
  Thu Apr  8 11:34:20.0603 2021  7  --- LINE END ---
  Thu Apr  8 11:34:20.0607 2021  6  RTSP Client 192.168.112.26:39082 Connection closed with fd: 10
  Thu Apr  8 11:34:20.3115 2021  6  RTSP Connection from 192.168.112.26 Port 39084 with fd: 10
[... 320 lines cut ...]
  Thu Apr  8 11:34:23.0824 2021  7  --- LINE END ---
  Thu Apr  8 11:34:23.0824 2021  6  Found StreamID x - SessionID: 0288386067
  Thu Apr  8 11:34:23.0824 2021  6  Stream: 0, New session but this stream is in use, skipping...
  Thu Apr  8 11:34:23.0824 2021  6  Stream: 1, New session but this stream is in use, skipping...
  Thu Apr  8 11:34:23.0824 2021  6  Stream: 2, Not capable of handling msys=dvbs2
  Thu Apr  8 11:34:23.0824 2021  6  Stream: 3, Not capable of handling msys=dvbs2
[... cut ...]
Apr  8 11:35:31 vdr6 vdr: [1412] SATIP-ERROR: Connect failed [device 0]
Apr  8 11:35:31 vdr6 vdr: [1415] SATIP-ERROR: Connect failed [device 1]
Apr  8 11:35:31 vdr6 vdr: [1415] SATIP-ERROR: Detected invalid status code 503: rtsp://192.168.112.66/ [device 1]
Apr  8 11:35:31 vdr6 vdr: [1415] SATIP-ERROR: Connect failed [device 1]
Apr  8 11:35:31 vdr6 vdr: [1412] SATIP-ERROR: Detected invalid status code 503: rtsp://192.168.112.66/ [device 0]
Apr  8 11:35:31 vdr6 vdr: [1412] SATIP-ERROR: Connect failed [device 0]
```

Barracuda09, 2021-04-08:
Hi @nikolauzi11 

Something went wrong before this:
```
Thu Apr  8 11:34:19.8085 2021  7  RTSP Stream data from client None with IP 192.168.112.26 on Port 39080: SETUP rtsp://192.168.112.66/?src=1&freq=11494&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&sr=22000&fec=23 RTSP/1.0
  Thu Apr  8 11:34:19.8085 2021  7  CSeq: 2
  Thu Apr  8 11:34:19.8085 2021  7  Transport: RTP/AVP;unicast;client_port=35032-35033
  Thu Apr  8 11:34:19.8085 2021  7  User-Agent: vdr-satip/2.4.1 (device 0)
  Thu Apr  8 11:34:19.8085 2021  7  --- LINE END ---
  Thu Apr  8 11:34:19.8085 2021  6  Found StreamID x - SessionID: 0267341282
  Thu Apr  8 11:34:19.8085 2021  6  Stream: 0, New session but this stream is in use, skipping...
  Thu Apr  8 11:34:19.8085 2021  6  Stream: 1, New session but this stream is in use, skipping...
  Thu Apr  8 11:34:19.8085 2021  6  Stream: 2, Not capable of handling msys=dvbs2
  Thu Apr  8 11:34:19.8085 2021  6  Stream: 3, Not capable of handling msys=dvbs2
  Thu Apr  8 11:34:19.8085 2021  3  Found no Stream/Client of interest!
  Thu Apr  8 11:34:19.8085 2021  7  RTSP/1.0 503 Service Unavailable
  Thu Apr  8 11:34:19.8085 2021  7  Server: SatPI WebServer v0.1
  Thu Apr  8 11:34:19.8085 2021  7  Location: 
  Thu Apr  8 11:34:19.8085 2021  7  CSeq: 2
  Thu Apr  8 11:34:19.8085 2021  7  cache-control: no-cache
  Thu Apr  8 11:34:19.8085 2021  7  Content-Type: video/MP2T
  Thu Apr  8 11:34:19.8085 2021  7  --- LINE END ---
```

Stream: 0 Stream: 1 are already in use, those are the physical tuners, which are not released yet. Could you please make a log of starting SatPI and first search/tune of VDR.

[153 more replies]

closed 2022-09-22 by Barracuda09