Barracuda09/SATPI #128: Request: Print current pid list in use in Frontend Info

lars18th, 2021-04-22:
Hi,

To best understand what is doing SATPI, and debugging some clients, it will be interesting if inside the "Frontend Info" page (for all type of frontends, including the virtual one) you  print the LIST OF PIDS in use.

I hope you want to implement this.
Thank you!

Barracuda09, 2021-04-22:
Hi @lars18th 

Yes it will be usefull. But first I am trying to make SatPI more suitable in combination with VDR. Therefor I change as well #115.
Because VDR has a 'bug?' that the StreamID needs to be > 0 to request PIDS at the moment it starts with 0. But this has more impact then I expexted.

lars18th, 2021-04-22:
Thank you for the response. I'll wait the time that it's necessary.
God job :+1:

lars18th, 2021-04-25:
Hi @Barracuda09 ,

This request is identical to #79. Sorry to repeat it. Then I close this. :sweat_smile:

closed 2021-04-25 by lars18th