Barracuda09/SATPI #131: Error while crosscompiling

Wonderwuzi2000, 2021-05-15:
Downloaded the newest version of satpi today and tried to cross compile if for testing on a VU+ Vuduo2 STB.

Started with: make debug ENIGMA=yes LIBDVBCSA=yes DVBCA=yes

Received the following error message:

src/main.cpp: In function ‘int main(int, char**)’:
src/main.cpp:349:4: error: ‘DVBCA’ was not declared in this scope
  349 |    DVBCA dvbca;
      |    ^~~~~
src/main.cpp:350:4: error: ‘dvbca’ was not declared in this scope
  350 |    dvbca.startThread();
      |    ^~~~~

Barracuda09, 2021-05-15:
Hi @Wonderwuzi2000 

Thanks for giving this a try!

But unfortunately does **DVBCA=yes** not work, as it is still a work in progress.

Wonderwuzi2000, 2021-05-15:
Hi @Barracuda09 

Sorry - i misunderstood the wiki by thinking i could give a "a work in progress" a first try.