Barracuda09/SATPI #133: TeVii S662 dvbs2 tunning problem

redblue-pkt, 2021-05-27:
Hello,
I have a problem with the correct operation of your program and the tevii662 tuner. It does not tune the frequency. The problem occurs on various platforms. Vlc also does not work.

`[                                 src/main.cpp:341] --- Starting SatPI version: 1.5.437~g0610cee Enigma ---
[                                 src/main.cpp:342] Number of processors online: 16
[                                 src/main.cpp:343] Default network buffer size: 104 KBytes
[                        src/InterfaceAttr.cpp:130] eth0: 192.168.1.233 [74:78:27:3e:06:60]
[                        src/StreamManager.cpp:068] Current DVB_API_VERSION: 5.11
[                        src/StreamManager.cpp:069] Enumerating all devices...
[                   src/input/dvb/Frontend.cpp:179] Detecting frontends in: /dev/dvb
[                       src/base/M3UParser.cpp:059] Error: could not open file: /var/lib/satpi/mapping.m3u
[                   src/input/dvb/Frontend.cpp:488] Frontend Name: Montage Technology M88DS3103
[                   src/input/dvb/Frontend.cpp:561] Frontend Type: Satellite (DVB-S)
[                   src/input/dvb/Frontend.cpp:565] Frontend Type: Satellite (DVB-S2)
[                   src/input/dvb/Frontend.cpp:607] Frontend Freq: 950000 Hz to 2150000 Hz
[                   src/input/dvb/Frontend.cpp:608] Frontend srat: 1000000 symbols/s to 45000000 symbols/s
[                   src/input/dvb/Frontend.cpp:182] Frontends found: 1
[                  src/input/file/TSReader.cpp:056] Setting up TS Reader using path: /var/lib/satpi
[                       src/base/M3UParser.cpp:059] Error: could not open file: /var/lib/satpi/mapping.m3u
[                src/input/stream/Streamer.cpp:059] Setting up TS Streamer
[                src/decrypt/dvbapi/Client.cpp:369] Setting up DVBAPI client
[                       src/base/M3UParser.cpp:059] Error: could not open file: /var/lib/satpi/mapping.m3u
[                       src/base/M3UParser.cpp:059] Error: could not open file: /var/lib/satpi/mapping.m3u
[                       src/base/M3UParser.cpp:059] Error: could not open file: /var/lib/satpi/mapping.m3u
[                       src/base/M3UParser.cpp:059] Error: could not open file: /var/lib/satpi/mapping.m3u
[                           src/Properties.cpp:085] Setting HTTP Port to: 8875
[                           src/Properties.cpp:092] Setting RTSP Port to: 554
[                           src/Properties.cpp:097] Setting WEB Path to: /usr/share/satpi/web
[                           src/Properties.cpp:101] Setting App Data Path to: /var/lib/satpi
[                           src/HttpServer.cpp:058] Setting up HTTP server
[                     src/upnp/ssdp/Server.cpp:104] Setting up SSDP server with BOOTID: 50  annouce interval: 60 Sec
[                           src/RtspServer.cpp:046] Setting up RTSP server
[                    src/socket/SocketAttr.cpp:138] HTTP Connection from 192.168.1.185 Port 60288 with fd: 8
[                src/decrypt/dvbapi/Client.cpp:229] Connecting to OSCam Server failed
[                    src/socket/SocketAttr.cpp:138] RTSP Connection from 192.168.1.136 Port 52400 with fd: 9
[                          src/HttpcServer.cpp:145] RTSP Stream data from client None with IP 192.168.1.136 on Port 52400: SETUP rtsp://192.168.1.233:554/?src=1&freq=11608&pol=h&msys=dvbs2&sr=27500&fec=34&ro=0.35&mtype=8psk&plts=off&pids>
CSeq: 1
Transport: RTP/AVP/TCP;interleaved=0-1
--- LINE END ---
[                        src/StreamManager.cpp:186] Found FrondtendID x - StreamID x - SessionID: 0244860943
[                               src/Stream.cpp:230] Frontend: 0, StreamClient[0] with SessionID 0244860943 for dvbs2
[                   src/input/dvb/Frontend.cpp:404] Frontend: 0, Parsing transport parameters...
[               src/input/dvb/FrontendData.cpp:120] Frontend: 0, New frequency requested, clearing old channel data...
[                   src/input/dvb/Frontend.cpp:411] Frontend: 0, Parsing transport parameters (Finished)
[                          src/HttpcServer.cpp:224] RTSP/1.0 200 OK
CSeq: 1
Session: 0244860943;timeout=60
Transport: RTP/AVP/TCP;unicast;client_ip=192.168.1.136;interleaved=0-1
com.ses.streamID: 100
--- LINE END ---
[                               src/Stream.cpp:291] Frontend: 0, Found Streaming type: RTP/TCP
[                   src/input/dvb/Frontend.cpp:415] Frontend: 0, Updating frontend...
[                   src/input/dvb/Frontend.cpp:695] Frontend: 0, Opened /dev/dvb/adapter0/frontend0 for Read/Write with fd: 10
[              src/input/dvb/delivery/DVBS.cpp:162] Frontend: 0, Start tuning process for DVB-S(2)...
[              src/input/dvb/delivery/DVBS.cpp:204] Frontend: 0, Opened /dev/dvb/adapter0/frontend0 for Writing DiSEqC command with fd: 10
[      src/input/dvb/delivery/DiSEqcSwitch.cpp:107] Frontend: 0, Sending DiSEqC: [e0] [10] [38] [f2] - DiSEqC Src: 0
[              src/input/dvb/delivery/DVBS.cpp:232] Frontend: 0, Set Properties: Frequency 1858000
[                   src/input/dvb/Frontend.cpp:702] Frontend: 0, Waiting on lock...
[                   src/input/dvb/Frontend.cpp:716] Frontend: 0, Not locked yet   (FE status 0x0)...
[                   src/input/dvb/Frontend.cpp:716] Frontend: 0, Not locked yet   (FE status 0x0)...
[                   src/input/dvb/Frontend.cpp:716] Frontend: 0, Not locked yet   (FE status 0x0)...
[                   src/input/dvb/Frontend.cpp:716] Frontend: 0, Not locked yet   (FE status 0x0)...
[                   src/input/dvb/Frontend.cpp:716] Frontend: 0, Not locked yet   (FE status 0x0)...
[                   src/input/dvb/Frontend.cpp:716] Frontend: 0, Not locked yet   (FE status 0x0)...
[                   src/input/dvb/Frontend.cpp:716] Frontend: 0, Not locked yet   (FE status 0x0)...
[                   src/input/dvb/Frontend.cpp:716] Frontend: 0, Not locked yet   (FE status 0x0)...
[                src/decrypt/dvbapi/Client.cpp:229] Connecting to OSCam Server failed
[                   src/input/dvb/Frontend.cpp:716] Frontend: 0, Not locked yet   (FE status 0x0)...
[                   src/input/dvb/Frontend.cpp:716] Frontend: 0, Not locked yet   (FE status 0x0)...
[                   src/input/dvb/Frontend.cpp:716] Frontend: 0, Not locked yet   (FE status 0x0)...
[... 32 lines cut ...]
[                   src/input/dvb/Frontend.cpp:716] Frontend: 0, Not locked yet   (FE status 0x0)...
[                   src/input/dvb/Frontend.cpp:716] Frontend: 0, Not locked yet   (FE status 0x0)...
[                   src/input/dvb/Frontend.cpp:716] Frontend: 0, Not locked yet   (FE status 0x0)...
[                   src/input/dvb/Frontend.cpp:716] Frontend: 0, Not locked yet   (FE status 0x0)...
[                   src/input/dvb/Frontend.cpp:716] Frontend: 0, Not locked yet   (FE status 0x0)...
[                   src/input/dvb/Frontend.cpp:716] Frontend: 0, Not locked yet   (FE status 0x0)...
[                   src/input/dvb/Frontend.cpp:716] Frontend: 0, Not locked yet   (FE status 0x0)...
[                   src/input/dvb/Frontend.cpp:716] Frontend: 0, Not locked yet   (FE status 0x0)...
[                   src/input/dvb/Frontend.cpp:716] Frontend: 0, Not locked yet   (FE status 0x0)...
[                   src/input/dvb/Frontend.cpp:716] Frontend: 0, Not locked yet   (FE status 0x0)...
[                   src/input/dvb/Frontend.cpp:716] Frontend: 0, Not locked yet   (FE status 0x0)...
[                   src/input/dvb/Frontend.cpp:433] Frontend: 0, Updating frontend (Failed)
[                          src/HttpcServer.cpp:224] RTSP/1.0 200 OK
RTP-Info: url=rtsp://192.168.1.233/stream=100
CSeq: 2
Session: 0244860943
Range: npt=0.000-
--- LINE END ---
[                    src/socket/SocketAttr.cpp:138] RTSP Connection from 192.168.1.136 Port 52404 with fd: 12
[                     src/socket/TcpSocket.cpp:106] RTSP Client 192.168.1.136:52400 Connection closed with fd: 9
[                          src/HttpcServer.cpp:145] RTSP Stream data from client None with IP 192.168.1.136 on Port 52402: SETUP rtsp://192.168.1.233:554/?src=1&freq=11608&pol=h&msys=dvbs2&sr=27500&fec=34&ro=0.35&mtype=8psk&plts=off&pids>
CSeq: 1
Transport: RTP/AVP/TCP;interleaved=0-1
--- LINE END ---
`

Barracuda09, 2021-05-28:
Hi @redblue-pkt 

First of all, thanks for your feedback.

Could you give a bit more info on your setup?
- Which Satellite
- DiSEqC Switch connections
- LNB Type
- Enigma Box?

redblue-pkt, 2021-05-30:
My setup: Hotbird with no diseqc connections, universal lnb
Server: orange pi zero, linux: armbian
Client: orange pi pc2, linux: my compilation

I was able to run vlc using the correct pids. But on the neutrino, I can't run it. I can see the epg being downloaded. And that's it.