Barracuda09/SATPI #138: Satpi inverto2 enigma 2 with tvheadend

nickmers, 2021-09-09:
Hello, sorry for my bad English. Thanks for the project. its amazing. I had running enigma2 with lnb universal, satpi connected with tvheadend. All working great.

Now, I bought a lnb unicable (inverto 2). I cant get running satpi with tvheadend with this type lnb. There are specific configuration for this ?

Before, in enigma2 just activate tunner A and B linked to A, rest no cofigured. Now I have cascade configuration tunner but doesnt work in tvheadend. 

Sorry if my question is bad here.

Barracuda09, 2021-09-10:
Hi @nickmers 

First of all, thanks for trying SatPI

Yes the unicable and FBC tuners need a different configuration, please have a look here:
- https://github.com/Barracuda09/SATPI/wiki/04.-Setup-Unicable-and-JESS
- https://github.com/Barracuda09/SATPI/wiki/06.-Setup-DVB-S2(X)-FBC-Tuners
- https://github.com/Barracuda09/SATPI/wiki/07.-SatPI-on-Vu-or-GigaBlue-(Enigma2)

Kind regards,
Marc

nickmers, 2021-09-10:
Hi Marc, thanks for ask really fast. I saw your post before but I cant get running. I will try explain better.

Config in gigablue ue uhd 4k

Tunner A:  advanced, astra, lnb 1, priority (auto), LOF (unicable/jess), maker (inverto), type( idlu-32ul40-unboo-opp), channel scr 1, frequency 1210, connected no.

Tunner B: same that A but scr 2, frecuency 1420, connected yes, connected to A.

Rest of tuners: similar changing scr, frecuency and connected to A.

After that, I go to satpi interface and disable frontend 0 for leave to gigablue.

Frontend 1: enable, fbc connection A, linked, send disecq via root, disecq unicable/jess, channel frecuency 1210, channel slot 1.

Rest of fronted: Similar changing channel frecuency and channel slot.

I dont know if Im doing something wrong.

Thanks a lot

Barracuda09, 2021-09-10:
Hi @nickmers 

I think this should have worked, but you need to disable the tuners in Enigma that you are using in SatPI.

IMPORTANT: SatPI does not use the config what is setup in Enigma.

But what exacly is not working in TvHeadend? 

Kind regards,
Marc

nickmers, 2021-09-10:
Then, in enigma I need put no configured in the tunners that I will manage with Satpi correct?

Now, with unicable in tvheadend I cant open no channel.

Before, with lnb universal, tunner A configured, B in loop to A and rest of tunners no configured was working perfect.

Thanks

Barracuda09, 2021-09-10:
> Then, in enigma I need put no configured in the tunners that I will manage with Satpi correct?

Yes, SatPI uses its own. You need to disable tuners in Enigma (I am no expert in Enigma, I do not use it myself)

You do not have to setup unicable in TvHeadend.

Maybe you could try VLC? with something like this (and adapt it for your situation)
http://ip.of.your.box:8875/?freq=11052.75&sr=22000&msys=dvbs2&mtype=8psk&pol=h&fec=23&pids=0,1,16,17,18,1039,5420,5421,5424

Barracuda09, 2021-09-10:
@nickmers 

The use band starts in SatPI with 0 (So scr 0 is the first uuse band freq)

nickmers, 2021-09-10:
Hi again. Im crazy about this haha. Im testing all week. Now I have tested removing satpi and installing minisatip to comprobe if the problem was satpi or my config with satpi.

With minisatip is working (not ok because many screen pixels and stops) but the channels open.

I dont know if you would interesting in get this working with satpi and this tha I comment help you in something, but if I can help in something I would like a lot.

Barracuda09, 2021-09-10:
Hi @nickmers 

Well it is working here with SatPI, But without Enigma running (I do not need it at all). So it should be a config issue.

nickmers, 2021-09-10:
Ok I will try get running with satpi. If I get I will share my config if that helps. Thanks a lot

Barracuda09, 2021-09-10:
Hi @nickmers 

Yes if you could do that, that would be great!

IMPORTANT:  The user band starts in SatPI with 0 (So scr 0 is the first user band freq)

nickmers, 2021-09-10:
Hello again. I will continue testing. Is rare. With universal lnb, I can have enigma2 on and stream without problems but with lnb unicable inverto 2 I cant have enigma2 on. If its on, the streams on tvheadend not working.

Barracuda09, 2022-02-08:
@nickmers 

Which receiver are you using?

nickmers, 2022-02-11:
Hello @Barracuda09 . I use gigablue uhd ue 4k. I tried but I couldn't get it to work.

Barracuda09, 2022-02-11:
Hi @nickmers 

Ok, thanks for the feedback.

I have a Gigablue quad 4k and also experience problems with Unicable. 
I am still not sure what the problem is. 
If I do not use Frontend 0 (tuner A) so disable it in SatPI, then it works for some reason.
But have to enable `Send DiSEqC Via Linked Root Tuner`

Barracuda09, 2022-02-15:
Hi @nickmers 

Could it be that the LNB `Lof High/Low/Switch` should be changed?

nickmers, 2022-03-23:
> Hola @nickmers
> 
> Puede ser que `Lof High/Low/Switch`se deba cambiar el LNB?

Maybe, I will bought again a unicable and testing.Im intersting to use unicable with gigablue. 

Thanks for answer

Barracuda09, 2022-03-23:
Hi @nickmers 

You could try to change the `Lof High/Low/Switch` to match the Inverto2 LNB

JobbeDeluxe, 2023-05-30:
Hello I have a "Octagon SX88V2 4K DUAL OS" there is "1.6.2.97~g3827b41" Installed and running. But in tvheadend all adapter will be showen as RED, so not connectable. they are showen, yes but not accasable. Is there a way that is was preconfigd with a mismatch? i haven*t change anything. Please let me know what info is needed.... i would like to get this runnig, too.

Barracuda09, 2023-05-30:
[image]

JobbeDeluxe, 2023-06-02:
I feel stupid, that had worked. I never need to activate on my own on all other devices.
Thanks for the info that worked for me.

Barracuda09, 2023-06-02:
@JobbeDeluxe 

No problem, thank you for the feedback!

Barracuda09, 2023-06-08:
Hi @nickmers 

Did you manage to make this work with the inverto2 LNB.
As far as I understand is that this LNB only has 1 LOF of 10400 MHz so you could set it up as described here:
[Setup the correct LNB](https://github.com/Barracuda09/SATPI/wiki/Setup-the-correct-LNB)