Barracuda09/SATPI #141: DIGITALBOX IMPERIAL HD 6i

Cristiann70, 2021-11-10:
Hello
I have a DIGITALBOX IMPERIAL HD 6i and I don't understand how I can install the receiver, but I do not get it, in raspberry and compiled satpi

Barracuda09, 2021-11-10:
Hi @Cristiann70 

Thanks for you interest in SatPI.
Sorry I do not know how to do this as well, I only used it as SAT>IP clients. So I have not used it as server.

Cristiann70, 2021-11-10:
Thank you for the answer and for your work
  Sat-ip imperial it recognizes the satip installed raspberry appears the ip of the connected receiver. My question is:
I can with satip installed in raspberry pi use the imperial turners dvb  to get the vlc channels.

Barracuda09, 2021-11-10:
Hi @Cristiann70 

No, SatPI (server) should be installed on the Imperial receiver to use the tuners by a SAT>IP client like VLC.

lars18th, 2021-11-28:
Hi @Cristiann70 ,

The Digitalbox Imperial HD 6i is a SAT>IP _client_. The SATPI is a SAT>IP _server_. So you can install SATPI in a Raspberry Pi with a DVB tuner and play the channel in the Digitalbox.

If you want a STB with SAT>IP _server_ capabilities, then purchase one with Linux (Enigma2) and then you'll can install SATPI on it. Or purchase another one with a SAT>IP server integrated on it (like some based on ALI chipsets, like Smart or Xoro; but from my expirience forget them).

I hope this helps you.

closed 2024-04-10 by Barracuda09