Barracuda09/SATPI #144: MIPSEL: Not listening on TCP ports

robbieb43, 2021-12-14:
Hi Marc

As you know I have been trying to unsuccessfully use satpi on my elderly DM800 mipsel box.

I have opened a new issue but if you think this is is a dead-end issue due to old hardware or old kernels then I am happy to re-close it, but at least it might save others having to tread the same path.  So for the benefit of others, in this [issue](https://github.com/Barracuda09/SATPI/discussions/143)  you kindly helped me through the cross compilation process but I fund the resultant executable did not open the http & rtsp ports and there was no logging info.  Subsequently I have tracked down a bin compiled by others of satpi_1.5-0-r5 mipsel and I get very similar results in that it fails to open the http ports and the CPU goes to 99%.   However there is now some startup information if run in foreground:

[                                src/Satpi.cpp:281] --- starting SatPI version: 1.5.158~gd03a698 ---
[                                src/Satpi.cpp:282] Number of processors online: 1
[                                src/Satpi.cpp:283] Default network buffer size: 50 KBytes
[                        src/InterfaceAttr.cpp:119] eth0: 192.168.1.71 [00:09:34:2b:57:57]
[                        src/StreamManager.cpp:074] Current DVB_API_VERSION: 5.10
[                   src/input/dvb/Frontend.cpp:156] Detecting frontends in: /dev/dvb
[                   src/input/dvb/Frontend.cpp:567] Frontend Name: cx24116
[                   src/input/dvb/Frontend.cpp:589] Unable to enumerate the delivery systems, retrying via old API Call
[                   src/input/dvb/Frontend.cpp:644] Frontend Type: Satellite (DVB-S2)
[                   src/input/dvb/Frontend.cpp:639] Frontend Type: Satellite (DVB-S)
[                   src/input/dvb/Frontend.cpp:686] Frontend Freq: 925000 Hz to 2175000 Hz
[                   src/input/dvb/Frontend.cpp:687] Frontend srat: 2 symbols/s to 45000000 symbols/s
[                   src/input/dvb/Frontend.cpp:159] Frontends found: 1
[                  src/input/file/TSReader.cpp:042] Setting up TS Reader using path: .
[                src/input/stream/Streamer.cpp:042] Setting up TS Streamer
[                    src/socket/SocketAttr.cpp:070] socket: Invalid argument (code 22)
[                     src/socket/TcpSocket.cpp:110] TCP Server handle failed
[                    src/socket/SocketAttr.cpp:070] socket: Invalid argument (code 22)
[                     src/socket/TcpSocket.cpp:110] TCP Server handle failed
[                           src/HttpServer.cpp:055] Setting up HTTP server
[                           src/RtspServer.cpp:047] Setting up RTSP server

Interestingly there is nothing after the Setting-up HTTP server/Setting up RTSP server lines.  I note also the complaint about the old API.   I guess this may be pointing back an issue with the old enigma2 version.  

If the above does not provide any new pointers or clues I suggest I close the issue based on the rationale already applied.

Many thanks

Rob

Barracuda09, 2021-12-14:
Hi @robbieb43 

This is quite an old version of SatPI d03a6983a536ea83f9c62892992850a88ba7ada8 (from around 2016) did you run it as `root` or with `sudo`

Barracuda09, 2021-12-14:
Hi @robbieb43

Here is something I have compiled + libs for MIPS and MIPSEL
[satpi_mips_libs.tar.gz](https://github.com/Barracuda09/SATPI/files/7714137/satpi_mips_libs.tar.gz)
[satpi_mipsel_libs.tar.gz](https://github.com/Barracuda09/SATPI/files/7714220/satpi_mipsel_libs.tar.gz)

robbieb43, 2021-12-15:
Hi Marc

 

Effectively yes, as the DM800 runs everything as root :-/  - I think the idea that it is an external factor like permissions is a good one though and I will have a dig around the config aspects.

 

Rob

robbieb43, 2021-12-15:
Very kind – I will give these a try later.