Barracuda09/SATPI #145: SatPi on Enigma2 with TVheadend - wrong tuner order

SaulGoodman1337, 2022-01-04:
Hi,
first of all, thanks for SatPI!

i have now 2 Cables from my LNB to my 4 Tuners.
In TVHeadend i have configured:

Tuner 1: Master
Tuner 2: Slave of 1
Tuner 3: Master
Tuner 4: Slave of 3

Now if i use Tuner 1 and then would like to use another tuner - TVHeadend is correctly use Tuner 3 BUT SatPI is using Tuner 1 (0-3).
It seems that the tuner order of 0-3 is not correct for TVHeadend. Satpi uses always the next tuner. 
i expected that Satpi is Using Tuner 2 when iam using Tuner 3 in TVHeadend.

hope that anybody can help me

SaulGoodman1337, 2022-01-05:
I will try to express myself more clearly.
Is it possible to influence the order of the tuners?

Barracuda09, 2022-01-05:
Hi @SaulGoodman1337 

Could you give a log of SatPI when this situation occurs (just the two channel requests)?

There is some option in the protocol that should handle this requests. 

Thanks and regards,
Marc

SaulGoodman1337, 2022-01-06:
> Hi @SaulGoodman1337
> 
> Could you give a log of SatPI when this situation occurs (just the two channel requests)?
> 
> There is some option in the protocol that should handle this requests.
> 
> Thanks and regards, Marc

Thanks for helping Marc.
Sure.

`
  Thu Jan  6 10:51:50.2465 2022  7  --- LINE END ---
  Thu Jan  6 10:51:50.2465 2022  7  Session: 0245332568
  Thu Jan  6 10:51:50.2465 2022  7  Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
  Thu Jan  6 10:51:50.2465 2022  7  CSeq: 12
  Thu Jan  6 10:51:50.2465 2022  7  RTSP/1.0 200 OK
  Thu Jan  6 10:51:50.2464 2022  6  Frontend: 0, StreamClient[0] with SessionID 0245332568
  Thu Jan  6 10:51:50.2463 2022  6  Found FrondtendID x - StreamID x - SessionID: 0245332568
  Thu Jan  6 10:51:50.2462 2022  7  --- LINE END ---
  Thu Jan  6 10:51:50.2462 2022  7  CSeq: 12
  Thu Jan  6 10:51:50.2462 2022  7  Session: 0245332568
  Thu Jan  6 10:51:50.2462 2022  7  RTSP Stream data from client None with IP 192.168.178.2 on Port 36260: OPTIONS rtsp://192.168.178.63/ RTSP/1.0
  Thu Jan  6 10:51:45.5597 2022  6  Root Device Client 192.168.178.2 : tries to discover the network, sending reply back
  Thu Jan  6 10:51:35.5589 2022  6  Root Device Client 192.168.178.2 : tries to discover the network, sending reply back
  Thu Jan  6 10:51:32.0770 2022  6  Root Device Client 192.168.178.2 : tries to discover the network, sending reply back
  Thu Jan  6 10:51:32.0767 2022  6  Root Device Client 192.168.178.2 : tries to discover the network, sending reply back
  Thu Jan  6 10:51:30.2590 2022  6  HTTP Client 192.168.178.95:63168 Connection closed with fd: 10
  Thu Jan  6 10:51:30.2579 2022  6  HTTP Client 192.168.178.95:63172 Connection closed with fd: 18
  Thu Jan  6 10:51:30.2578 2022  6  HTTP Client 192.168.178.95:63171 Connection closed with fd: 14
  Thu Jan  6 10:51:30.2577 2022  6  HTTP Client 192.168.178.95:63169 Connection closed with fd: 11
  Thu Jan  6 10:51:30.2575 2022  6  HTTP Client 192.168.178.95:63170 Connection closed with fd: 12
  Thu Jan  6 10:51:25.5579 2022  6  Root Device Client 192.168.178.2 : tries to discover the network, sending reply back
  Thu Jan  6 10:51:20.2218 2022  7  --- LINE END ---
  Thu Jan  6 10:51:20.2218 2022  7  Session: 0245332568
  Thu Jan  6 10:51:20.2218 2022  7  Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
  Thu Jan  6 10:51:20.2218 2022  7  CSeq: 11
  Thu Jan  6 10:51:20.2218 2022  7  RTSP/1.0 200 OK
  Thu Jan  6 10:51:20.2216 2022  6  Frontend: 0, StreamClient[0] with SessionID 0245332568
  Thu Jan  6 10:51:20.2205 2022  6  Found FrondtendID x - StreamID x - SessionID: 0245332568
  Thu Jan  6 10:51:20.2203 2022  7  --- LINE END ---
  Thu Jan  6 10:51:20.2203 2022  7  CSeq: 11
  Thu Jan  6 10:51:20.2203 2022  7  Session: 0245332568
  Thu Jan  6 10:51:20.2203 2022  7  RTSP Stream data from client None with IP 192.168.178.2 on Port 36260: OPTIONS rtsp://192.168.178.63/ RTSP/1.0
  Thu Jan  6 10:51:15.5583 2022  6  Root Device Client 192.168.178.2 : tries to discover the network, sending reply back
  Thu Jan  6 10:51:05.5583 2022  6  Root Device Client 192.168.178.2 : tries to discover the network, sending reply back
  Thu Jan  6 10:50:59.3165 2022  6  RTSP Client 192.168.178.2:36814 Connection closed with fd: 8
  Thu Jan  6 10:50:59.3159 2022  7  --- LINE END ---
  Thu Jan  6 10:50:59.3159 2022  7  Session: 0272911665
  Thu Jan  6 10:50:59.3159 2022  7  CSeq: 4
  Thu Jan  6 10:50:59.3159 2022  7  RTSP/1.0 200 OK
  Thu Jan  6 10:50:59.3155 2022  6  Frontend: 1, Closing /dev/dvb/adapter0/frontend1 fd: 9
  Thu Jan  6 10:50:59.3105 2022  6  Frontend: 1, Destroy RTCP/TCP stream to 192.168.178.2:36814
  Thu Jan  6 10:50:59.3105 2022  6  Frontend: 1, Destroy RTP/TCP stream to 192.168.178.2:36814
  Thu Jan  6 10:50:59.3105 2022  6  Frontend: 1, Teardown StreamClient[0] with SessionID 0272911665
  Thu Jan  6 10:50:59.3104 2022  6  Frontend: 1, StreamClient[0] with SessionID 0272911665
  Thu Jan  6 10:50:59.3103 2022  6  Found FrondtendID 1 - StreamID 101 - SessionID 0272911665
  Thu Jan  6 10:50:59.3101 2022  7  --- LINE END ---
  Thu Jan  6 10:50:59.3101 2022  7  CSeq: 4
  Thu Jan  6 10:50:59.3101 2022  7  Session: 0272911665
  Thu Jan  6 10:50:59.3101 2022  7  RTSP Stream data from client None with IP 192.168.178.2 on Port 36814: TEARDOWN rtsp://192.168.178.63/stream=101 RTSP/1.0
  Thu Jan  6 10:50:55.5547 2022  6  Root Device Client 192.168.178.2 : tries to discover the network, sending reply back
  Thu Jan  6 10:50:53.5392 2022  7  --- LINE END ---
  Thu Jan  6 10:50:53.5392 2022  7  Range: npt=0.000-
  Thu Jan  6 10:50:53.5392 2022  7  Session: 0272911665
  Thu Jan  6 10:50:53.5392 2022  7  CSeq: 3
  Thu Jan  6 10:50:53.5392 2022  7  RTP-Info: url=rtsp://192.168.178.63/stream=101
  Thu Jan  6 10:50:53.5392 2022  7  RTSP/1.0 200 OK
  Thu Jan  6 10:50:53.5391 2022  6  Frontend: 1, Updating frontend (Failed)
  Thu Jan  6 10:50:53.4790 2022  6  Frontend: 1, Not locked yet   (FE status 0x0)...
[... 79 lines cut ...]
  Thu Jan  6 10:50:51.3352 2022  7  Frontend: 1, Parsing transport parameters (Finished)
  Thu Jan  6 10:50:51.3345 2022  6  Frontend: 1, New frequency requested, clearing old channel data...
  Thu Jan  6 10:50:51.3345 2022  6  Frontend: 1, Parsing transport parameters...
  Thu Jan  6 10:50:51.3344 2022  6  Frontend: 1, StreamClient[0] with SessionID 0272911665 for dvbs2
  Thu Jan  6 10:50:51.3344 2022  6  Frontend: 0, New session but this stream is in use, skipping...
[... cut ...]
  Thu Jan  6 10:50:50.1385 2022  6  Frontend: 0, StreamClient[0] with SessionID 0245332568
  Thu Jan  6 10:50:50.1384 2022  6  Found FrondtendID x - StreamID x - SessionID: 0245332568
  Thu Jan  6 10:50:50.1382 2022  7  --- LINE END ---
  Thu Jan  6 10:50:50.1382 2022  7  CSeq: 10
  Thu Jan  6 10:50:50.1382 2022  7  Session: 0245332568
  Thu Jan  6 10:50:50.1382 2022  7  RTSP Stream data from client None with IP 192.168.178.2 on Port 36260: OPTIONS rtsp://192.168.178.63/ RTSP/1.0`
  
And here I have a screenshot where the problem should be visible
[image]

Barracuda09, 2022-01-06:
Hi @SaulGoodman1337 

Which version do you use? Which receiver are you using? (which cpu does it have?) so I can perhaps give you an updated version

SaulGoodman1337, 2022-01-06:
XTrend Et10000, 4 sat Tuner, MIPS 
Version 1.5.442~gf04c050

Barracuda09, 2022-01-06:
Hi @SaulGoodman1337 

Here is a new version, it is a debug version, just copy it over the original. (Need to stop SatPI first)

[satpi_mips_debug.tar.gz](https://github.com/Barracuda09/SATPI/files/7822375/satpi_mips_debug.tar.gz)

SaulGoodman1337, 2022-01-06:
> Hi @SaulGoodman1337
> 
> Here is a new version, it is a debug version, just copy it over the original. (Need to stop SatPI first)
> 
> [satpi_mips_debug.tar.gz](https://github.com/Barracuda09/SATPI/files/7822375/satpi_mips_debug.tar.gz)

-bash: /usr/bin/satpi: cannot execute binary file: Exec format error
??

Barracuda09, 2022-01-06:
What does this show `file /usr/bin/satpi`

SaulGoodman1337, 2022-01-06:
> What does this show `file /usr/bin/satpi`

> What does this show `file /usr/bin/satpi`

`root@et10000:/usr/bin# file satpi
satpi: ELF 32-bit MSB executable, MIPS, MIPS32 version 1 (SYSV), dynamically linked, interpreter /lib/ld.so.1, for GNU/Linux 3.2.0, with debug_info, not stripped
root@et10000:/usr/bin# cat /proc/cpuinfo
system type             : BCM7425B2 STB platform
machine                 : Unknown
processor               : 0
cpu model               : Broadcom BMIPS5000 V1.1  FPU V0.1
BogoMIPS                : 864.25
cpu MHz                 : 1305.100
wait instruction        : yes
microsecond timers      : yes
tlb_entries             : 64
extra interrupt vector  : yes
hardware watchpoint     : no
isa                     : mips1 mips2 mips32r1
ASEs implemented        :
shadow register sets    : 1
kscratch registers      : 0
package                 : 0
core                    : 0
VCED exceptions         : not available
VCEI exceptions         : not available`

Barracuda09, 2022-01-06:
What does the old one show with `file /usr/bin/satpi`

SaulGoodman1337, 2022-01-06:
i dont have a backup so this is for opkg

root@et10000:/usr/bin# file opkg
opkg: ELF 32-bit LSB executable, MIPS, MIPS32 version 1 (SYSV), dynamically linked, interpreter /lib/ld.so.1, BuildID[sha1]=d28da492fdba5dc677a8288d44937a6851c8e680, for GNU/Linux 3.2.0, stripped

Barracuda09, 2022-01-06:
Ok its mipsel (LSB)

[satpi_mipsel_debug.tar.gz](https://github.com/Barracuda09/SATPI/files/7822528/satpi_mipsel_debug.tar.gz)

SaulGoodman1337, 2022-01-06:
> Ok its mipsel (LSB)
> 
> [satpi_mipsel_debug.tar.gz](https://github.com/Barracuda09/SATPI/files/7822528/satpi_mipsel_debug.tar.gz)

sorry, my fault.

Now Tuner 1 in TVheadend wants Tuner 1 on SatPi
FrontendID 1 in TVheadend must goes to Tuner 0

Barracuda09, 2022-01-06:
Could you give another log of SatPI. 
You can enable in SatPI config page `Log messages to syslog` then you can log longer. then get the log from `dmesg`

SaulGoodman1337, 2022-01-06:
Thu Jan  6 16:12:30.4136 2022  7  <CRLF>
  Thu Jan  6 16:12:30.4136 2022  7  Range: npt=0.000-
  Thu Jan  6 16:12:30.4136 2022  7  Session: 0279847755
  Thu Jan  6 16:12:30.4136 2022  7  CSeq: 2
  Thu Jan  6 16:12:30.4136 2022  7  RTP-Info: url=rtsp://192.168.178.63/stream=101
  Thu Jan  6 16:12:30.4136 2022  7  RTSP/1.0 200 OK
  Thu Jan  6 16:12:30.4136 2022  7  [                          src/HttpcServer.cpp:217] Send reply in 5 ms
  Thu Jan  6 16:12:30.4133 2022  6  [                   src/input/dvb/Frontend.cpp:435] Frontend: 1, Updating frontend (Finished in 2 ms)
  Thu Jan  6 16:12:30.4123 2022  7  [                          src/mpegts/Filter.h:158] Frontend: 1, Set filter PID: 1795
  Thu Jan  6 16:12:30.4120 2022  7  [                          src/mpegts/Filter.h:158] Frontend: 1, Set filter PID: 1791
  Thu Jan  6 16:12:30.4114 2022  7  [                          src/mpegts/Filter.h:158] Frontend: 1, Set filter PID: 0102
  Thu Jan  6 16:12:30.4110 2022  6  [                          src/mpegts/Filter.h:133] Frontend: 1, Updating PID filters...
  Thu Jan  6 16:12:30.4107 2022  6  [                   src/input/dvb/Frontend.cpp:414] Frontend: 1, Updating frontend...
  Thu Jan  6 16:12:30.4103 2022  7  [                   src/input/dvb/Frontend.cpp:410] Frontend: 1, Parsing transport parameters (Finished)
  Thu Jan  6 16:12:30.4098 2022  6  [                   src/input/dvb/Frontend.cpp:403] Frontend: 1, Parsing transport parameters...
  Thu Jan  6 16:12:30.4093 2022  6  [                               src/Stream.cpp:207] Frontend: 1, StreamClient[0] with SessionID 0279847755
  Thu Jan  6 16:12:30.4090 2022  6  [                        src/StreamManager.cpp:192] Found FrondtendID 1 - StreamID 101 - SessionID 0279847755
  Thu Jan  6 16:12:30.4082 2022  7  <CRLF>
  Thu Jan  6 16:12:30.4082 2022  7  CSeq: 2
  Thu Jan  6 16:12:30.4082 2022  7  Session: 0279847755
  Thu Jan  6 16:12:30.4082 2022  7  [                          src/HttpcServer.cpp:150] RTSP Stream data from client None with IP 192.168.178.2 on Port 58778: PLAY rtsp://192.168.178.63/stream=101?addpids=1,16,17,18,102,1791,1795 RTSP/1.0
  Thu Jan  6 16:12:30.4071 2022  7  <CRLF>
  Thu Jan  6 16:12:30.4071 2022  7  com.ses.streamID: 101
  Thu Jan  6 16:12:30.4071 2022  7  Transport: RTP/AVP/TCP;unicast;client_ip=192.168.178.2;interleaved=0-1
  Thu Jan  6 16:12:30.4071 2022  7  Session: 0279847755;timeout=60
  Thu Jan  6 16:12:30.4071 2022  7  CSeq: 1
  Thu Jan  6 16:12:30.4071 2022  7  RTSP/1.0 200 OK
  Thu Jan  6 16:12:30.4071 2022  7  [                          src/HttpcServer.cpp:217] Send reply in 1086 ms
  Thu Jan  6 16:12:30.4068 2022  6  [              src/output/StreamThreadBase.cpp:133] Frontend: 1, Start RTP/TCP stream to 192.168.178.2:58778
  Thu Jan  6 16:12:30.4057 2022  6  [          src/output/StreamThreadRtcpBase.cpp:063] Frontend: 1, Start RTCP/TCP stream to 192.168.178.2:58778
  Thu Jan  6 16:12:30.4053 2022  6  [            src/output/StreamThreadRtpTcp.cpp:063] Frontend: 1, RTP/TCP set network buffer size: 450 KBytes
  Thu Jan  6 16:12:30.4049 2022  6  [                   src/input/dvb/Frontend.cpp:435] Frontend: 1, Updating frontend (Finished in 1078 ms)
  Thu Jan  6 16:12:30.4035 2022  7  [                          src/mpegts/Filter.h:158] Frontend: 1, Set filter PID: 0018
  Thu Jan  6 16:12:30.4032 2022  7  [                          src/mpegts/Filter.h:158] Frontend: 1, Set filter PID: 0017
  Thu Jan  6 16:12:30.4027 2022  7  [                          src/mpegts/Filter.h:158] Frontend: 1, Set filter PID: 0016
  Thu Jan  6 16:12:30.4024 2022  7  [                          src/mpegts/Filter.h:158] Frontend: 1, Set filter PID: 0001
  Thu Jan  6 16:12:30.4021 2022  7  [                          src/mpegts/Filter.h:158] Frontend: 1, Set filter PID: 0000
  Thu Jan  6 16:12:30.4012 2022  6  [                   src/input/dvb/Frontend.cpp:762] Frontend: 1, Set DMX_SET_SOURCE with (Src: 1 - Offset: 0)
  Thu Jan  6 16:12:30.4007 2022  6  [                   src/input/dvb/Frontend.cpp:746] Frontend: 1, Set DMX buffer size to 18874368 Bytes
  Thu Jan  6 16:12:30.3975 2022  6  [                   src/input/dvb/Frontend.cpp:740] Frontend: 1, Opened /dev/dvb/adapter0/demux1 using fd: 11
  Thu Jan  6 16:12:30.3972 2022  6  [                          src/mpegts/Filter.h:133] Frontend: 1, Updating PID filters...
  Thu Jan  6 16:12:30.3969 2022  6  [                   src/input/dvb/Frontend.cpp:697] Frontend: 1, Not locked yet   (Timeout 1017 ms)...
  Thu Jan  6 16:12:29.4745 2022  6  [                   src/input/dvb/Frontend.cpp:692] Frontend: 1, Not locked yet   (FE status 0x00)...
  Thu Jan  6 16:12:29.4743 2022  6  [                   src/input/dvb/Frontend.cpp:678] Frontend: 1, Tuned, waiting on lock...
  Thu Jan  6 16:12:29.4740 2022  7  [              src/input/dvb/delivery/DVBS.cpp:233] Frontend: 1, Set Properties: Frequency 1420750
  Thu Jan  6 16:12:29.4409 2022  6  [         src/input/dvb/delivery/DiSEqcLnb.cpp:073] Frontend: 1, Sending LNB: Mini-Switch Src: 1
  Thu Jan  6 16:12:29.4405 2022  6  [              src/input/dvb/delivery/DVBS.cpp:205] Frontend: 1, Opened /dev/dvb/adapter0/frontend1 for Writing DiSEqC command with fd: 10
  Thu Jan  6 16:12:29.4402 2022  6  [              src/input/dvb/delivery/DVBS.cpp:163] Frontend: 1, Start tuning process for DVB-S(2)...
[... cut ...]
  Thu Jan  6 16:11:52.9942 2022  6  [                        src/StreamManager.cpp:192] Found FrondtendID 1 - StreamID -1 - SessionID 0263668040
  Thu Jan  6 16:11:52.9921 2022  7  <CRLF>
  Thu Jan  6 16:11:52.9921 2022  7  CSeq: 1
  Thu Jan  6 16:11:52.9921 2022  7  Transport: RTP/AVP/TCP;interleaved=0-1
  Thu Jan  6 16:11:52.9921 2022  7  [                          src/HttpcServer.cpp:150] RTSP Stream data from client None with IP 192.168.178.2 on Port 58498: SETUP rtsp://192.168.178.63/?src=2&fe=1&freq=11170.75&sr=22000&msys=dvbs2&mtype=8psk&pol=h&fec=34&ro=0.35&plts=on&pids=0 RTSP/1.0

Barracuda09, 2022-01-06:
Hi @SaulGoodman1337 

Sorry, I have still a bit trouble understanding the problem. This version has fix ed4a84d0d6df9c75b3dd307975cfcecef7e4d4bb ,so it should use the requested frontend.

SaulGoodman1337, 2022-01-06:
> Hi @SaulGoodman1337
> 
> Sorry, I have still a bit trouble understanding the problem. This version has fix [ed4a84d](https://github.com/Barracuda09/SATPI/commit/ed4a84d0d6df9c75b3dd307975cfcecef7e4d4bb) ,so it should use the requested frontend.

I think the problem is Tvheadend is using Frontend 1-4.
SATpi is using 0-3. Tvheadend requests FE1 as first Frontend.Can this be ? Makes on me so the impression

Barracuda09, 2022-01-06:
> Tvheadend requests FE1 as first Frontend.Can this be ?

Yes you could be right. I try to check and fix this.

Barracuda09, 2022-01-06:
Hi @SaulGoodman1337 

Could you please try this version?
[satpi_mipsel_debug.tar.gz](https://github.com/Barracuda09/SATPI/files/7823507/satpi_mipsel_debug.tar.gz)

SaulGoodman1337, 2022-01-06:
> Hi @SaulGoodman1337
> 
> Could you please try this version? [satpi_mipsel_debug.tar.gz](https://github.com/Barracuda09/SATPI/files/7823507/satpi_mipsel_debug.tar.gz)

now i get a force close. only on tuner0 (tvheadend=1). no log entry.

SaulGoodman1337, 2022-01-21:
hi,
could you possibly compile this for me for the gigablue UE 4K? Sorry, I have in the meantime changed the receiver and LNB and therefore no longer contact me. I would still like to test the whole thing

Barracuda09, 2022-01-21:
Sure, here is the ARM build
[satpi_armv7ve_V1.6.2-11-g6b90b13.tar.gz](https://github.com/Barracuda09/SATPI/files/7915453/satpi_armv7ve_V1.6.2-11-g6b90b13.tar.gz)

SaulGoodman1337, 2022-01-23:
Found FrondtendID: 3 (fe=4)  StreamID: 103  SessionID: 0302071157
 
looks like it's working. i'll test it for a few more days and get back to you. thank you.

Barracuda09, 2022-01-23:
@SaulGoodman1337 

Thanks for testing

Barracuda09, 2022-10-03:
Hi @SaulGoodman1337 

After reading this again, I think I fixed this with issue #158.

The problem was that FrondtendID in SatPI starts at 0 and **src=x** parameter starts at 1.

Thanks again for your feedback and testing

closed 2022-11-13 by SaulGoodman1337