Barracuda09/SATPI #146: Playback often breaks

taojishou, 2022-01-08:
Sat Jan  8 19:03:29.2097 2022  7  Frontend: 0, Send Filter Data with size 106 for demux 0 filter 1 PID 1283 TableID 0x81 0x70 0x61 0x89 0x51
  Sat Jan  8 19:03:29.2263 2022  6  Frontend: 0, SDT - PID 0017: sectionLength: 159  tableDataSize: 188  secNr: 0  lastSecNr: 0  currSecNr: 0
  Sat Jan  8 19:03:29.2264 2022  6  Frontend: 0, SDT - Section Length: 159  Transport Stream ID: 27  Version: 195  secNr: 0  lastSecNr: 0  NetworkID: 7690  CRC: 0x2A96AB66
  Sat Jan  8 19:03:29.2265 2022  6  Frontend: 0,  serviceID: 0x0A8D - 02701  EIT: 0xFF  NetworkName:   ChannelName: ÖØÇìÎÀÊÓ£¨¸ßÇå£©
  Sat Jan  8 19:03:29.2265 2022  6  Frontend: 0,  serviceID: 0x0A8F - 02703  EIT: 0xFF  NetworkName:   ChannelName: ¹ã¶«ÌåÓý£¨¸ßÇå£©
  Sat Jan  8 19:03:29.2265 2022  6  Frontend: 0,  serviceID: 0x0A8E - 02702  EIT: 0xFF  NetworkName:   ChannelName: ÒËºÍ¹ºÎï£¨¸ßÇå£©
  Sat Jan  8 19:03:29.2266 2022  6  Frontend: 0,  serviceID: 0x0A90 - 02704  EIT: 0xFF  NetworkName:   ChannelName: CCTV-11Ï·Çú£¨¸ßÇå£©
  Sat Jan  8 19:03:29.3095 2022  7  Frontend: 0, Send Filter Data with size 106 for demux 0 filter 1 PID 1283 TableID 0x81 0x70 0x61 0x89 0x51
  Sat Jan  8 19:03:29.4092 2022  7  Frontend: 0, Send Filter Data with size 106 for demux 0 filter 1 PID 1283 TableID 0x81 0x70 0x61 0x89 0x51
  Sat Jan  8 19:03:29.5089 2022  7  Frontend: 0, Send Filter Data with size 106 for demux 0 filter 1 PID 1283 TableID 0x81 0x70 0x61 0x89 0x51
  Sat Jan  8 19:03:29.6087 2022  7  Frontend: 0, Send Filter Data with size 106 for demux 0 filter 1 PID 1283 TableID 0x81 0x70 0x61 0x89 0x51
  Sat Jan  8 19:03:29.6780 2022  6  Frontend: 0, Start filter PID: 1283  demux: 0  filter: 1 (data 0x80 mask 0xFF 0x00)
  Sat Jan  8 19:03:29.7840 2022  7  Frontend: 0, Received even(0x00) CW: 0x57 0xA2 0x36 0x2F 0x16 0x0F 0xE2 0x07  index: 0
  Sat Jan  8 19:03:29.7841 2022  7  Frontend: 0, Received odd(0x01) CW: 0x43 0x9E 0x00 0xE1 0x82 0x59 0xDA 0xB5  index: 0
  Sat Jan  8 19:03:29.7841 2022  7  Frontend: 0, Receive ECM Info System: tongfang  Reader: dgtv  Source: local  Protocol: mouse  ECM Time: 106
  Sat Jan  8 19:03:34.1819 2022  6  Frontend: 0, Start filter PID: 0017  demux: 0  filter: 2 (data 0x42 mask 0xFF 0x00)
  Sat Jan  8 19:03:34.5780 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:35.1097 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:35.6416 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:36.1731 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:36.6959 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:37.2276 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:37.7591 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:38.2990 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:38.8226 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:39.3626 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:39.8871 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:40.4260 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:40.6835 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:40.9912 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:41.5226 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:42.0466 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:42.5781 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:43.1098 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:43.3588 2022  7  Frontend: 0, Parity changed from 1 to 0, decrypting batch size 23
  Sat Jan  8 19:03:43.3598 2022  7  Frontend: 0, Parity changed from 0 to 1, decrypting batch size 1
  Sat Jan  8 19:03:43.3607 2022  7  Frontend: 0, Parity changed from 1 to 0, decrypting batch size 9
  Sat Jan  8 19:03:43.3617 2022  7  Frontend: 0, Parity changed from 0 to 1, decrypting batch size 1
  Sat Jan  8 19:03:43.3626 2022  7  Frontend: 0, Parity changed from 1 to 0, decrypting batch size 5
  Sat Jan  8 19:03:43.3635 2022  7  Frontend: 0, Parity changed from 0 to 1, decrypting batch size 1
  Sat Jan  8 19:03:43.3645 2022  7  Frontend: 0, Parity changed from 1 to 0, decrypting batch size 14
  Sat Jan  8 19:03:43.3654 2022  7  Frontend: 0, Parity changed from 0 to 1, decrypting batch size 1
  Sat Jan  8 19:03:43.3671 2022  7  Frontend: 0, Parity changed from 1 to 0, decrypting batch size 14
  Sat Jan  8 19:03:43.3681 2022  7  Frontend: 0, Parity changed from 0 to 1, decrypting batch size 1
  Sat Jan  8 19:03:43.3690 2022  7  Frontend: 0, Parity changed from 1 to 0, decrypting batch size 5
  Sat Jan  8 19:03:43.6413 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:43.6914 2022  7  Frontend: 0, Send Filter Data with size 106 for demux 0 filter 1 PID 1283 TableID 0x80 0x70 0x61 0x89 0x52
  Sat Jan  8 19:03:43.6919 2022  6  Frontend: 0, Start filter PID: 1283  demux: 0  filter: 1 (data 0x81 mask 0xFF 0x00)
  Sat Jan  8 19:03:43.7926 2022  7  Frontend: 0, Received odd(0x01) CW: 0x00 0x0E 0x80 0x8E 0x08 0xD4 0x4A 0x26  index: 0
  Sat Jan  8 19:03:43.7928 2022  7  Frontend: 0, Receive ECM Info System: tongfang  Reader: dgtv  Source: local  Protocol: mouse  ECM Time: 100
  Sat Jan  8 19:03:44.1730 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:44.6978 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:45.2295 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:45.7600 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:46.2933 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:46.8244 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:47.3568 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:47.8870 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:48.4270 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:48.6587 2022  6  Root Device Client 10.6.7.150 : tries to discover the network, sending reply back
  Sat Jan  8 19:03:48.6844 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:48.9929 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:49.4937 2022  6  Frontend: 0, Start filter PID: 0001  demux: 0  filter: 3 (data 0x01 mask 0xFF 0x00)
[... cut ...]
  Sat Jan  8 19:03:54.8244 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:55.3575 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:55.8878 2022  7  Frontend: 0, Send Filter Data with size 168 for demux 0 filter 2 PID 0017 TableID 0x42 0xF0 0x9F 0x00 0x1B
  Sat Jan  8 19:03:56.1810 2022  6  Root Device Client 10.6.7.115 : tries to discover the network, sending reply back

taojishou, 2022-01-08:
We found that when accessing the HTTP stream, the frontend overview could not get the resources 

frontendoverview.html:23 Uncaught TypeError: Cannot read properties of null (reading 'getElementsByTagName')
    at buildcontent (frontendoverview.html:23)
    at xmlloaded (frontendoverview.html:65)
    at XMLHttpRequest.xmlhttp.onreadystatechange (loadxmldoc.js:17)

Barracuda09, 2022-01-09:
> We found that when accessing the HTTP stream, the frontend overview could not get the resources
>
> frontendoverview.html:23 Uncaught TypeError: Cannot read properties of null (reading [...]

Maybe you need to update the web directory as well on the server sized

Barracuda09, 2022-02-08:
@taojishou 

Did you try updating the WebUI to fix https://github.com/Barracuda09/SATPI/issues/146#issuecomment-1007969416 ?

Maybe there is a problem with `ChannelName: ÖØÇìÎÀÊÓ£¨¸ßÇå£©` it seems not to show in the correct `code page`
Text is encoded in UTF-8 at the moment

Could you give some more info about playback problem?

Barracuda09, 2022-03-01:
@taojishou 

Could you please try again? 
I did some fixes for OSCam dvbapi have tested with r11704